doudz/zigate #1: Issue with Ikea lights

Emmegiemme, 2018-02-17:
Hello there,

great work! I love your job and effort on this matter!

I am using your last version 0.8.2, I have been able to paring and greatly control an Ikea bulb by sending ON/OFF command ("z.action_onoff(....)") and control the light level ("z.action_move_level_onoff(....)"). 

Unfortunately the documented methods ("z.devices[0].endpoints" and "z.devices[0].properties") they do not work as they reply with "{ }" and with an empty string.

Of course, I can ensure that I am okay with the pairing, as the command z.devices[0].addr it does reply with the short zigbee address of my bulb.

Cheers
Mirco

doudz, 2018-02-17:
Currently endpoints discovery only happens at pairing time.
Did you pair the device with the version 0.8.2 ?
If not I suggest to pair the device again or to send a node descriptor request
`z.node_descriptor_request(addr)`

doudz, 2018-02-17:
And `z.active_endpoint_request(addr)` to discover the endpoints

Emmegiemme, 2018-02-17:
Hi doudz, thank you for the reply!
I cannot test right now, but I will test asap!

Cheers

Emmegiemme, 2018-02-19:
Okay,
above the function reported on the online manual "z.permit_join()", I had to perform the "z.start_network_scan()" in order to get those functions up and running.
Actually, was I call start_network_scan, the "z.devices" function has reported the full description of the bulb (eg. "[Device 2669 TRADFRI bulb E27 opal 1000lm]" instead of the only address "[2669]" that I was getting with the command "z.permit_join()"

Emmegiemme, 2018-02-19:
Apology, but one last point, it is not cleat to me how I can declare that my bulb has coordinator. 
As I understood, if I can do that, I should able to control the bulb either via Zigate and by the Ikea remote, am I wrong?

Thanks!

doudz, 2018-02-20:
network_scan has nothing to do with the device discovery.
After pairing the lib use the following sequence :
- Wait Device announce
- Ask for node description
- Ask for active endpoints
- For each endpoint discovered ask for clusters descriptions
- For each cluster discovered ask for well known attribute

doudz, 2018-02-20:
For now, by default the python lib is the coordinator.
But you're right, technically you could do that but really I don't know how ;)
I suppose you have to set the zigate as a router instead of coordinator.

If you use an automation system (like Jeedom or Home Assistant, ...) the workaround is to pair both bulb and remote to the zigate and use the system to control everything

Emmegiemme, 2018-02-20:
Gotcha!
My final goal would be integrated your library in a python script that running as a service will expose on MQTT the functions per device, in order to do not need a special implementation for a particularly home automation system and make it easier to use with one or another.

doudz, 2018-02-20:
That's exactly what I was thinking about today

doudz, 2018-02-21:
a simple mqtt broker will be available in the next release ;)

doudz, 2018-03-01:
To be able to control your Ikea bulb from zigate and from remote you have to :
- Pair the bulb to zigate with permit_join
- connect ikea remote to bulb using the touchlink button of the remote

closed 2018-03-01 by doudz