doudz/zigate #2: Discovered endpoints list empty

lboue, 2018-03-10:
Hello,

I setup zigate-0.10.7 version on my raspberry pi (kernel 4.9.59-v7) which have the Zigate USB dongle (firmware 3.0d) connected. Here is what my the python script outputs:
```
pi@raspberrypi:~/py-zigate$ python3 py-zigate.py 
Zigate version:
3.0d
is_permitting_join ? :
True

devices:
[Device 7629 , Device 1bd7 , Device 6f44 TRADFRI bulb E14 WS opal 400lm, Device 74d2 , Device b26c ]
7629
1bd7
discovered endpoints:
{}
well known attributes:
pi@raspberrypi:~/py-zigate$ 

```
I can notice the TRADFRI bulb: 'Device 6f44 TRADFRI bulb E14 WS opal 400lm'
But 'discovered endpoints' and 'well known attributes' are empty. 

Do you have any clue to fix that?

Regards,
Ludovic

doudz, 2018-03-10:
You have to call refresh_device() to re discover all informations

lboue, 2018-03-10:
Does it work for all devices?
```
print('Zigate version:')
print(z.get_version_text())

z.get_devices_list()

# refresh devices list
z.refresh_device()
```

I am getting an error with the refresh_device method:
```
$ python3 py-zigate.py
Zigate version:
3.0d
Traceback (most recent call last):
  File "py-zigate.py", line 20, in <module>
    z.refresh_device()
TypeError: refresh_device() missing 1 required positional argument: 'addr'
```

doudz, 2018-03-10:
You have to specify the short address of the device
z.refresh_device('1bd7')

or
d = z.get_device_from_addr('1bd7')
d.refresh_device()

doudz, 2018-03-12:
it works on any device but be aware that battery powered devices need to be awake (by pushing button for example)
note that the device is automatically "refresh" at pairing time

closed 2018-03-12 by doudz

lboue, 2018-03-12:
Hello,

I am sorry but it still doesn't work. I did an EEPROM reset and flash again firmware 3.0d. Pairing is OK but endpoints still empty:
```
cat /home/pi/.zigate.json
[
    {
        "addr": "a83a",
        "endpoints": [],
        "info": {
            "addr": "a83a",
            "id": 0,
            "ieee": "00158d0001d5c468",
            "last_seen": "2018-03-12 17:23:44",
            "power_type": 0,
            "rssi": 162
        }
    },
    {
        "addr": "d292",
        "endpoints": [],
        "info": {
            "addr": "d292",
            "id": 1,
            "ieee": "000b57fffe0d820a",
            "last_seen": "2018-03-12 17:23:44",
            "power_type": 1,
            "rssi": 156
        }
    }
```

Here is the debug log:
```
pi@raspberrypi:~/py-zigate$ ./py-zigate-list.py 
DEBUG:zigate:Searching ZiGate port
DEBUG:zigate:ZiGate found at /dev/ttyUSB0
DEBUG:zigate:Try loading persistent file
DEBUG:zigate:Load success
DEBUG:zigate:REQUEST : 0x0021 b'\x03\x18\xc8\x00'
DEBUG:zigate:Msg to send b'\x00!\x00\x04\xf6\x03\x18\xc8\x00'
DEBUG:zigate:Encoded Msg to send b'\x01\x02\x10!\x02\x10\x02\x14\xf6\x02\x13\x18\xc8\x02\x10\x03'
DEBUG:zigate:Waiting for status message for command 0x0021
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:33, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0021:0
DEBUG:zigate:REQUEST : 0x0023 b'\x00'
DEBUG:zigate:Msg to send b'\x00#\x00\x01"\x00'
DEBUG:zigate:Encoded Msg to send b'\x01\x02\x10#\x02\x10\x02\x11"\x02\x10\x03'
DEBUG:zigate:Waiting for status message for command 0x0023
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:35, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0023:0
DEBUG:zigate:Check network state
DEBUG:zigate:REQUEST : 0x0024 
DEBUG:zigate:Msg to send b'\x00$\x00\x00$'
DEBUG:zigate:Encoded Msg to send b'\x01\x02\x10$\x02\x10\x02\x10$\x03'
DEBUG:zigate:Waiting for status message for command 0x0024
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:36, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0024:0
DEBUG:zigate:REQUEST : 0x0009 
DEBUG:zigate:Msg to send b'\x00\t\x00\x00\t'
DEBUG:zigate:Encoded Msg to send b'\x01\x02\x10\x02\x19\x02\x10\x02\x10\x02\x19\x03'
DEBUG:zigate:Waiting for status message for command 0x0009
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:9, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8009 - Network state response : addr:0000, ieee:00158d0001b22e3a, pan:25054, extend_pan:3847775962517828605, channel:11, rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Do nothing special for response RESPONSE 0x8009 - Network state response : addr:0000, ieee:00158d0001b22e3a, pan:25054, extend_pan:3847775962517828605, channel:11, rssi:0
DEBUG:zigate:STATUS code to command 0x0009:0
DEBUG:zigate:Waiting for message 0x8009
DEBUG:zigate:Stop waiting, got message 0x8009
DEBUG:zigate:REQUEST : 0x0015 
DEBUG:zigate:Msg to send b'\x00\x15\x00\x00\x15'
DEBUG:zigate:Encoded Msg to send b'\x01\x02\x10\x15\x02\x10\x02\x10\x15\x03'
DEBUG:zigate:Waiting for status message for command 0x0015
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:21, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8015 - Device list : devices:[OrderedDict([('id', 0), ('addr', 'a83a'), ('ieee', '00158d0001d5c468'), ('power_type', 0), ('rssi', 162)]), OrderedDict([('id', 1), ('addr', 'd292'), ('ieee', '000b57fffe0d820a'), ('power_type', 1), ('rssi', 154)])], rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Known devices in zigate : {'a83a', 'd292'}
DEBUG:zigate:STATUS code to command 0x0015:0
DEBUG:zigate:Waiting for message 0x8015
DEBUG:zigate:Stop waiting, got message 0x8015
DEBUG:zigate:Check Need refresh Device a83a 
DEBUG:zigate:Need refresh : no type
DEBUG:zigate:Need refresh : no endpoints
DEBUG:zigate:Dispatch ZIGATE_DEVICE_NEED_REFRESH Device a83a 
DEBUG:zigate:Check Need refresh Device d292 
---------------------------------------------------------------
DEBUG:zigate:Need refresh : no type
DEBUG:zigate:Need refresh : no endpoints
DEBUG:zigate:Dispatch ZIGATE_DEVICE_NEED_REFRESH Device d292 
---------------------------------------------------------------
DEBUG:zigate:Auto saving /home/pi/.zigate.json
DEBUG:zigate:Previous devices to delete : set()
DEBUG:zigate:Dispatch ZIGATE_DEVICE_UPDATED
DEBUG:zigate:Dispatch ZIGATE_DEVICE_UPDATED
DEBUG:zigate:REQUEST : 0x0015 
DEBUG:zigate:Msg to send b'\x00\x15\x00\x00\x15'
DEBUG:zigate:Encoded Msg to send b'\x01\x02\x10\x15\x02\x10\x02\x10\x15\x03'
DEBUG:zigate:Waiting for status message for command 0x0015
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:21, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8015 - Device list : devices:[OrderedDict([('id', 0), ('addr', 'a83a'), ('ieee', '00158d0001d5c468'), ('power_type', 0), ('rssi', 162)]), OrderedDict([('id', 1), ('addr', 'd292'), ('ieee', '000b57fffe0d820a'), ('power_type', 1), ('rssi', 154)])], rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Known devices in zigate : {'a83a', 'd292'}
DEBUG:zigate:Previous devices to delete : set()
DEBUG:zigate:Dispatch ZIGATE_DEVICE_UPDATED
DEBUG:zigate:Dispatch ZIGATE_DEVICE_UPDATED
DEBUG:zigate:STATUS code to command 0x0015:0
devices:
[Device a83a , Device d292 ]
a83a
discovered endpoints:
{}
well known attributes:
End of script
```

I don't know what to do

Regards,
Ludovic