doudz/zigate #6: MQTT function

Emmegiemme, 2018-03-18:
Hi Sébastien,
well done! 

I have seen that you have deployed the MQTT function as we discussed! 
As soon as possible (it means as soon as I will stop following "white rabbits" on my home automation project) I will give a shot and run a test with it! ;-)

Cheers!

doudz, 2018-03-18:
Currently I use it with Home Assistant

Emmegiemme, 2018-03-18:
Cool, I will try to use it with mosquitto on OpenHab

doudz, 2018-06-19:
Closing since no update

closed 2018-06-19 by doudz

pdecat, 2018-06-23:
Hi @doudz, since you said you were doing it, I'm trying to integrate my ZiGate (plugged with USB-TTL module) with Home Assistant using your MQTT broker implementation as it certainly is the most elegant way to do it.

However I'm currently struggling figuring out the proper configuration.

Is device discovery supposed to work with [Home Assistant's MQTT discovery](https://www.home-assistant.io/docs/mqtt/discovery/)?

Are the topics for discovery, command and state supposed to match out of the box?
Or is customization needed?

Here what I've tried right now but I doubt this would ever work as Home Assistant expects the discovery topic to follow the `<discovery_prefix>/<component>/[<node_id>/]<object_id>/<>` format:

```
mqtt:
  discovery: true
  discovery_prefix: zigate/device_changed

light:
  - platform: mqtt
    command_topic: "zigate/command"
```

Also, I guess `state_value_template` are needed to extract values from messages as described at https://www.home-assistant.io/components/light.mqtt/

Would you mind sharing an example Home Assistant configuration snippet for your MQTT broker?

pdecat, 2018-06-23:
Hmm, I just stumbled on https://github.com/Koenkk/zigbee2mqtt

doudz, 2018-06-23:
Oh... That's what I was looking for before buying a zigate...

doudz, 2018-06-23:
Currently my mqtt broker doesn't support discovery.
I'll send you a example of what I use on my system soon

doudz, 2018-06-23:
```
sensor:
  - platform: mqtt
    state_topic: "zigate/attribute_changed/21d3/01/0402/0000"
    name: 'Temperature'
    unit_of_measurement: "°C"
    value_template: '{{ value_json.value }}'
```
This is a Xiaomi temperature sensor, addr 21d3, endpoint 01, cluster 0x0402, attribute 0x0000

pdecat, 2018-06-25:
Thanks, that helped!

FWIW, here is the best I have managed to achieve so far with the MQTT Template Light platform to control my IKEA Tradfri Lights (on/off and level control working):

```
light:
  - platform: mqtt_template
    command_topic: "zigate/command"
    command_on_template: >
      {%- if brightness is defined -%}
      {"function": "action_move_level_onoff", "args": ["5c2f", 1, 1, {{ brightness / 255 * 100 }}]}
      {%- else -%}
      {"function": "action_onoff", "args": ["5c2f", 1, 1]}
      {%- endif -%}
    command_off_template: '{"function": "action_onoff", "args": ["5c2f", 1, 0]}'
    optimistic: true
    brightness_template: '' # Not None to allow controlling brightness
    state_topic: "zigate/attribute_changed/5c2f/01/0006/0000"
    state_template: '{{ value_json["value"] }}'
```

This still misses light level state changes and initial device state messages pushed to `zigate/device_changed/5c2f`.

Now, none of the current MQTT light platforms is gathering all required features to allow for a perfect fit:

* `light.mqtt` is missing command templates features to be able to format command messages as JSON
* `light.mqtt_template` is missing a `brightness_state_topic` option to process `zigate/attribute_changed/5c2f/01/0008/0000` messages.
* `light.mqtt_json` is missing command templates and a `brightness_state_topic` option to process `zigate/attribute_changed/5c2f/01/0008/0000` messages.

At this point, I'm considering to adapt your project (what's the license by the way?) to send all device and attribute change events to the same topic and use the `mqtt.light_template` platform some configuration like the following:

```
light:
  - platform: mqtt_template
    command_topic: "zigate/command"
    command_on_template: >
      {%- if brightness is defined -%}
      {"function": "action_move_level_onoff", "args": ["5c2f", 1, 1, {{ brightness / 255 * 100 }}]}
      {%- else -%}
      {"function": "action_onoff", "args": ["5c2f", 1, 1]}
      {%- endif -%}
    command_off_template: '{"function": "action_onoff", "args": ["5c2f", 1, 0]}'
    state_topic: "zigate/device_changed/5c2f"
    state_template: '{{ value_json["endpoints"][0]["clusters"][0]["attributes"]["value"] }}'
    brightness_template: '{{ value_json["endpoints"][0]["clusters"][1]["attributes"]["value"] }}'
```

doudz, 2018-06-26:
Yes I agree, I think there's no other way .
About the licence, I have no choose right now, but the MIT licence is probably the best choice

lolorc, 2018-06-27:
> Oh... That's what I was looking for before buying a zigate...

quite off-topic, but why did you choose a zigate instead ?
I'm owning one and I've just bought stuff to try to replace it by zigbee2mqtt.

qm3ster, 2018-11-13:
@lolorc what was your problem with ZiGate?