doudz/zigate #19: Trying to get the 'current_level'

pipiche38, 2018-07-30:
@cloudz, I'm trying to use your plugin to interact with a USB Zigate. While the connectivity seems to work. Bascially the example in the README works fine.

However, I'm not getting any results for the 2 commandes 
```
# get all discovered endpoints
print(" get all discovered endpoints")
z.devices[0].endpoints

# get well known attributes
print(" get well known attributes")
for attribute in z.devices[0].properties:
     print(attribute)
```

In addition, I'm trying to use the 'current_level' in order to ready the level of my shutter, with the following command, but nothing.

```
# Recupère le status du Volet
print("z.devices[0].get_property('current_level')")
z.devices[0].get_property('current_level')
```
Should I get the result via CallBack ? How ?

doudz, 2018-07-30:
First, be sure that the device is really the one you want, z.devices is a list of all devices
Asking z.devices[0] means the first device of the list
You can use z.get_device_from_addr() to get a particular device

If there no endpoints on z.devices[0].endpoints, maybe you need to refresh information
```z.devices[0].refresh_device()```
Calling ```z.devices[0].get_property('current_level')``` supposed that the value `current_level` as been already fetched by calling read_attribute_request or by device self reporting
Current_level is coming from cluster 8, attribute 0 so you have to call `z.read_attribute_request(ADDR, ENDPOINT, 8, 0)` (you have to specify the addr and the endpoint)

pipiche38, 2018-07-31:
Thanks very much. Very helpful

closed 2018-07-31 by pipiche38