doudz/zigate #22: Associate a switch to a light

ChoOo7, 2018-08-10:
Bonjour

J'essayer d'associer directement un interrupteur à un bandeau de led en utilisant ta librairie.
Inspiré de http://faire-ca-soi-meme.fr/domotique/2018/03/14/une-telecommande-zigbee-pour-les-controler-tous/ 

Mon approche est la suivante : 
J'ai un brokker lancé

J'appelle une premiere fois
$payload = '{"function": "add_group", "args": ["b89c", 11]}';
$mqtt->publish("zigate/command", $payload, 0);

Et dans un listener je recupère l'adresse du groupe.

Je lance ensuite

$groupAddr = "0b71";

echo "\nadd group to light 1";
$payload = '{"function": "add_group", "args": ["b89c", 11, "'.$groupAddr.'"]}';
$mqtt->publish("zigate/command", $payload, 0);

echo "\nadd group to switch 1";
$payload = '{"function": "add_group", "args": ["fd1d", 1, "'.$groupAddr.'"]}';
$mqtt->publish("zigate/command", $payload, 0);

echo "\nview grouips";
$payload = '{"function": "get_group_membership", "args": ["b89c", 11]}';
$mqtt->publish("zigate/command", $payload, 0);

Les 2 premières commandes se passent sans erreur, la dernière échoue : 
ERROR:root:Error calling function get_group_membership

En, en résultat... :/ appuyer sur mon bouton ne change pas le status on/off de mon interrupteur :(

J'ai essayé le bouton d’appareillage  : 
(adresse différente car entre temps je l'ai supprimé puis ajouté de nouveau)

string(42) "zigate/attribute_changed/0cba/01/0000/0005"
string(146) "{"attribute": 5, "addr": "0cba", "cluster": 0, "data": "lumi.sensor_switch.aq2", "name": "type", "value": "lumi.sensor_switch.aq2", "endpoint": 1}"
string(4) "XXXX"
string(42) "zigate/attribute_changed/0cba/01/0000/ff01"
string(177) "{"attribute": 65281, "cluster": 0, "data": "0121030c0328210421a81305214600062402000200000a210000", "value": 3.075, "endpoint": 1, "addr": "0cba", "unit": "V", "name": "battery"}"
string(4) "XXXX"

et suppression / association, rien de mieux.

Mes dispositifs : 
switch : lumi.sensor_switch.aq2
osram : bandeau de led
jiawen : bandeau de led

Sinon, si j'intercepte l'event avec le broker et que je lance une action, cela marche, mais j'aimerais avoir une installation indépendante du raspberry

Merci pour ton aide

doudz, 2018-08-10:
Je n'ai pas testé le fonctionnement des groupes, j'ignore pourquoi la fonction get_group_membership, est-ce que tu utilises bien la dernière version ?

ChoOo7, 2018-08-10:
J’étais sur la version de la semaine dernière je dirais
Je viens de pip3 install zigate --upgrade
et après upgrade, j'ai le même comportement

doudz, 2018-08-10:
ok dommage.
Je ne travaille plus sur zigate en ce moment, je reprendrai dans quelques semaines / mois.

Si tu trouves le problème n'hésite pas à proposer quelque chose

ChoOo7, 2018-08-10:
Déjà aujourd'hui j'ai compris que lancer pleins de process java utilisant la zigate directement n'était pas la bonne approche, et qu'il fallait un seul process (le broker) qui y accédait.
Depuis l'ensemble marche pas mal mieux.

Je ne suis pas fortiche en python, mais pour l'histoire de save je regarderais. Apres je suis encore jeune sur zigate & cie, je ne comprends pas toujours ce que je fais ou comment cela devrait fonctionner :/

Mais à la rentrée en septembre, je ne résisterais surement pas à aller regarder / corriger cela :)

doudz, 2018-11-13:
Est-ce que le problème est toujours d'actualité ?