doudz/zigate #23: Error saving configuration as JSON

ChoOo7, 2018-08-10:
Hello

In my brokker stack trace, I can see

ERROR:zigate:Failed to save persistent file /root/.zigate.json
ERROR:zigate:Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 171, in save_state
    sort_keys=True, indent=4, separators=(',', ': '))
  File "/usr/lib/python3.5/json/__init__.py", line 178, in dump
    for chunk in iterable:
  File "/usr/lib/python3.5/json/encoder.py", line 429, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/usr/lib/python3.5/json/encoder.py", line 403, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.5/json/encoder.py", line 403, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.5/json/encoder.py", line 436, in _iterencode
    o = _default(o)
  File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 1256, in default
    return json.JSONEncoder.default(self, obj)
  File "/usr/lib/python3.5/json/encoder.py", line 179, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: {('fd1d', 1), ('b89c', 11), ('ee65', 3)} is not JSON serializable

Am i doin'g something wrong ?

Thanks

doudz, 2018-08-10:
This is a bug, it tries to save group but groups are almost untested, and it doesn't seem to work.

I'm not working on zigate this days, feel free to submit a patch

ChoOo7, 2018-08-10:
Ok thanks.

We keep in touch

doudz, 2018-10-11:
Fix https://github.com/doudz/zigate/commit/eb5812a9dce807a081f79c20912c67f5b3f9912a

closed 2018-10-11 by doudz