doudz/zigate #26: Support "Xiaomi Aqara vibration sensor"

llaumgui, 2018-09-03:
Cf: http://faire-ca-soi-meme.fr/domotique/2018/09/03/test-xiaomi-aqara-vibration-sensor/#Compatibilite_ZiGate

llaumgui, 2018-09-11:
Je viens de le recevoir... J'ai bien l'objet qui se créé dans ma solution domotique (Jeedom). Par contre j'ai beau le taper fort, j'ai pas de trace qui remonte dans mes logs.

Je regarde ce qui peut y avoir de spécifique dans le code du daemon mais j'ai un peu de mal à comprendre l'ajout d'équipement...

llaumgui, 2018-09-11:
D'autant plus étrange qu'il est dans la liste...
https://zigate.fr/produits-xiaomi-compatibles-zigate/

doudz, 2018-09-12:
J'ai poussé un commit pour supporter le capteur de vibration, à tester

llaumgui, 2018-09-15:
Je viens de tester, dès que je le secou, ça plante:
`Exception in thread Thread-2:
Traceback (most recent call last):
File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
self.run()
File "/usr/lib/python3.5/threading.py", line 862, in run
self._target(*self._args, **self._kwargs)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 140, in _event_loop
self.decode_data(packet)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 349, in decode_data
response = RESPONSES.get(msg_type, Response)(value, rssi)
File "/usr/local/lib/python3.5/dist-packages/zigate/responses.py", line 45, in __init__
self.decode()
File "/usr/local/lib/python3.5/dist-packages/zigate/responses.py", line 291, in decode
data = struct.unpack(fmt, self.data['data'])[0]
struct.error: unpack requires a bytes object of length 4`

doudz, 2018-09-15:
idéalement peux-tu mettre en mode DEBUG et m'envoyer le log ?

llaumgui, 2018-09-17:
On parle du mode débug du plugin ? Il y a pas de mode débug dnas la lib ?

ioull, 2018-09-17:
Voila ce que j'ai 
`
[2018-09-17 19:55:56][DEBUG] : callZiGate permit_join
[2018-09-17 19:55:56][DEBUG][root] : Message received in socket
[2018-09-17 19:55:56][DEBUG][root] : {'args': '', 'action': 'permit_join'}
[2018-09-17 19:55:56][DEBUG][zigate] : REQUEST : 0x0049 FFFC1E00
[2018-09-17 19:55:56][DEBUG][zigate] : Msg to send b'0049000450fffc1e00'
[2018-09-17 19:55:56][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10I\x02\x10\x02\x14P\xff\xfc\x1e\x02\x10\x03'
[2018-09-17 19:55:56][DEBUG][zigate] : Waiting for status message for command 0x0049
[2018-09-17 19:55:56][DEBUG][zigate] : RESPONSE 0x8000 - Status response : status:0, sequence:116, packet_type:73, error:b'', rssi:0
[2018-09-17 19:55:56][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-17 19:55:56][DEBUG][zigate] : STATUS code to command 0x0049:0
[2018-09-17 19:55:56][DEBUG][root] : {'success': True, 'result': 0}
[2018-09-17 19:55:56][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 0 )
[2018-09-17 19:56:01][DEBUG][zigate] : RESPONSE 0x004D - Device announce : addr:9cde, ieee:00158d0002a51c56, mac_capability:10000000, rssi:156
[2018-09-17 19:56:01][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-17 19:56:01][DEBUG][zigate] : Device Announce
[2018-09-17 19:56:01][DEBUG][zigate] : Dispatch ZIGATE_DEVICE_ADDED
[2018-09-17 19:56:01][DEBUG][root] : External command ZIGATE_DEVICE_ADDED : {'device': Device 9cde , 'zigate': <zigate.core.ZiGate object at 0x7fb5ba46b198>}
[2018-09-17 19:56:01][DEBUG][zigate] : REQUEST : 0x0042 9cde
[2018-09-17 19:56:01][DEBUG][zigate] : Msg to send b'00420002029cde'
[2018-09-17 19:56:01][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10B\x02\x10\x02\x12\x02\x12\x9c\xde\x03'
[2018-09-17 19:56:01][DEBUG][zigate] : Waiting for status message for command 0x0042
[2018-09-17 19:56:01][DEBUG][root] : Send to jeedom :  {'device': Device 9cde , 'action': 'ZIGATE_DEVICE_ADDED'}
[2018-09-17 19:56:01][DEBUG][requests.packages.urllib3.connectionpool] : Starting new HTTP connection (1): 127.0.0.1
[2018-09-17 19:56:01][DEBUG] : Array (     [device] => Array         (             [info] => Array                 (                     [addr] => 9cde                     [ieee] => 00158d0002a51c56                     [mac_capability] => 10000000                     [rssi] => 156                 )              [endpoints] => Array                 (                 )              [addr] => 9cde         )      [action] => ZIGATE_DEVICE_ADDED )
[2018-09-17 19:56:01][DEBUG] : createcommands for 9cde
[2018-09-17 19:56:01][DEBUG] : callZiGate available_actions Array (     [0] => 9cde )
[2018-09-17 19:56:01][DEBUG][root] : Message received in socket
[2018-09-17 19:56:01][DEBUG][root] : {'args': ['9cde'], 'action': 'available_actions'}
[2018-09-17 19:56:01][DEBUG][root] : {'success': True, 'result': {}}
[2018-09-17 19:56:01][DEBUG] : result callZiGate Array (     [success] => 1     [result] => Array         (         )  )
[2018-09-17 19:56:01][DEBUG] : create action refresh for endpoint 0
[2018-09-17 19:56:01][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=hHWPN5tmZPbDbHuIl8j6GSgcwR07xZrI HTTP/1.1" 200 16
[2018-09-17 19:56:01][DEBUG][root] : Jeedom reply :  {'success': True}
[2018-09-17 19:56:03][DEBUG] : callZiGate get_version_text
[2018-09-17 19:56:03][DEBUG][root] : Message received in socket
[2018-09-17 19:56:03][DEBUG][root] : {'args': '', 'action': 'get_version_text'}
[2018-09-17 19:56:03][DEBUG][root] : {'success': True, 'result': '3.0d'}
[2018-09-17 19:56:03][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 3.0d )
[2018-09-17 19:56:04][ERROR][zigate] : No response after command 0x0042
[2018-09-17 19:56:04][DEBUG][zigate] : REQUEST : 0x0045 9cde
[2018-09-17 19:56:04][DEBUG][zigate] : Msg to send b'00450002059cde'
[2018-09-17 19:56:04][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10E\x02\x10\x02\x12\x02\x15\x9c\xde\x03'
[2018-09-17 19:56:04][DEBUG][zigate] : Waiting for status message for command 0x0045
[2018-09-17 19:56:07][ERROR][zigate] : No response after command 0x0045
[2018-09-17 19:56:07][DEBUG][zigate] : RESPONSE 0x8000 - Status response : status:0, sequence:117, packet_type:66, error:b'', rssi:0
[2018-09-17 19:56:07][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-17 19:56:07][DEBUG][zigate] : RESPONSE 0x004D - Device announce : addr:9cde, ieee:00158d0002a51c56, mac_capability:10000000, rssi:168
[2018-09-17 19:56:07][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-17 19:56:07][DEBUG][zigate] : Device Announce
[2018-09-17 19:56:07][DEBUG][zigate] : Dispatch ZIGATE_DEVICE_UPDATED
[2018-09-17 19:56:07][DEBUG][root] : External command ZIGATE_DEVICE_UPDATED : {'device': Device 9cde , 'zigate': <zigate.core.ZiGate object at 0x7fb5ba46b198>}
[2018-09-17 19:56:07][DEBUG][root] : Send to jeedom :  {'device': Device 9cde , 'action': 'ZIGATE_DEVICE_UPDATED'}
[2018-09-17 19:56:07][DEBUG][requests.packages.urllib3.connectionpool] : Starting new HTTP connection (1): 127.0.0.1
[2018-09-17 19:56:07][DEBUG] : Array (     [device] => Array         (             [info] => Array                 (                     [addr] => 9cde                     [ieee] => 00158d0002a51c56                     [mac_capability] => 10000000                     [rssi] => 168                 )              [endpoints] => Array                 (                 )              [addr] => 9cde       [... cut ...]
[2018-09-17 19:56:25][DEBUG] : result callZiGate Array (     [success] => 1     [result] => Array         (             [1] => Array                 (                 )              [2] => Array                 (                 )          )  )
[2018-09-17 19:56:25][DEBUG] : create action refresh for endpoint 0
[2018-09-17 19:56:25][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=hHWPN5tmZPbDbHuIl8j6GSgcwR07xZrI HTTP/1.1" 200 16
[2018-09-17 19:56:25][DEBUG][root] : Jeedom reply :  {'success': True}
`

llaumgui, 2018-09-17:
La même version avec les sauts de ligne:
```
[2018-09-17 19:55:56][DEBUG] : callZiGate permit_join 
[2018-09-17 19:55:56][DEBUG][root] : Message received in socket 
[2018-09-17 19:55:56][DEBUG][root] : {'args': '', 'action': 'permit_join'} 
[2018-09-17 19:55:56][DEBUG][zigate] : REQUEST : 0x0049 FFFC1E00 
[2018-09-17 19:55:56][DEBUG][zigate] : Msg to send b'0049000450fffc1e00' 
[2018-09-17 19:55:56][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10I\x02\x10\x02\x14P\xff\xfc\x1e\x02\x10\x03' 
[2018-09-17 19:55:56][DEBUG][zigate] : Waiting for status message for command 0x0049 
[2018-09-17 19:55:56][DEBUG][zigate] : RESPONSE 0x8000 - Status response : status:0, sequence:116, packet_type:73, error:b'', rssi:0 
[2018-09-17 19:55:56][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED 
[2018-09-17 19:55:56][DEBUG][zigate] : STATUS code to command 0x0049:0 
[2018-09-17 19:55:56][DEBUG][root] : {'success': True, 'result': 0} 
[2018-09-17 19:55:56][DEBUG] : result callZiGate Array ( [success] => 1 [result] => 0 ) 
[2018-09-17 19:56:01][DEBUG][zigate] : RESPONSE 0x004D - Device announce : addr:9cde, ieee:00158d0002a51c56, mac_capability:10000000, rssi:156 
[2018-09-17 19:56:01][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED 
[2018-09-17 19:56:01][DEBUG][zigate] : Device Announce 
[2018-09-17 19:56:01][DEBUG][zigate] : Dispatch ZIGATE_DEVICE_ADDED 
[2018-09-17 19:56:01][DEBUG][root] : External command ZIGATE_DEVICE_ADDED : {'device': Device 9cde , 'zigate': <zigate.core.ZiGate object at 0x7fb5ba46b198>} 
[2018-09-17 19:56:01][DEBUG][zigate] : REQUEST : 0x0042 9cde 
[2018-09-17 19:56:01][DEBUG][zigate] : Msg to send b'00420002029cde' 
[2018-09-17 19:56:01][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10B\x02\x10\x02\x12\x02\x12\x9c\xde\x03' 
[2018-09-17 19:56:01][DEBUG][zigate] : Waiting for status message for command 0x0042 
[2018-09-17 19:56:01][DEBUG][root] : Send to jeedom : {'device': Device 9cde , 'action': 'ZIGATE_DEVICE_ADDED'} 
[2018-09-17 19:56:01][DEBUG][requests.packages.urllib3.connectionpool] : Starting new HTTP connection (1): 127.0.0.1 
[2018-09-17 19:56:01][DEBUG] : Array ( [device] => Array ( [info] => Array ( [addr] => 9cde [ieee] => 00158d0002a51c56 [mac_capability] => 10000000 [rssi] => 156 ) [endpoints] => Array ( ) [addr] => 9cde ) [action] => ZIGATE_DEVICE_ADDED ) 
[2018-09-17 19:56:01][DEBUG] : createcommands for 9cde 
[2018-09-17 19:56:01][DEBUG] : callZiGate available_actions Array ( [0] => 9cde ) 
[2018-09-17 19:56:01][DEBUG][root] : Message received in socket 
[2018-09-17 19:56:01][DEBUG][root] : {'args': ['9cde'], 'action': 'available_actions'} 
[2018-09-17 19:56:01][DEBUG][root] : {'success': True, 'result': {}} 
[2018-09-17 19:56:01][DEBUG] : result callZiGate Array ( [success] => 1 [result] => Array ( ) ) 
[2018-09-17 19:56:01][DEBUG] : create action refresh for endpoint 0 
[2018-09-17 19:56:01][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=hHWPN5tmZPbDbHuIl8j6GSgcwR07xZrI HTTP/1.1" 200 16 
[2018-09-17 19:56:01][DEBUG][root] : Jeedom reply : {'success': True} 
[2018-09-17 19:56:03][DEBUG] : callZiGate get_version_text 
[2018-09-17 19:56:03][DEBUG][root] : Message received in socket 
[2018-09-17 19:56:03][DEBUG][root] : {'args': '', 'action': 'get_version_text'} 
[2018-09-17 19:56:03][DEBUG][root] : {'success': True, 'result': '3.0d'} 
[2018-09-17 19:56:03][DEBUG] : result callZiGate Array ( [success] => 1 [result] => 3.0d ) 
[2018-09-17 19:56:04][ERROR][zigate] : No response after command 0x0042 
[2018-09-17 19:56:04][DEBUG][zigate] : REQUEST : 0x0045 9cde 
[2018-09-17 19:56:04][DEBUG][zigate] : Msg to send b'00450002059cde' 
[2018-09-17 19:56:04][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10E\x02\x10\x02\x12\x02\x15\x9c\xde\x03' 
[2018-09-17 19:56:04][DEBUG][zigate] : Waiting for status message for command 0x0045 
[2018-09-17 19:56:07][ERROR][zigate] : No response after command 0x0045 
[2018-09-17 19:56:07][DEBUG][zigate] : RESPONSE 0x8000 - Status response : status:0, sequence:117, packet_type:66, error:b'', rssi:0 
[2018-09-17 19:56:07][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED 
[2018-09-17 19:56:07][DEBUG][zigate] : RESPONSE 0x004D - Device announce : addr:9cde, ieee:00158d0002a51c56, mac_capability:10000000, rssi:168 
[2018-09-17 19:56:07][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED 
[2018-09-17 19:56:07][DEBUG][zigate] : Device Announce 
[2018-09-17 19:56:07][DEBUG][zigate] : Dispatch ZIGATE_DEVICE_UPDATED 
[2018-09-17 19:56:07][DEBUG][root] : External command ZIGATE_DEVICE_UPDATED : {'device': Device 9cde , 'zigate': <zigate.core.ZiGate object at 0x7fb5ba46b198>} 
[2018-09-17 19:56:07][DEBUG][root] : Send to jeedom : {'device': Device 9cde , 'action': 'ZIGATE_DEVICE_UPDATED'} 
[2018-09-17 19:56:07][DEBUG][requests.packages.urllib3.connectionpool] : Starting new HTTP connection (1): 127.0.0.1 
[2018-09-17 19:56:07][DEBUG] : Array ( [device] => Array ( [info] => Array ( [addr] => 9cde [ieee] => 00158d0002a51c56 [mac_capability] => 10000000 [rssi] => 168 ) [endpoints] => Array ( ) [addr] => 9cde ) [action] => ZIGATE_DEVICE_UPDATED ) 
[2018-09-17 19:56:07][DEBUG] : createcommands for 9cde 
[2018-09-17 19:56:07][DEBUG] : callZiGate available_actions Array ( [0] => 9cde ) 
[2018-09-17 19:56:07][DEBUG][root] : Message received in socket 
[2018-09-17 19:56:07][DEBUG][root] : {'args': ['9cde'], 'action': 'available_actions'} 
[... cut ...]
[2018-09-17 19:56:25][DEBUG] : result callZiGate Array ( [success] => 1 [result] => Array ( [1] => Array ( ) [2] => Array ( ) ) ) 
[2018-09-17 19:56:25][DEBUG] : create action refresh for endpoint 0 
[2018-09-17 19:56:25][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=hHWPN5tmZPbDbHuIl8j6GSgcwR07xZrI HTTP/1.1" 200 16 
[2018-09-17 19:56:25][DEBUG][root] : Jeedom reply : {'success': True}
```

Pour moi tu n'as  pas d'erreur comme moi...

doudz, 2018-09-17:
Oui en regardant comme ça je vois pas de problème

llaumgui, 2018-09-17:
Alors après tests :

- J'ai plus de crash Python.
- J'ai un objet qui c'est créé mais qui est pas complet AMHA:

[image]

- J'arrive plus à le suprimer, il revient à chaque restart.
- Comme il est déjà là, pas moyen de le réinclude.

Mais j'ai beau tapper ça envoi rien.

llaumgui, 2018-09-17:
En virant le patch, l’objet se créé bien... Mais rien ne passe...

llaumgui, 2018-09-17:
@ioull > tu as posté tes logs... Mais tu as un problème particulier ?
@doudz > On est OK que pour tester j'applique le patch, je supprime le __pycache__ et je lance le daemon pour regénérer le __pycache__ ?

ioull, 2018-09-17:
Non non c'était pour montrer les logs du capteur vibration

doudz, 2018-09-18:
@llaumgui met ton plugin en mode DEBUG, utilise l'objet afin de produire le crash et post le log ici

llaumgui, 2018-09-18:
Le problème c'est qu'il ne plante plus... Mais aucune info.

llaumgui, 2018-09-18:
Je testerais un reset ce soir. Pour sortir l'objet puis un réappairage...

llaumgui, 2018-09-21:
Ayé, j'ai reproduit !!!!!
```
[2018-09-21 21:11:12][DEBUG] : callZiGate get_version_text 
[2018-09-21 19:11:12][DEBUG][root] : Message received in socket
[2018-09-21 19:11:12][DEBUG][root] : {'action': 'get_version_text', 'args': ''}
[2018-09-21 19:11:12][DEBUG][root] : {'success': True, 'result': '3.0d'}
[2018-09-21 21:11:12][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 3.0d ) 
[2018-09-21 21:11:20][DEBUG] : callZiGate remove_device Array (     [0] => 63c7 ) 
[2018-09-21 19:11:20][DEBUG][root] : Message received in socket
[2018-09-21 19:11:20][DEBUG][root] : {'action': 'remove_device', 'args': ['63c7']}
[2018-09-21 19:11:20][DEBUG][zigate] : REQUEST : 0x0026 b'\x00\x15\x8d\x00\x02\xa4\x92l\x00\x15\x8d\x00\x02\xa4\x92l'
[2018-09-21 19:11:20][DEBUG][zigate] : Msg to send b'002600103600158d0002a4926c00158d0002a4926c'
[2018-09-21 19:11:20][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10&\x02\x10\x106\x02\x10\x15\x8d\x02\x10\x02\x12\xa4\x92l\x02\x10\x15\x8d\x02\x10\x02\x12\xa4\x92l\x03'
[2018-09-21 19:11:20][DEBUG][zigate] : Waiting for status message for command 0x0026
[2018-09-21 19:11:20][DEBUG][zigate] : RESPONSE 0x8000 - Status response : status:169, sequence:0, packet_type:38, error:b'', rssi:0
[2018-09-21 19:11:20][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-21 19:11:20][ERROR][zigate] : Command 0x0026 failed Failed (ZigBee event codes) 169 : b''
[2018-09-21 19:11:20][DEBUG][zigate] : STATUS code to command 0x0026:169
[2018-09-21 19:11:20][DEBUG][root] : {'success': True, 'result': 169}
[2018-09-21 21:11:20][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 169 ) 
[2018-09-21 21:11:20][DEBUG] : callZiGate get_version_text 
[2018-09-21 19:11:20][DEBUG][root] : Message received in socket
[2018-09-21 19:11:20][DEBUG][root] : {'action': 'get_version_text', 'args': ''}
[2018-09-21 19:11:20][DEBUG][root] : {'success': True, 'result': '3.0d'}
[2018-09-21 21:11:20][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 3.0d ) 
[2018-09-21 21:11:26][DEBUG] : callZiGate permit_join 
[2018-09-21 19:11:26][DEBUG][root] : Message received in socket
[2018-09-21 19:11:26][DEBUG][root] : {'action': 'permit_join', 'args': ''}
[2018-09-21 19:11:26][DEBUG][zigate] : REQUEST : 0x0049 FFFC1E00
[2018-09-21 19:11:26][DEBUG][zigate] : Msg to send b'0049000450fffc1e00'
[2018-09-21 19:11:26][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10I\x02\x10\x02\x14P\xff\xfc\x1e\x02\x10\x03'
[2018-09-21 19:11:26][DEBUG][zigate] : Waiting for status message for command 0x0049
[2018-09-21 19:11:27][DEBUG][zigate] : RESPONSE 0x8000 - Status response : status:0, sequence:228, packet_type:73, error:b'', rssi:0
[2018-09-21 19:11:27][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-21 19:11:27][DEBUG][zigate] : STATUS code to command 0x0049:0
[2018-09-21 19:11:27][DEBUG][root] : {'success': True, 'result': 0}
[2018-09-21 21:11:27][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 0 ) 
[2018-09-21 19:11:35][DEBUG][zigate] : RESPONSE 0x004D - Device announce : addr:1190, ieee:00158d0002a4926c, mac_capability:10000000, rssi:84
[2018-09-21 19:11:35][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-09-21 19:11:35][DEBUG][zigate] : Device Announce
[2018-09-21 19:11:35][WARNING][zigate] : Device already exists with another addr 63c7, removing it.
[2018-09-21 19:11:35][DEBUG][zigate] : Dispatch ZIGATE_DEVICE_REMOVED
[2018-09-21 19:11:35][DEBUG][zigate] : Dispatch ZIGATE_DEVICE_ADDED
[2018-09-21 19:11:35][DEBUG][root] : External command ZIGATE_DEVICE_ADDED : {'zigate': <zigate.core.ZiGate object at 0xffffaec2f2e8>, 'device': Device 1190 }
[2018-09-21 19:11:35][DEBUG][zigate] : REQUEST : 0x0042 1190
[2018-09-21 19:11:35][DEBUG][zigate] : Msg to send b'00420002c11190'
[2018-09-21 19:11:35][DEBUG][zigate] : Encoded Msg to send b'\x01\x02\x10B\x02\x10\x02\x12\xc1\x11\x90\x03'
[2018-09-21 19:11:35][DEBUG][zigate] : Waiting for status message for command 0x0042
[2018-09-21 19:11:35][DEBUG][root] : Send to jeedom :  {'action': 'ZIGATE_DEVICE_ADDED', 'device': Device 1190 }
[2018-09-21 19:11:35][DEBUG][requests.packages.urllib3.connectionpool] : Starting new HTTP connection (1): 127.0.0.1
[2018-09-21 21:11:35][DEBUG] : Array (     [action] => ZIGATE_DEVICE_ADDED     [device] => Array         (             [addr] => 1190             [endpoints] => Array                 (                 )              [info] => Array                 (                     [addr] => 1190                     [ieee] => 00158d0002a4926c                     [mac_capability] => 10000000                     [rssi] => 84                 )          )  ) 
[2018-09-21 21:11:36][DEBUG] : createcommands for 1190
[2018-09-21 21:11:36][DEBUG] : callZiGate available_actions Array (     [0] => 1190 ) 
[2018-09-21 19:11:36][DEBUG][root] : Message received in socket
[2018-09-21 19:11:36][DEBUG][root] : {'action': 'available_actions', 'args': ['1190']}
[2018-09-21 19:11:36][DEBUG][root] : {'success': True, 'result': {}}
[2018-09-21 21:11:36][DEBUG] : result callZiGate Array (     [success] => 1     [result] => Array         (         )  ) 
[2018-09-21 21:11:36][DEBUG] : create action refresh for endpoint 0
[2018-09-21 19:11:36][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=bXcO5Pj3l35FFPj3rN4kWSnEL94iU32g HTTP/1.1" 200 16
[2018-09-21 19:11:36][DEBUG][root] : Jeedom reply :  {'success': True}
[... cut ...]
[2018-09-21 19:11:46][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=bXcO5Pj3l35FFPj3rN4kWSnEL94iU32g HTTP/1.1" 200 16
[2018-09-21 19:11:46][DEBUG][root] : Jeedom reply :  {'success': True}
[2018-09-21 21:11:56][DEBUG] : callZiGate get_version_text 
[2018-09-21 19:11:56][DEBUG][root] : Message received in socket
[2018-09-21 19:11:56][DEBUG][root] : {'action': 'get_version_text', 'args': ''}
[2018-09-21 19:11:56][DEBUG][root] : {'success': True, 'result': '3.0d'}
[2018-09-21 21:11:56][DEBUG] : result callZiGate Array (     [success] => 1     [result] => 3.0d ) 
```

llaumgui, 2018-09-28:
Du coup @doudz ça t'aide ?

doudz, 2018-09-28:
A vrai dire non, car c'est le décodage de la réponse qui plante mais il manque une ligne de DEBUG pour comprendre ce qu'il se passe.
Je viens de pousser une toute petite modif pour ajouter une ligne de debug

llaumgui, 2018-09-28:
OK. En // tu as vu ma PR ?

doudz, 2018-09-28:
oui j'ai vu. C'est intéressant, je verrai ça la semaine prochaine

llaumgui, 2018-10-10:
Voila:
```
[2018-10-10 19:00:23][DEBUG][zigate] : Received response 0x8102: b'\xa4{\x9a\x02\x00\x12\x00U\x00!\x00\x02\x00\x02'
[2018-10-10 19:00:23][DEBUG][zigate] : RESPONSE 0x8102 - Individual Attribute Report : sequence:164, addr:7b9a, endpoint:2, cluster:18, attribute:85, status:0, data_type:33, size:2, data:2, rssi:156
[2018-10-10 19:00:23][DEBUG][zigate] : Dispatch ZIGATE_RESPONSE_RECEIVED
[2018-10-10 19:00:23][DEBUG][zigate] : Set expire timer for 2-18-85 in 2
[2018-10-10 19:00:23][DEBUG][zigate] : Dispatch ZIGATE_ATTRIBUTE_UPDATED
[2018-10-10 19:00:23][DEBUG][root] : External command ZIGATE_ATTRIBUTE_UPDATED : {'zigate': <zigate.core.ZiGate object at 0xffffbda8f080>, 'device': Device 7b9a lumi.sensor_cube, 'attribute': {'endpoint': 2, 'addr': '7b9a', 'expire': 2, 'value': 'wakeup', 'name': 'movement', 'attribute': 85, 'expire_value': '', 'data': 2, 'cluster': 18}}
[2018-10-10 19:00:24][DEBUG][root] : Send to jeedom :  {'attribute': {'endpoint': 2, 'addr': '7b9a', 'expire': 2, 'value': 'wakeup', 'name': 'movement', 'attribute': 85, 'expire_value': '', 'data': 2, 'cluster': 18}, 'action': 'ZIGATE_ATTRIBUTE_UPDATED', 'device': Device 7b9a lumi.sensor_cube}
[2018-10-10 19:00:24][DEBUG][requests.packages.urllib3.connectionpool] : Starting new HTTP connection (1): 127.0.0.1
[2018-10-10 21:00:24][DEBUG] : Array (     [attribute] => Array         (             [endpoint] => 2             [addr] => 7b9a             [expire] => 2             [value] => wakeup             [name] => movement             [attribute] => 85             [expire_value] =>              [data] => 2             [cluster] => 18         )      [action] => ZIGATE_ATTRIBUTE_UPDATED     [device] => Array         (             [endpoints] => Array                 (                     [0] => Array                         (                             [in_clusters] => Array                                 (                                 )                              [endpoint] => 1                             [out_clusters] => Array                                 (                                 )                              [clusters] => Array                                 (                                     [0] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [name] => application_version                                                             [value] => 3                                                             [attribute] => 1                                                             [data] => 3                                                         )                                                      [1] => Array                                                         (                                                             [name] => type                                                             [value] => lumi.sensor_cube                                                             [attribute] => 5                                                             [data] => lumi.sensor_cube                                                         )                                                      [2] => Array                                                         (                                                             [value] => 2.975                                                             [unit] => V                                                             [name] => battery                                                             [attribute] => 65281                                                             [data] => 01219f0b0328180421a8130521c300062407000000000a210000972100009821ac03992101039a210300                                                         )                                                  )                                              [cluster] => 0                                         )                                  )                              [profile] => 0                             [device] => 0                         )                      [1] => Array                         (                             [in_clusters] => Array                                 (                                 )                              [endpoint] => 2                             [out_clusters] => Array                                 (                                 )                              [clusters] => Array                                 (                                     [0] => Array                                         (                                             [attributes] => Array                                                 (                                                     [0] => Array                                                         (                                                             [name] => movement                                                             [expire] => 2                                                             [value] => wakeup                                                             [attribute] => 85                                                             [expire_value] =>                 [... cut ...] ffer] => 127                     [descriptor_capability] => 00000000                     [bit_field] => 0100000000000010                 )          )  )
[2018-10-10 21:00:27][DEBUG] : create command 1.257.85
[2018-10-10 19:00:27][DEBUG][requests.packages.urllib3.connectionpool] : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=bXcO5Pj3l35FFPj3rN4kWSnEL94iU32g HTTP/1.1" 200 16
[2018-10-10 19:00:27][DEBUG][root] : Jeedom reply :  {'success': True}
```

Désolé pour le temps.

doudz, 2018-10-11:
Merci.
J'ai fait une correction, ça devrait fonctionner maintenant.
La trame qui ne passait pas est le "mode libération"
https://faire-ca-soi-meme.fr/domotique/2018/09/03/test-xiaomi-aqara-vibration-sensor/

> Mode libération
> Type: 0x8102 (Attribute Report)
> SQN: 0x0C
> Src Addr: 0xCDD1
> Src Ep: 0x01
> Cluster ID: 0x0101 (General: Door Lock)
> Attribute ID: 0x0508
> Attribute Size: 0x0008
> Attribute Type: 0x25 (Unknown)
> Status: 0x00
> 
> Cette trame permet d’éviter l’inondation d’actions. Tant que cette trame n’est pas lancée, on ne peut pas lancer d’autres actions.

C'est une trame vide donc ça ne changera rien au fonctionnement, mais il n'y aura plus de crash

closed 2018-10-11 by doudz

llaumgui, 2018-10-11:
OK, je te ferai mon retours ce WE.