doudz/zigate #29: Support de la télécommande ikea Tradfri

ChoOo7, 2018-10-12:
Bonjour

J'ai bien lu ton article
https://faire-ca-soi-meme.fr/test/2017/06/06/ikea-tradfri-zigbee-3-0-compatibilite-zigate/

Ce que je cherche à faire, ce n'est pas d'associer ma télécommande à une lampe, mais à associer la télécommande à ma zigate, pour ensuite binder les evenements sur le broker pour lancer diverses actions.

Je n'arrive pas à associer la télécommande à la zigate / à recevoir des evenements

Lorsque je tiens le bouton appuyé 10 seconde,s rien coté zigate

Et si je fias un factory reset, j'arrive à avoir des choses
string(26) "zigate/device_changed/38b8"
string(128) "{"endpoints": [], "addr": "38b8", "info": {"addr": "38b8", "ieee": "d0cf5efffe05aa5d", "mac_capability": "10000000", "rssi": 3}}"
string(4) "XXXX"
string(26) "zigate/device_changed/38b8"
string(128) "{"endpoints": [], "addr": "38b8", "info": {"addr": "38b8", "ieee": "d0cf5efffe05aa5d", "mac_capability": "10000000", "rssi": 3}}"

Mais rien ensuite concernant les clics sur les boutons

Est-ce quelque chose de possible ?

Merci

doudz, 2018-10-12:
ATTENTION je ne suis pas Akila le créateur de la zigate.
Pour info, actuellement la zigate ne permet pas de récupérer les actions de la télécommande, peut-être dans une prochaine mise à jour du firmware.
Pour le moment, la télécommande est visible mais c'est tout.

closed 2018-10-12 by doudz

ChoOo7, 2018-10-12:
En effet pardon ! :) merci pour ton travail

ChoOo7, 2018-10-12:
Yep j'avais capté ce point

Mais si je veux utiliser la zigate pour intercepter les messages de ma télécommande pour ensuite lancer des CURL ou autre, je suis b####
Cela marche bien avec un bouton aqara, mais pas dans mon cas :/ Dommage