doudz/zigate #32: Appel des constante

llaumgui, 2018-10-27:
Salut,

avec la 0.20.0, pour appeler les constantes c'est plus zigate.ZIGATE_FAILED_TO_CONNECT mais zigate.const.ZIGATE_FAILED_TO_CONNECT

La solution serait de faire comme pour la version et faire des trucs du genre ZIGATE_FAILED_TO_CONNECT = const.ZIGATE_FAILED_TO_CONNECT

tu en penses quoi ? Le but étant de ne pas importer de choses non utilisé (flake8)

llaumgui, 2018-10-27:
Pour illustrer: https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/commit/9ae4c91a6e1c588080320949947117150a381b91#diff-d5e3656be8dbcce9765f37cf1bdc1e20

Faut valider ça avant la release 0.20.0. Car le lumi.vibration marche !

doudz, 2018-10-27:
Je vais proposer qqc dans la semaine.
Par contre attention dans ton exemple, zigate et zigatewifi ne sont pas des constantes

llaumgui, 2018-10-27:
Mais du coup ça sera quoi ? zigate. ou zigate.const ?

llaumgui, 2018-10-27:
Merci pour la remarque, j'avais pas mis la sensibilté à la case ;-).

doudz, 2018-10-29:
C'est corrigé, on ne change rien, ça reste zigate.ZIGATE_FAILED_TO_CONNECT

closed 2018-10-29 by doudz

llaumgui, 2018-10-29:
Effectivement on peut mettre un noqa 😀