doudz/zigate #35: unable to start ZIgate MQTT brooker

soors, 2018-10-30:
Hello
nice work with Zigate python library.
Although the zigate recognises my smoke detectors running from python3, i do not succeed in starting the mqtt broker. I always get the following "connection refused error ": 

$ sudo python3 -m zigate.mqtt_broker --device auto --mqtt_host localhost:1883
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/zigate/zigate/mqtt_broker.py", line 111, in <module>
    broker.start()
  File "/home/pi/zigate/zigate/mqtt_broker.py", line 38, in start
    self.connect()
  File "/home/pi/zigate/zigate/mqtt_broker.py", line 35, in connect
    self.client.connect(host, port)
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 839, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 962, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.5/socket.py", line 712, in create_connection
    raise err
  File "/usr/lib/python3.5/socket.py", line 703, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

Any help on this?

Thx

doudz, 2018-10-30:
Could you check your mqtt is running ? (on localhost and port 1883)

soors, 2018-10-30:
No indeed mqtt on host was not running.
I misunderstood the zigate.mqtt_broker python application : I thought a new Paho Mqtt broker would be started also.
By pointing too a running mqtt broker hosted on another IP adress, the zigbee data is now visible in Node Red. 
Although i still get the following error :

python3 -m zigate.mqtt_broker --device auto --mqtt_host 192.168.10.200:1883 --mqtt_username xxxxxx --mqtt_password xxxxxx
ERROR:zigate:Failed to eval "struct.unpack('H', unhexlify(value)[2:4])[0]/1000." using "0121c10c0328180421a81305210b00062401000000000a2100000821041064200096230000000000"
ERROR:zigate:Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/zigate/clusters.py", line 91, in update
    'self': self})
  File "<string>", line 1, in <module>
NameError: name 'struct' is not defined

INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB0
INFO:root:Publish zigate/device_changed/6de8
INFO:root:Publish zigate/device_changed/18a9
INFO:root:MQTT connected with result code 0

Thanks and keep up the good work.

closed 2018-10-30 by soors

doudz, 2018-10-31:
Fixed in 0.20.2