doudz/zigate #36: Delete device question/problem

pawelka, 2018-10-30:
In code remove device command is:
0026+ieee+ieee

commented one is:
0026+addr+ieee

On zigate site https://zigate.fr/wiki/commandes-zigate/ is commented version. 

Why you used ieee+ieee? I'm asking, because I don't know if it's by purpose or bug.

doudz, 2018-11-01:
Well, the fact is I don't manage to make it works, that's why I tried both command

https://github.com/fairecasoimeme/ZiGate/issues/29
https://github.com/fairecasoimeme/ZiGate/issues/55

doudz, 2018-11-15:
Should be fixed in 0.21.1

closed 2018-11-15 by doudz

biau, 2018-11-15:
this bug could explain this in hass?
2018-11-15 18:11:21 DEBUG (Thread-7) [zigate] Previous devices missing : {'207e', '07ca', 'dd8c', 'b746', '0097', 'ae25', '8065', '09c5', '06bd', '8a15', 'a3c3', '06f9', 'b40b', '36cd'}

Did you test 0.21.1 with hass ?

ruimarinho, 2018-11-15:
@biau which plugin are you using for hass integration?

doudz, 2018-11-15:
I suppose he use https://github.com/doudz/homeassistant-zigate

ruimarinho, 2018-11-15:
Are you restarting development of the plugin? That would be fantastic! The community needs it :)

doudz, 2018-11-15:
The github repo is not archived anymore ;)

biau, 2018-11-15:
Yes that's devices I had to rejoin to the zigate and these addresses does not exist anymore.
Is there any command I can send to remove them? 

Great news for all zigate users and thanks for your work.

doudz, 2018-11-16:
To remove this sévices, call cleanup