doudz/zigate #37: Obtenir la version

llaumgui, 2018-10-31:
Histoire d'améliorer le debug ainsi que le reporting, je souhaiterais récupérer la version de la lib à partir du daemon... Actuellement on peut récupérer la version du firmware mais c'est tout.

Te serait-il possible de passer la version de ta librairie via une commande ?

C'est pour ce bug: https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues/75

doudz, 2018-11-01:
Mauvais endroit pour cet issue à mon avis....
Pour avoir la version : 
```
import zigate
zigate.__version__
```

closed 2018-11-01 by doudz

llaumgui, 2018-11-01:
Oui mais tu peux pas le faire avec un callZigate. Le but était de récupérer l'info à partir du daemon... Sinon je dois faire un shell exec en php pour le récupérer.

doudz, 2018-11-01:
Pas du tout, ya beaucoup plus simple, il suffit d'ajouter une fonction dans le démon