doudz/zigate #38: Read Write attributes with manufacturer specific command

soors, 2018-11-02:
Hello

is it possible to extend the read_attribute_request and write_attribute_request command with the manufacturer specific code when sending the command over MQTT? Or is there a workaround?

It seems in order to read the status of a Xiamo detector,  the manufacturer specific code 0x115F has to be added. And this is the same, to set its sensitivity.

If I am not mistaken, the Zigate command 0x0100 (read) and 0x0110 (write), could deal with it.

Soors

doudz, 2018-11-04:
Sure I could add it in the next release

doudz, 2018-11-07:
Fixed in https://github.com/doudz/zigate/commit/c29d042faeadfd51433f0ef8de87046072e7aaa4
You could now add manufacturer_specific

closed 2018-11-07 by doudz