doudz/zigate #40: Cannot parse network join response

pawelka, 2018-11-15:
Zigate firmware 3.0e
Zigate library 0.21.1 

2018-11-15 23:44:27 DEBUG (SyncWorker_3) [zigate] Check network state
2018-11-15 23:44:27 DEBUG (SyncWorker_3) [zigate] REQUEST : 0x0024
2018-11-15 23:44:27 DEBUG (SyncWorker_3) [zigate] Msg to send b'0024000024'
2018-11-15 23:44:27 DEBUG (SyncWorker_3) [zigate] Encoded Msg to send b'\x01\x02\x10$\x02\x10\x02\x10$\x03'
2018-11-15 23:44:27 DEBUG (SyncWorker_3) [zigate] Waiting for status message for command 0x0024
2018-11-15 23:44:27 DEBUG (Thread-3) [zigate] Received response 0x8024: b'\x04'
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 155, in _event_loop
    self.decode_data(packet)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 373, in decode_data
    response = RESPONSES.get(msg_type, Response)(value, rssi)
  File "/config/deps/lib/python3.6/site-packages/zigate/responses.py", line 46, in __init__
    self.decode()
  File "/config/deps/lib/python3.6/site-packages/zigate/responses.py", line 107, in decode
    sdata, msg_data = self.__decode(fmt, keys, msg_data)
  File "/config/deps/lib/python3.6/site-packages/zigate/responses.py", line 118, in __decode
    sdata = OrderedDict(zip(keys, struct.unpack(fmt, data[:size])))
struct.error: unpack requires a buffer of 12 bytes

pawelka, 2018-11-15:
If I understand zigate commands correctly this is because command was send too fast. TestGUI works because of manual delays.

"4 = Busy (Node is carrying out a Lengthy operation and is Currently Unable to handle the incoming command)" - it's a description for 8000, but there is a chance that applies for network join too.

doudz, 2018-11-16:
Yes you're that's the same status code
I have to handle this case
is it happening everytimes ?

pawelka, 2018-11-16:
Everytime :-/

I added (after learing python basics) 3s delay in your lib before start network and this change didn’t help :-/

doudz, 2018-11-16:
Did you add time.sleep(3) just before start_network() at line 248 ?
```
        self.set_type(TYPE_COORDINATOR)
        LOGGER.debug('Check network state')
        time.sleep(3)
        self.start_network()
        network_state = self.get_network_state()
```

pawelka, 2018-11-16:
Yes, this was my change.

doudz, 2018-11-16:
Ok I'll handle it
I think it's related to https://github.com/fairecasoimeme/ZiGate/issues/15

I think there's no problem, the busy status probably just indicate that the network is already started.
In previous firmware, this command just not answer.

doudz, 2018-11-16:
Should be fixed in 0.21.2

closed 2018-11-16 by doudz

pawelka, 2018-11-16:
Works, thank you.