doudz/zigate #42: ZIGATE_DEVICE_REMOVED signal with ieee - improvement

pawelka, 2018-11-15:
device is removed from list before signal, so there is no way to get device from Zigate. 

Options:
- send ieee additionally
- send whole device

doudz, 2018-11-16:
Why would you need a removed device?

pawelka, 2018-11-16:
I’m trying to improve your integration with HASS and  I switched from addr to ieee which are not changing over time (for example kids playing with switch and unpair /repair it). Entity id using ieee, so such playing doesn't broke hass configuration.

I know that I can handle mapping on plug-in site, but from other signals I can retrieve ieee.

doudz, 2018-11-16:
ok I understand. Using ieee is a good thing
BUT
- addr changing doesn't really break things because of customize.yaml, you can have device zigate.adbc with real addr 1234, since the current integration use ieee as unique_id, HASS continue to works
- Additionnally, release 0.20.0 brings a new signal ZIGATE_DEVICE_RENAMED which inform a addr change. It's useful when a device is paired again but has not sent leave signal to zigate (when out of range for example).

doudz, 2018-11-16:
ZIGATE_DEVICE_REMOVED now provide whole device in 0.22.0

closed 2018-11-16 by doudz

pdecat, 2018-11-17:
Hi @pawelka, I was hitting the same issue with my fork and noticed [that's what you used as](https://github.com/pawelka/homeassistant-zigate/commit/191f17d790669c1881d60ad0dbcc7fe3c65fdec7) your base in https://github.com/pawelka/homeassistant-zigate/commits/master

Perhaps we should consider contributing those changes back to https://github.com/doudz/homeassistant-zigate/

@doudz would such intrusive changes be accepted?

doudz, 2018-11-17:
Sure, any contribution is welcome.
The final goal is official integration

pawelka, 2018-11-17:
Sure, I'll contribute. I'll clean up my fork code in upcoming weeks.