doudz/zigate #52: Issues with TRADFRI bulb E27 WS clear 950lm

ruimarinho, 2018-12-02:
I was able to pair this bulb correctly with endpoint discovery looking fine as well, but the integration is not perfect yet.

1. I am able of turning it off immediately and the light bulb turns off as expected but a couple of seconds later, the switch is turned on again automatically. The light remains off in an inconsistent state (see next issue).

```
2018-12-01 00:32:54 DEBUG (SyncWorker_6) [zigate] REQUEST : 0x0092 b'\x02\xd3\xd0\x01\x01\x00'
2018-12-01 00:32:54 DEBUG (SyncWorker_6) [zigate] Msg to send b'009200069502d3d0010100'
2018-12-01 00:32:54 DEBUG (SyncWorker_6) [zigate] Encoded Msg to send b'0102109202100216950212d3d002110211021003'
2018-12-01 00:32:54 DEBUG (SyncWorker_6) [zigate] Waiting for status message for command 0x0092
2018-12-01 00:32:54 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-01 00:32:54 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00640092'
2018-12-01 00:32:54 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:100, packet_type:146, error:b'', rssi:0
2018-12-01 00:32:54 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-01 00:32:54 DEBUG (SyncWorker_6) [zigate] STATUS code to command 0x0092:0
```

2. After turning the light off, it turns on immediately when changing its brightness. Right now this is the only way of turning it on:

```
2018-12-01 00:33:48 DEBUG (SyncWorker_18) [zigate] REQUEST : 0x0081 b'\x02\xd3\xd0\x01\x01\x01\xd2\x00\x00'
2018-12-01 00:33:48 DEBUG (SyncWorker_18) [zigate] Msg to send b'008100095a02d3d0010101d20000'
2018-12-01 00:33:48 DEBUG (SyncWorker_18) [zigate] Encoded Msg to send b'01021081021002195a0212d3d0021102110211d20210021003'
2018-12-01 00:33:48 DEBUG (SyncWorker_18) [zigate] Waiting for status message for command 0x0081
2018-12-01 00:33:48 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-01 00:33:48 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'006e0081'
2018-12-01 00:33:48 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:110, packet_type:129, error:b'', rssi:0
2018-12-01 00:33:48 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-01 00:33:48 DEBUG (SyncWorker_18) [zigate] STATUS code to command 0x0081:0
```

3. Changing the color temperature appears to record the right state but does not trigger a change on the light bulb:

```
2018-12-01 00:34:22 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0092 b'\x02\xd3\xd0\x01\x01\x01'
2018-12-01 00:34:22 DEBUG (SyncWorker_1) [zigate] Msg to send b'009200069402d3d0010101'
2018-12-01 00:34:22 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'0102109202100216940212d3d002110211021103'
2018-12-01 00:34:22 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x0092
2018-12-01 00:34:22 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-01 00:34:22 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00760092'
2018-12-01 00:34:22 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:118, packet_type:146, error:b'', rssi:0
2018-12-01 00:34:22 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-01 00:34:22 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x0092:0
```

If it helps, this is what the zigate entity looks like in home assistant:

```nginx
battery_voltage: null
battery_level: 100
rssi_percent: 15
type: TRADFRI bulb E27 WS clear 950lm
manufacturer: 117c
receiver_on_when_idle: true
missing: false
addr: d3d0
ieee: d0cf5efffe22c8d9
mac_capability: 10001110
rssi: 39
max_rx: 82
max_tx: 82
server_mask: 0
descriptor_capability: 00000000
max_buffer: 82
bit_field: 0100000000000001
last_seen: 2018-12-01 00:28:47
friendly_name: Device d0cf5efffe22c8d9 (d3d0) TRADFRI bulb E27 WS clear 950lm
icon: mdi:access-point
```

And the light entity:

```nginx
min_mireds: 153
max_mireds: 500
brightness: 33
color_temp: 2702
addr: d3d0
ieee: d0cf5efffe22c8d9
endpoint: 1
friendly_name: Device d0cf5efffe22c8d9 (d3d0) TRADFRI bulb E27 WS clear 950lm 1
supported_features: 35
```

Brightness does not appear to change on the entity state, but it does indeed change on the bulb.

biau, 2018-12-02:
I had issues with tradfri bulbs the first time. Did you tried to refresh it from hass?
zigate.refresh_device

ruimarinho, 2018-12-02:
Interesting, I did not. Will report back when I get to it again.

doudz, 2018-12-03:
Actually, the binding process to allow self status report is now perfect, so yes calling `zigate.refresh_device` will help

ruimarinho, 2018-12-03:
I have just tested it and the behavior remains the same after calling `zigate.refresh_device` and also after restarting Home Assistant.

doudz, 2018-12-04:
Are you using the last 0.23.1 version of the lib ?

ruimarinho, 2018-12-04:
I was using 0.22.0. Upgrading now and will call `zigate.refresh_device` after that.

doudz, 2018-12-10:
Could you try to call `zigate.raw_command` with the following ?
```json
{"cmd":"0x0030",
"data":"0015c9d0cf5efffe22c8d90100060300158d0000de19c501"}
```
and then
```json
{"cmd":"0x0120",
"data":"00172002d3d00101000600000000010010000000000000000000"}
```
and see what happen in log ?

ruimarinho, 2018-12-10:
First command:

```
2018-12-10 20:39:51 DEBUG (SyncWorker_12) [zigate] REQUEST : 0x0030 0015c9d0cf5efffe22c8d90100060300158d0000de19c501
2018-12-10 20:39:51 DEBUG (SyncWorker_12) [zigate] Msg to send b'00300018180015c9d0cf5efffe22c8d90100060300158d0000de19c501'
2018-12-10 20:39:51 DEBUG (SyncWorker_12) [zigate] Encoded Msg to send b'0102103002101818021015c9d0cf5efffe22c8d902110210021602130210158d02100210de19c5021103'
2018-12-10 20:39:51 DEBUG (SyncWorker_12) [zigate] Waiting for status message for command 0x0030
2018-12-10 20:39:51 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-10 20:39:51 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'80000030'
2018-12-10 20:39:51 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:128, sequence:0, packet_type:48, error:b'', rssi:0
2018-12-10 20:39:51 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-10 20:39:51 ERROR (ZiGate-Event Loop) [zigate] Command 0x0030 failed Failed (ZigBee event codes) 128 : b''
2018-12-10 20:39:51 DEBUG (SyncWorker_12) [zigate] STATUS code to command 0x0030:128
```

Second command:

```
2018-12-10 20:40:19 DEBUG (SyncWorker_19) [zigate] REQUEST : 0x0120 02d3d00101000600000000010010000000000000000000
2018-12-10 20:40:19 DEBUG (SyncWorker_19) [zigate] Msg to send b'012000172002d3d00101000600000000010010000000000000000000'
2018-12-10 20:40:19 DEBUG (SyncWorker_19) [zigate] Encoded Msg to send b'01021120021017200212d3d002110211021002160210021002100210021102101002100210021002100210021002100210021003'
2018-12-10 20:40:19 DEBUG (SyncWorker_19) [zigate] Waiting for status message for command 0x0120
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00b60120'
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:182, packet_type:288, error:b'', rssi:0
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8702: b'd4010102d3d00000000000003f'
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:212, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:15262699137158610944, sequence:63, rssi:8
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-10 20:40:19 DEBUG (SyncWorker_19) [zigate] STATUS code to command 0x0120:0
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8701: b'0000'
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8701 - Route Discovery Confirmation : status:0, network_status:0, rssi:0
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8120: b'b6d3d001000600'
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8120 - Configure Reporting response : sequence:182, addr:d3d0, endpoint:1, cluster:6, status:0, rssi:102
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8102: b'02d3d001000600000010000101'
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:2, addr:d3d0, endpoint:1, cluster:6, attribute:0, status:0, data_type:16, size:1, data:True, rssi:102
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2018-12-10 20:40:19 DEBUG (ZiGate-Event Loop) [custom_components.zigate] Update attribute for device Device d0cf5efffe22c8d9 (d3d0) TRADFRI bulb E27 WS clear 950lm {'endpoint': 1, 'cluster': 6, 'addr': 'd3d0', 'attribute': 0, 'data': True, 'name': 'onoff', 'value': True}
```

doudz, 2018-12-11:
The last command succed, so now I suppose on/off status is reported, is it ?

doudz, 2018-12-11:
For level reporting you could try:

```
{"cmd":"0x0120",
"data":"02d3d00101000600000000010010000000000000000000"}
```

ruimarinho, 2018-12-27:
Whenever I turn it off, it always goes back to on:

```
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] REQUEST : 0x0092 b'\x02\xd3\xd0\x01\x01\x00'
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] Msg to send b'009200069502d3d0010100'
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] Encoded Msg to send b'0102109202100216950212d3d002110211021003'
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] Waiting for status message for command 0x0092
2018-12-27 02:16:27 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xfc\x02\x10\xeb\x02\x10\x92\x02\x10\x03'
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00eb0092'
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:235, packet_type:146, error:b'', rssi:0
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] STATUS code to command 0x0092:0
```

The only possible way of turning it off is by setting brightness to 0:

```
2018-12-27 02:18:24 DEBUG (SyncWorker_13) [zigate] REQUEST : 0x0081 b'\x02\xd3\xd0\x01\x01\x01\x00\x00\x00'
2018-12-27 02:18:24 DEBUG (SyncWorker_13) [zigate] Msg to send b'008100098802d3d0010101000000'
2018-12-27 02:18:24 DEBUG (SyncWorker_13) [zigate] Encoded Msg to send b'0102108102100219880212d3d002110211021102100210021003'
2018-12-27 02:18:24 DEBUG (SyncWorker_13) [zigate] Waiting for status message for command 0x0081
2018-12-27 02:18:24 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xea\x02\x10\xee\x02\x10\x81\x02\x10\x03'
2018-12-27 02:18:24 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-27 02:18:24 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00ee0081'
2018-12-27 02:18:24 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:238, packet_type:129, error:b'', rssi:0
2018-12-27 02:18:24 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-27 02:18:24 DEBUG (SyncWorker_13) [zigate] STATUS code to command 0x0081:0
```

doudz, 2018-12-27:
```
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] REQUEST : 0x0092 b'\x02\xd3\xd0\x01\x01\x00'
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] Msg to send b'009200069502d3d0010100'
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] Encoded Msg to send b'0102109202100216950212d3d002110211021003'
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] Waiting for status message for command 0x0092
2018-12-27 02:16:27 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xfc\x02\x10\xeb\x02\x10\x92\x02\x10\x03'
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00eb0092'
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:235, packet_type:146, error:b'', rssi:0
2018-12-27 02:16:27 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-27 02:16:27 DEBUG (SyncWorker_18) [zigate] STATUS code to command 0x0092:0
```
This log is when you turn off, and it seems to work, but after that the bulb goes on without anything in log ?

doudz, 2018-12-27:
Looks like I have to buy a IKEA bulb since I have no problem with my OSRAM bulb

doudz, 2019-01-01:
The current implementation generate too much commands, I know.
But it will change soon, since I start to implement templates and the next firmware 3.0f will provide better support for attribute discovery

doudz, 2019-01-01:
I managed to have the same bug with my OSRAM bulb, I'll fix it next week

doudz, 2019-01-07:
it's now fixed in dev

ruimarinho, 2019-01-08:
I'll give it a test and report back the result! Thank you!

ruimarinho, 2019-01-16:
@doudz I am still unable to control the temperate color. I tried calling discover_device. Is there an optimal path for removing the existing entity and re-add a fresh one?

doudz, 2019-01-28:
I didn't saw you still have the problem, is it ok now with release 0.26.4 ?

doudz, 2019-02-05:
closing since no news

closed 2019-02-05 by doudz