doudz/zigate #53: Le refresh d'un dispositif entraine un lag

ChoOo7, 2018-12-05:
Bonjour
J'utilise la lib python, avec le broker

Lorsque j'envoie une commande de refresh device au broker, je dois attendre  de nombreuses secondes que la mise à jour se fasse.
Si pendant ce temps là j'utilise mon bouton aqara par exemple, je ne vois pas mon événement dans listener, ou plutôt je ne vois bien plus tard

J'ai l'impression que lorsque l'on demande un refresh, la zigate se met en "pause" tant qu'elle n'a pas obtenu les informations qu'elle est sencée recevoir, et traite les autres messages ensuite

Exemple de commande envoyée : 
zigate/command
{"function": "refresh_device", "args": ["7d56"]}

Je demande l'état d'une ampoule Ikea. Je souhaite avoir des attributs à jour
(mon but final est de savoir son état allumé / etteint et son niveau de luminosité).

S'il y a un autre moyen de connaitre l'état de mon ampoule, je suis preneur :)

Je ne suis pas certain de devoir déclarer un bug / un comportement genant sur ta lib ou sur la zigate elle même.

Merci :)

doudz, 2018-12-06:
Oui effectivement ça provoque un LAG important car, pour le moment, la fonction refresh "inonde" la zigate de commande
Je réfléchie à une meilleure option mais je n'ai pas trop le temps de travailler dessus pour le moment.

ChoOo7, 2018-12-06:
Ok merci
Si en effet on peut avoir quelque chose de + smart pour raffraichir juste ce que l'on souhaite, ça pourrait être pas mal
Merci

doudz, 2018-12-06:
Aujourd'hui la fonction refresh est plus qu'un rafraichissement puisqu'elle redécouvre entièrement l'équipement.
Si l'idée est simplement de forcer la lecture des paramètres connus alors oui on peut faire plus simple.

doudz, 2018-12-06:
Si tu passes par MQTT, tu peux utiliser directement la fonction read_attribute_request en précisant les paramètres addr, endpoint, cluster, attribute

ChoOo7, 2018-12-06:
Super, je vais tester cela, merci.

    def read_attribute_request(self, addr, endpoint, cluster, attribute,
                               direction=0, manufacturer_specific=0, manufacturer_id=0):
        '''
        Read Attribute request
        attribute can be a unique int or a list of int
        '''

Sachant que mon device est
"{"endpoints": [{"clusters": [{"cluster": 0, "attributes": [{"name": "zcl_version", "attribute": 0, "data": 1, "value": 1}, {"name": "application_version", "attribute": 1, "data": 17, "value": 17}, {"name": "stack_version", "attribute": 2, "data": 87, "value": 87}, {"name": "hardware_version", "attribute": 3, "data": 1, "value": 1}, {"name": "manufacturer", "attribute": 4, "data": "IKEA of Sweden", "value": "IKEA of Sweden"}, {"name": "type", "attribute": 5, "data": "TRADFRI bulb E27 WS opal 980lm", "value": "TRADFRI bulb E27 WS opal 980lm"}, {"name": "power_source", "attribute": 7, "data": 0, "value": 0}]}, {"cluster": 8, "attributes": [{"name": "current_level", "attribute": 0, "data": 2, "value": 0}]}, {"cluster": 6, "attributes": [{"name": "onoff", "attribute": 0, "data": true, "value": true}]}, {"cluster": 768, "attributes": [{"name": "remaining_time", "attribute": 2, "data": 0, "value": 0}, {"name": "current_x", "attribute": 3, "data": 30138, "value": 0.459869384765625}, {"name": "current_y", "attribute": 4, "data": 26909, "value": 0.4105987548828125}, {"name": "colour_temperature", "attribute": 7, "data": 333, "value": 3003}, {"name": "colour_mode", "attribute": 8, "data": 2, "value": 2}, {"name": "capabilities", "attribute": 16394, "data": "", "value": ""}]}], "endpoint": 1, "device": 544, "out_clusters": [5, 25, 32, 4096], "profile": 49246, "in_clusters": [0, 3, 4, 5, 6, 8, 768, 2821, 4096]}], "addr": "7d56", "info": {"bit_field": "0100000000000001", "mac_capability": "10001110", "max_buffer": 82, "addr": "7d56", "last_seen": "2018-12-05 20:54:27", "ieee": "90fd9ffffe8b41a6", "manufacturer": "117c", "rssi": 153, "power_type": 1, "descriptor_capability": "00000000", "max_rx": 82, "server_mask": 0, "id": 50, "max_tx": 82}}"

Si ce qui m’intéresse est
{"cluster": 8, "attributes": [{"name": "current_level", "attribute": 0, "data": 2, "value": 0}]}
et
{"cluster": 768, "attributes": [ {"name": "colour_temperature", "attribute": 7, "data": 333, "value": 3003}

Je dois faire
read_attribute_request("7d56", 1, 8, 0)
et
read_attribute_request("7d56", 1, 768, 7)

J'ai tout bon ?
Merci

doudz, 2018-12-06:
Oui c'est ça

ChoOo7, 2018-12-06:
Et..... ca marche au top !  merci
T'as pas mis un petit Donate que l'on puisse te payer un café ?