doudz/zigate #54: struct.error: required argument is not an integer

pawelka, 2018-12-16:
I'm not sure what happened. I touch nothing maybe it's related with xiaomi disappearing devices, but required argument can be handled 

IEEE is missing for 7f72, please pair it again !
Exception dispatching signal ZIGATE_RESPONSE_RECEIVED
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 111, in dispatch_signal
    dispatcher.send(signal, sender, *arguments, **named)
  File "/usr/local/lib/python3.7/site-packages/pydispatch/dispatcher.py", line 338, in send
    **named
  File "/usr/local/lib/python3.7/site-packages/pydispatch/robustapply.py", line 55, in robustApply
    return receiver(*arguments, **named)
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 430, in interpret_response
    d._bind_report(endpoint)
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 1567, in _bind_report
    self._zigate.bind(self.ieee, endpoint_id, 0x0006)
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 816, in bind
    dst_addr, dst_endpoint)
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 806, in _bind_unbind
    cluster, dst_addr_mode, dst_addr, dst_endpoint)
struct.error: required argument is not an integer

doudz, 2018-12-16:
Could you provide log entries just before ?

pawelka, 2018-12-16:
More logs:

2018-12-16 11:04:30 WARNING (ZiGate-Event Loop) [zigate] No response after command 0x0100
2018-12-16 11:04:30 DEBUG (ZiGate-Event Loop) [zigate] REQUEST : 0x0100 b'\x02\xe7\x8d\x01\x01\x00\x00\x00\x00\x00\x00\x02\x00\x10\xff\x01'
2018-12-16 11:04:30 DEBUG (ZiGate-Event Loop) [zigate] Msg to send b'010000109502e78d0101000000000000020010ff01'
2018-12-16 11:04:30 DEBUG (ZiGate-Event Loop) [zigate] Encoded Msg to send b'0102110210021010950212e78d021102110210021002100210021002100212021010ff021103'
2018-12-16 11:04:30 DEBUG (ZiGate-Event Loop) [zigate] Waiting for status message for command 0x0100
2018-12-16 11:04:33 WARNING (ZiGate-Event Loop) [zigate] No response after command 0x0100
2018-12-16 11:04:33 DEBUG (ZiGate-Event Loop) [zigate] REQUEST : 0x0100 b'\x02\xe7\x8d\x01\x01\x00\x01\x00\x00\x00\x00\x06\x00\x00\x00 \x00!\x000\x001\x003'
2018-12-16 11:04:33 DEBUG (ZiGate-Event Loop) [zigate] Msg to send b'010000184502e78d010100010000000006000000200021003000310033'
2018-12-16 11:04:33 DEBUG (ZiGate-Event Loop) [zigate] Encoded Msg to send b'0102110210021018450212e78d0211021102100211021002100210021002160210021002102002102102103002103102103303'
2018-12-16 11:04:33 DEBUG (ZiGate-Event Loop) [zigate] Waiting for status message for command 0x0100
2018-12-16 11:04:36 WARNING (ZiGate-Event Loop) [zigate] No response after command 0x0100
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00f70043'
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:247, packet_type:67, error:b'', rssi:0
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8043: b'f700e78d1002010401000204001000060004000500'
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8043 - Simple descriptor : sequence:247, status:0, addr:e78d, length:16, endpoint:2, profile:260, device:256, bit_field:00000010, inout_clusters:b'\x04\x00\x10\x00\x06\x00\x04\x00\x05\x00', rssi:63, in_clusters:(16, 6, 4, 5), out_clusters:()
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] bind and report for cluster 0x0006
2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] IEEE is missing for e78d, please pair it again !
2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] Exception dispatching signal ZIGATE_RESPONSE_RECEIVED
2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] Traceback (most recent call last):
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 111, in dispatch_signal
    dispatcher.send(signal, sender, *arguments, **named)
  File "/usr/local/lib/python3.6/site-packages/pydispatch/dispatcher.py", line 338, in send
    **named
  File "/usr/local/lib/python3.6/site-packages/pydispatch/robustapply.py", line 55, in robustApply
    return receiver(*arguments, **named)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 430, in interpret_response
    d._bind_report(endpoint)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1567, in _bind_report
    self._zigate.bind(self.ieee, endpoint_id, 0x0006)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 816, in bind
    dst_addr, dst_endpoint)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 806, in _bind_unbind
    cluster, dst_addr_mode, dst_addr, dst_endpoint)
struct.error: required argument is not an integer

2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00f80043'
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:248, packet_type:67, error:b'', rssi:0
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8043: b'f800e78d1003010401000204001000060004000500'
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8043 - Simple descriptor : sequence:248, status:0, addr:e78d, length:16, endpoint:3, profile:260, device:256, bit_field:00000010, inout_clusters:b'\x04\x00\x10\x00\x06\x00\x04\x00\x05\x00', rssi:63, in_clusters:(16, 6, 4, 5), out_clusters:()
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-16 11:04:36 DEBUG (ZiGate-Event Loop) [zigate] bind and report for cluster 0x0006
2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] IEEE is missing for e78d, please pair it again !
2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] Exception dispatching signal ZIGATE_RESPONSE_RECEIVED
2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] Traceback (most recent call last):
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 111, in dispatch_signal
    dispatcher.send(signal, sender, *arguments, **named)
  File "/usr/local/lib/python3.6/site-packages/pydispatch/dispatcher.py", line 338, in send
    **named
[... cut ...]
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 430, in interpret_response
    d._bind_report(endpoint)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1567, in _bind_report
    self._zigate.bind(self.ieee, endpoint_id, 0x0006)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 816, in bind
    dst_addr, dst_endpoint)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 806, in _bind_unbind
    cluster, dst_addr_mode, dst_addr, dst_endpoint)
struct.error: required argument is not an integer

doudz, 2018-12-17:
This will be handled in next release but the real problem is `2018-12-16 11:04:36 ERROR (ZiGate-Event Loop) [zigate] IEEE is missing for e78d, please pair it again !`

closed 2018-12-17 by doudz