doudz/zigate #61: Is light grouping possible?

aurelijusrozenas, 2018-12-29:
Hi,

I have multiple ikea bulbs and I want to control them as one. I saw in some thread @doudz explain that groups are not supported/untested. Does this mean one of:
1. it's not possible with current zigate implementation to do that
2. it's possible and I should use default HA light grouping as in [official docs](https://www.home-assistant.io/components/light.group/) (was not able to setup, but maybe I just need different configuration)
3. it's possible but I should use some zigate specific configuration

doudz, 2018-12-29:
Grouping is possible and supported, but it depends on what you want to do.
If You want to control a group of light from HA only, you can use the case 2.

But if you want to control a group of light from a Ikea remote for exemple, you have to put each light in the same zigbee group. You can achieve it using Ikea remote or using some zigate commands .

aurelijusrozenas, 2018-12-29:
Thanks for quick response! I was not successful so far but I will try again, it's good to know it's possible :)

closed 2018-12-29 by aurelijusrozenas