doudz/zigate #63: Legrand Celiane

dcolens, 2018-12-31:
First, thanks for publishing the code of this project, it seems to be one of the best one out there!

I'm trying to pair a Legrand store switch (https://www.legrand.fr/pro/catalogue/42566-version-celiane-with-netatmo/interrupteur-filaire-connecte-pour-volet-roulant-celiane-with-netatmo-blanc) and I'm getting the following error when trying:
```python
>>> z.permit_join()
DEBUG:zigate:REQUEST : 0x0049 FFFC1E00
DEBUG:zigate:Msg to send b'0049000450fffc1e00'
DEBUG:zigate:Encoded Msg to send b'010210490210021450fffc1e021003'
DEBUG:zigate:Waiting for status message for command 0x0049
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15L\x80\x02\x10\x02\x10I\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'80000049'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:128, sequence:0, packet_type:73, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
ERROR:zigate:Command 0x0049 failed Failed (ZigBee event codes) 128 : b''
DEBUG:zigate:STATUS code to command 0x0049:128
128
```
- zigate firmware: `3.0e`
- host: `rpi3`
- python version: `3.5`

Do you have suggestions on what could go wrong ?

dcolens, 2018-12-31:
After a reboot of the rpi3, the output is different:
```python
>>> z.permit_join()
DEBUG:zigate:REQUEST : 0x0049 FFFC1E00
DEBUG:zigate:Msg to send b'0049000450fffc1e00'
DEBUG:zigate:Encoded Msg to send b'010210490210021450fffc1e021003'
DEBUG:zigate:Waiting for status message for command 0x0049
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15y\x02\x10\xb5\x02\x10I\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00b50049'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:181, packet_type:73, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0049:0
0
```
that looks better, however it still does not pair the device, is there some documentation on the expected flow I could read ?

dcolens, 2019-01-02:
Thanks for the reply, yes, I read that article, I don't have the Legrand gateway I tried pairig it directly with the zigate using a permit_join. Unfortunately nothing happens after the permit join, the led on the switch remains red and no debugs are shown on the zigate side.

doudz, 2019-01-02:
According to the article, you have to "reset" the plug to enable join, have you done it ?

> Petite astuce, (j’ai un peu galéré à trouver. ce n’est pas dans la notice) : le bouton de « reset » ou « RAZ » se trouve sous l’anneau plastique. Au départ je pensais que le bouton se trouvait à la place de la LED. Donc ne forcez pas avec un trombone dans ce trou, il y a la LED au fond 🙂

doudz, 2019-01-02:
Looks like there's limitation, if you still have problem, I suggest you to open an issue on https://github.com/fairecasoimeme/ZiGate

> La compatibilité ZiGate
> Bon ok… certains produits de la gamme Legrand Netatmo semblent compatibles avec la ZiGate… Mais jusqu’à quel point ?
> 
> Si vous appairez directement le micromodule à la ZiGate, vous n’aurez aucun problème. Il suffit de mettre la passerelle en mode « Permit Join » et de mettre sous tension le micromodule pour le voir apparaître.
> 
> Cependant si votre module a déjà été associé à une passerelle, il semble falloir faire un « factory reset » ou un « Erase PDM » pour pouvoir l’associer. Je travaille à comprendre pourquoi mais ce phénomène apparait en fonction des différentes versions.
> 
> Legrand semble avoir mis une sécurité supplémentaire afin de limiter l’accès à d’autres passerelles. Par exemple, sur les dernières versions mises à jour par Legrand, je m’aperçois qu’au moment de l’appairage, les modules demandent à la passerelle un attribut particulier. La passerelle Legrand renvoie alors un nombre aléatoire… Puis il communique ses propriétés.
> 
> Cela laisse entrevoir le côté « je suis ouvert mais pas trop quand même ».

dcolens, 2019-01-17:
FYI: I managed to pair the shutter switch from Legrand by resetting it (multiple times), now I get: 
```python
>>> z.devices
[Device Shutter switch with neutral (082a) 0004740000815460]
>>> z.devices[0].endpoints
{1: {'clusters': {0: Cluster 0x0000 General: Basic}, 'profile': 260, 'in_clusters': (0, 3, 258, 4, 5, 15, 64513), 'device': 514, 'out_clusters': (0, 64513, 25)}}
```

I guess shutters are not supported by the lib yet ? What would it take to add the actions for the shutter ?

pipiche38, 2019-01-17:
> thread [2].
> Links:

Might be managed via Group as Cluster 0x0004 is in

doudz, 2019-01-17:
@pipiche38 not exactly

@dcolens you're right shutters are not yet supported, it's managed by the cluster 0x0102 (listed as 258 in in_clusters)
but not sure it can be handle by zigate yet, I'll check

doudz, 2019-01-17:
I have added a basic support  to allow (I hope) reporting status. But right now it's not possible to control it.

ISO-B, 2019-01-19:
@dcolens Could you report what attributes you can get from shutters?

doudz, 2019-02-20:
any news about this issue ?

dcolens, 2019-02-20:
Ooh, did not realize you were pending on me ? How should I dump attributes ?

doudz, 2019-02-20:
Now (I suppose) you can control the device. But do you want/need some attributes reporting ?
Since the last release 0.28.0 do you get any attribute update ?

or if everything is fine for you, I'll close the issue :smile:

dcolens, 2019-02-20:
Yes, I can control, it fine. I Ran some tests, and the following attributes seem to be available (cluster 258): 
0x0001 - PhysicalClosedLimit – Lift
0x0007 - Config/Status
0x0008 - Current Position Lift Percentage
0x000A - Not sure what this is ?
0x0017 - Not sure what this is ?

with the latest zigate package, I can see:
```python
[{'attribute': 5,
  'data': ' Shutter switch with neutral\x00\x00\x00',
  'name': 'type',
  'type': str,
  'value': 'Shutter switch with neutral'},
 {'attribute': 8,
  'data': 100,
  'name': 'current_position_lift_percentage',
  'value': 100}]
```
which I assume maps to `0x0008 - Current Position Lift Percentage` above.

doudz, 2019-02-20:
So is it enough ? or do you think there's something missing ?

dcolens, 2019-02-21:
Good for me! Thanks a lot for taking the time to help 👍

closed 2019-02-21 by dcolens