doudz/zigate #67: zigate.flasher  - IndexError: index out of range

lboue, 2019-01-08:
Hello 

I am running Python 3.5.3 on my raspberrypi. I am trying to Erase EEPROM but it fails. 
```
pi@raspberrypi:~ $ python3 -V
Python 3.5.3

pi@raspberrypi:~ $ sudo pip3 install -U zigate
Collecting zigate
  Downloading https://files.pythonhosted.org/packages/7d/88/f34edb6e2615990c45656aa4bda404845998f47935c09e2adbaed655fbf3/zigate-0.25.0-py3-none-any.whl (48kB)
    100% |████████████████████████████████| 51kB 447kB/s 
Requirement already up-to-date: pydispatcher in /usr/local/lib/python3.5/dist-packages (from zigate)
Collecting pyserial (from zigate)
  Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 656kB/s 
Requirement already up-to-date: bottle in /usr/local/lib/python3.5/dist-packages (from zigate)
Installing collected packages: pyserial, zigate
  Found existing installation: pyserial 3.2.1
    Not uninstalling pyserial at /usr/lib/python3/dist-packages, outside environment /usr
  Found existing installation: zigate 0.25.0.dev0
    Uninstalling zigate-0.25.0.dev0:
      Successfully uninstalled zigate-0.25.0.dev0
Successfully installed pyserial-3.4 zigate-0.25.0
```

```
pi@raspberrypi:~ $ python3 -m zigate.flasher -h
usage: flasher.py [-h] -p {/dev/ttyAMA0,/dev/ttyUSB0} [-w WRITE] [-s SAVE]
                  [-e] [--pdm-only]

optional arguments:
  -h, --help            show this help message and exit
  -p {/dev/ttyAMA0,/dev/ttyUSB0}, --serialport {/dev/ttyAMA0,/dev/ttyUSB0}
                        Serial port, e.g. /dev/ttyUSB0
  -w WRITE, --write WRITE
                        Firmware bin to flash onto the chip
  -s SAVE, --save SAVE  File to save the currently loaded firmware to
  -e, --erase           Erase EEPROM
  --pdm-only            Erase PDM only, use it with --erase

I got Python IndexError: index out of range error:
pi@raspberrypi:~ $ python3 -m zigate.flasher -p /dev/ttyUSB0 -e
Found MAC-address: 00:15:8d:00:01:b2:2e:3a
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/dist-packages/zigate/flasher.py", line 366, in <module>
    main()
  File "/usr/local/lib/python3.5/dist-packages/zigate/flasher.py", line 362, in main
    erase_EEPROM(ser, args.pdm_only)
  File "/usr/local/lib/python3.5/dist-packages/zigate/flasher.py", line 323, in erase_EEPROM
    res = read_response(ser)
  File "/usr/local/lib/python3.5/dist-packages/zigate/flasher.py", line 108, in read_response
    return _unpack_raw_message(length, answer)
  File "/usr/local/lib/python3.5/dist-packages/zigate/flasher.py", line 119, in _unpack_raw_message
    return _responses.get(type_, Response)(type_, data, chksum)
  File "/usr/local/lib/python3.5/dist-packages/zigate/flasher.py", line 69, in __init__
    self.status = data[0]
IndexError: index out of range
```

Could you help me to fix that?

Regards,

doudz, 2019-01-09:
is the error raising after 5 seconds ?

lboue, 2019-01-09:
Bonjour,

Je passe en français ce sera plus simple :    

- Je ne ne suis pas sûr que l'erreur apparaisse après 5 sec, je vais vérifier ce soir
- Oui je peux passer sur la branche dev et faire un test, pas de souci

Cordialement,

lboue, 2019-01-09:
J'ai remplacé les `logger.debug` par des `print `et voilà ce que cela donne: 

```
pi@raspberrypi:/tmp/zigate-dev $ python3 -m zigate.flasher -p /dev/ttyUSB0 -e
Command 7 None False
Command 9 None True
Command 11 <LH False
Command 31 <LH False
Command 37 None False
Command 39 !B False
Command 44 <BL False
Command 50 None False
Command 54 B False
read_response length 3
read_response answer b'(\x00+'
unpack raw message 3 b'(\x00+'
Response 40 b'\x00' 43
read_response length 7
read_response answer b'3\x00\x10@\x86\x86d'
unpack raw message 7 b'3\x00\x10@\x86\x86d'
Response 51 b'\x00\x10@\x86\x86' 100
read_response length 5
read_response answer b'&\x00\xcc\xee\x01'
unpack raw message 5 b'&\x00\xcc\xee\x01'
Response 38 b'\x00\xcc\xee' 1
read_response length 11
read_response answer b' \x00\xff\xff\xff\xff\xff\xff\xff\xff+'
unpack raw message 11 b' \x00\xff\xff\xff\xff\xff\xff\xff\xff+'
Response 32 b'\x00\xff\xff\xff\xff\xff\xff\xff\xff' 43
read_response length 11
read_response answer b' \x00\x00\x15\x8d\x00\x01\xb2.:\x14'
unpack raw message 11 b' \x00\x00\x15\x8d\x00\x01\xb2.:\x14'
Response 32 b'\x00\x00\x15\x8d\x00\x01\xb2.:' 20
Found MAC-address: 00:15:8d:00:01:b2:2e:3a
read_response length 3
read_response answer b'-\x00.'
unpack raw message 3 b'-\x00.'
Response 45 b'\x00' 46
read_response length 2
read_response answer b'\xb6\xb4'
unpack raw message 2 b'\xb6\xb4'
Response 182 b'' 180
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/tmp/zigate-dev/zigate/flasher.py", line 375, in <module>
    main()
  File "/tmp/zigate-dev/zigate/flasher.py", line 371, in main
    erase_EEPROM(ser, args.pdm_only)
  File "/tmp/zigate-dev/zigate/flasher.py", line 328, in erase_EEPROM
    res = read_response(ser)
  File "/tmp/zigate-dev/zigate/flasher.py", line 111, in read_response
    return _unpack_raw_message(length, answer)
  File "/tmp/zigate-dev/zigate/flasher.py", line 123, in _unpack_raw_message
    return _responses.get(type_, Response)(type_, data, chksum)
  File "/tmp/zigate-dev/zigate/flasher.py", line 70, in __init__
    self.status = data[0]
IndexError: index out of range
read_response length 3
read_response answer b'(\x00+'
unpack raw message 3 b'(\x00+'
Response 40 b'\x00' 43

```

lboue, 2019-01-09:
J'ai fait 2 autres tests qui fonctionnent :

1. Lecture du firmware :
```
pi@raspberrypi:/tmp/zigate-dev $ python3 -m zigate.flasher -p /dev/ttyUSB0 -s zigate-current.bin
Found MAC-address: 00:15:8d:00:01:b2:2e:3a
reading old flash to zigate-current.bin
```

2. Ecriture
```
pi@raspberrypi:/tmp/zigate-dev $ python3 -m zigate.flasher -p /dev/ttyUSB0 -w ZiGate_3.0f_pre_release.bin
Found MAC-address: 00:15:8d:00:01:b2:2e:3a
writing new flash from ZiGate_3.0f_pre_release.bin
```

doudz, 2019-01-09:
Ceci dit peut-être que ça fonctionne malgré l'erreur ?

lboue, 2019-01-09:
Je ne crois pas puisqu'il connait toujours la liste des devices:
```
pi@raspberrypi:~/py-zigate $ python3 py-zigate-list.py 
DEBUG:zigate:Try loading persistent file
DEBUG:zigate:No file to load
INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB0
DEBUG:zigate:REQUEST : 0x0010 
DEBUG:zigate:Msg to send b'0010000010'
DEBUG:zigate:Encoded Msg to send b'01021010021002101003'
DEBUG:zigate:Waiting for status message for command 0x0010
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x95\x02\x10\x02\x10\x02\x10\x10\x02\x10\x03\x01\x80\x10\x02\x10\x02\x15\x9b\x02\x10\x02\x12\x02\x13\x02\x1f\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000010'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'', rssi:0
DEBUG:zigate:Received response 0x8010: b'0002030f'
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8010 - Version list : major:2, installer:30f, rssi:0, version:3.0f
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0010:0
DEBUG:zigate:Waiting for message 0x8010
DEBUG:zigate:Stop waiting, got message 0x8010
DEBUG:zigate:REQUEST : 0x0021 b'\x03\x18\xc8\x00'
DEBUG:zigate:Msg to send b'00210004f60318c800'
DEBUG:zigate:Encoded Msg to send b'0102102102100214f6021318c8021003'
DEBUG:zigate:Waiting for status message for command 0x0021
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xa4\x02\x10\x02\x10\x02\x10!\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000021'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:33, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0021:0
DEBUG:zigate:REQUEST : 0x0023 b'\x00'
DEBUG:zigate:Msg to send b'002300012200'
DEBUG:zigate:Encoded Msg to send b'010210230210021122021003'
DEBUG:zigate:Waiting for status message for command 0x0023
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xa6\x02\x10\x02\x10\x02\x10#\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000023'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:35, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0023:0
DEBUG:zigate:Check network state
DEBUG:zigate:REQUEST : 0x0024 
DEBUG:zigate:Msg to send b'0024000024'
DEBUG:zigate:Encoded Msg to send b'01021024021002102403'
DEBUG:zigate:Waiting for status message for command 0x0024
DEBUG:zigate:Raw packet received, b'\x01\x80$\x02\x10\x02\x12\xa2\x02\x14\x02\x10\x03\x01\x80\x02\x10\x02\x10\x02\x15\xa1\x02\x10\x02\x10\x02\x10$\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8024: b'04'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8024 - Network joined / formed : status:4, rssi:0
DEBUG:zigate:Received response 0x8000: b'00000024'
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:36, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0024:0
DEBUG:zigate:REQUEST : 0x0009 
DEBUG:zigate:Msg to send b'0009000009'
DEBUG:zigate:Encoded Msg to send b'010210021902100210021903'
DEBUG:zigate:Waiting for status message for command 0x0009
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x8c\x02\x10\x02\x10\x02\x10\x02\x19\x02\x10\x03\x01\x80\x02\x19\x02\x10\x16\xd7\x02\x10\x02\x10\x02\x10\x15\x8d\x02\x10\x02\x11\xb2.:>\xce\x02\x16\xa7\xaf\xcf\x88\x02\x13=\xfb\x02\x1b\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000009'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:9, error:b'', rssi:0
DEBUG:zigate:Received response 0x8009: b'000000158d0001b22e3a3ece06a7afcf88033dfb0b'
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8009 - Network state response : addr:0000, ieee:00158d0001b22e3a, pan:16078, extend_pan:479545191213252091, channel:11, rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0009:0
DEBUG:zigate:Waiting for message 0x8009
DEBUG:zigate:Stop waiting, got message 0x8009
DEBUG:zigate:Set Zigate Time (firmware >= 3.0f)
DEBUG:zigate:REQUEST : 0x0016 b'!\xe6\x892'
DEBUG:zigate:Msg to send b'001600046e21e68932'
DEBUG:zigate:Encoded Msg to send b'01021016021002146e21e6893203'
DEBUG:zigate:Waiting for status message for command 0x0016
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x93\x02\x10\x02\x10\x02\x10\x16\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000016'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:22, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0016:0
DEBUG:zigate:REQUEST : 0x0015 
DEBUG:zigate:Msg to send b'0015000015'
DEBUG:zigate:Encoded Msg to send b'01021015021002101503'
DEBUG:zigate:Waiting for status message for command 0x0015
```

lboue, 2019-01-10:
Je confirme que cela ne fonctionne pas avec flasher.py car j'ai fait l'Erase sur un vieux Windows XP avec l'outil NXP et cela à fonctionné du premier coup.

doudz, 2019-01-11:
Ok merci pour le retour.
Je vais améliorer tout ça prochainement.
Merci en tout cas d'avoir fait des essais

lboue, 2019-01-22:
Est-ce qu'on l'on peut faire quelque chose pour aider ?

doudz, 2019-01-22:
J'ai les sources du programme d'origine NXP, il faut "juste" comprendre ce qui est fait et faire pareil. Je pense que je ne peux pas juste faire un copier/coller pour des questions de license

ruimarinho, 2019-03-01:
Same issue on macOS @doudz:

```
❯ python3 -m zigate.flasher -p /dev/cu.usbserial -e -d
Found MAC-address: xx:xx:xx:xx:xx:xx:xx:xx
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.7/site-packages/zigate/flasher.py", line 374, in <module>
    main()
  File "/usr/local/lib/python3.7/site-packages/zigate/flasher.py", line 370, in main
    erase_EEPROM(ser, args.pdm_only)
  File "/usr/local/lib/python3.7/site-packages/zigate/flasher.py", line 328, in erase_EEPROM
    res = read_response(ser)
  File "/usr/local/lib/python3.7/site-packages/zigate/flasher.py", line 111, in read_response
    return _unpack_raw_message(length, answer)
  File "/usr/local/lib/python3.7/site-packages/zigate/flasher.py", line 123, in _unpack_raw_message
    return _responses.get(type_, Response)(type_, data, chksum)
  File "/usr/local/lib/python3.7/site-packages/zigate/flasher.py", line 70, in __init__
    self.status = data[0]
IndexError: index out of range
```

ruimarinho, 2019-07-31:
@doudz do we need to erase before upgrading from 3.0f to 3.1a?

doudz, 2019-08-01:
No, since 3.a1 the firmware has an internal migration function, so theorically you don't need to erase anymore.
If needed you can call erase persistent command.

ruimarinho, 2019-08-01:
@doudz the flasher portion worked very well on macOS. 3.1a migrated devices successfully, no need for re-pairing. Is the erase permanent raw command equivalent to the erase option on this module?

doudz, 2019-08-01:
Yes it's equivalent as far as I know

doudz, 2019-11-18:
Close since erasing is not really needed on recent firmware

closed 2019-11-18 by doudz