doudz/zigate #68: osram devices does not respond the first time

biau, 2019-01-10:
Hi,

I'm using the zigate lib with home assistant component.
I'm also using nodered for automation.
I have two devices type from osram: smart plug and a lightify outdoor flex rgwb.
The problem is that it looks like osram devices use some kind of sleep after some time or something like that and the result is that the device does not react at the first command.
When I say some time it's very short, max one minute.

Please find the log below and the use case:
I turn on the light addr c28c using hass gui the node red detect it and send a service call turn_on to osram strip 6201.
But the strip does not turn on.
I turn off c28c, node red send a turn_off to 6201, I can see now in debug logs the answer of 6201 to turn_off, but it was already off.
Then I turn on c28c again, 6201 turn on. Finally I turn off c28c and 6201 turn off.

I have the same behaviour with a smartplug and a aqara wireless switch. First click does not turn on smart plug, second click yes, and then I can continue turn on and off while clicking.

Don't know I my description is clear, please find de log:

> 2019-01-10 15:03:23 DEBUG (SyncWorker_15) [zigate] REQUEST : 0x0092 b'\x02\xc2\x8c\x01\x01\x01'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_15) [zigate] Msg to send b'00920006d902c28c010101'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_15) [zigate] Encoded Msg to send b'0102109202100216d90212c28c02110211021103'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_15) [zigate] Waiting for status message for command 0x0092
> 2019-01-10 15:03:23 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15o\x02\x10x\x02\x10\x92\x02\x10\x03\x01\x81\x02\x11\x02\x10\x02\x17\x90x\x02\x11\x02\x10\x02\x16\x02\x11\x02\x10i\x03\x01\x81\x02\x12\x02\x10\x02\x1eQ\xe3\xc2\x8c\x02\x11\x02\x10\x02\x16\x02\x10\x02\x10\x02\x10\x10\x02\x10\x02\x11\x02\x11f\x03\x01\x81\x02\x12\x02\x10\x11\xf6\xe3\xc2\x8c\x02\x11\x02\x10\x02\x16\xf0\x02\x10\x02\x10#\x02\x10\x02\x14\x02\x17\x02\x10\x02\x10xf\x03'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00780092'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:120, packet_type:146, error:b'', rssi:0
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8101: b'780100060100'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8101 - Default device response : sequence:120, endpoint:1, cluster:6, cmd:1, status:0, rssi:105
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'e3c28c01000600000010000101'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:227, addr:c28c, endpoint:1, cluster:6, attribute:0, status:0, data_type:16, size:1, data:True, rssi:102
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'e3c28c010006f0000023000407000078'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:227, addr:c28c, endpoint:1, cluster:6, attribute:61440, status:0, data_type:35, size:4, data:117440632, rssi:102
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
> 2019-01-10 15:03:23 DEBUG (SyncWorker_15) [zigate] STATUS code to command 0x0092:0
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] REQUEST : 0x0081 b'\x02b\x01\x01\x03\x01\xfe\x00\x05'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] Msg to send b'0081000911026201010301fe0005'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] Encoded Msg to send b'0102108102100219110212620211021102130211fe0210021503'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] Waiting for status message for command 0x0081
> 2019-01-10 15:03:23 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15}\x02\x10y\x02\x10\x81\x02\x10\x03'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00790081'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:121, packet_type:129, error:b'', rssi:0
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] STATUS code to command 0x0081:0
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] REQUEST : 0x00c0 b'\x02b\x01\x01\x03\x015\x00\x05'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] Msg to send b'00c000099b026201010301350005'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] Encoded Msg to send b'010210c0021002199b0212620211021102130211350210021503'
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] Waiting for status message for command 0x00c0
> 2019-01-10 15:03:23 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x11%\xe4\xc2\x8c\x02\x13\x02\x10\x02\x1c\x02\x10U\x02\x109\x02\x10\x02\x14B$\\)i\x03'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'e4c28c03000c00550039000442245c29'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:228, addr:c28c, endpoint:3, cluster:12, attribute:85, status:0, data_type:57, size:4, data:41.09000015258789, rssi:105
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15?\x02\x10z\x02\x10\xc0\x02\x10\x03'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Set expire timer for 3-12-85 in 2
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Cancel previous Timer <Timer(Thread-285, stopped daemon 140490216548096)>
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'007a00c0'
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:122, packet_type:192, error:b'', rssi:0
> 2019-01-10 15:03:23 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:23 DEBUG (SyncWorker_16) [zigate] STATUS code to command 0x00c0:0
[... 233 lines cut ...]
> 2019-01-10 15:03:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
> 2019-01-10 15:03:36 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x02\x1e\xbb\x02\x17\xc2\x8c\x02\x11\x02\x10\x02\x16\x02\x10\x02\x10\x02\x10\x10\x02\x10\x02\x11\x02\x10i\x03'
> 2019-01-10 15:03:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-10 15:03:36 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'07c28c01000600000010000100'
> 2019-01-10 15:03:36 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:7, addr:c28c, endpoint:1, cluster:6, attribute:0, status:0, data_type:16, size:1, data:False, rssi:105
> 2019-01-10 15:03:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
> 2019-01-10 15:03:36 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x11\x02\x10\x02\x17o\x81\x02\x13\x02\x10\x02\x16\x02\x10\x02\x10l\x03'
[... cut ...]
> 2019-01-10 15:03:54 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'52a45e010000ff01004200210121b30b0328150421a81305210c00062401000000000a2100006410000b210000'
> 2019-01-10 15:03:54 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:82, addr:a45e, endpoint:1, cluster:0, attribute:65281, status:0, data_type:66, size:33, data:0121b30b0328150421a81305210c00062401000000000a2100006410000b210000, rssi:126
> 2019-01-10 15:03:54 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-10 15:03:54 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED

doudz, 2019-01-11:
For device c28c, could you please try to call service "discover_device" and see if it resolves the problem ?

biau, 2019-01-12:
Nope does not solve the problem here.
So a call to turn on from hass ui works but the light does not respond so hass put it back off.
The only way to get the control back is to call a turn_off service.

biau, 2019-01-13:
Maybe the problem is more related to hass component I don't know.
But know I have done some demo to my kids showing the different colors. 
The last I used was blue, and now each time I turn off turn on the strip I use automation.
In hass the on/off state is correct if I turn off/on at least two times. But the color stay blue even if the service change color to warm white.
So the strip does not seems to report his color.

doudz, 2019-01-15:
reporting has been improve, check using 0.26

biau, 2019-01-17:
Mmh sorry for my dumb question but how can I do that, the latest publish version seems to be 0.25.1 ?

biau, 2019-01-20:
After a better look I've seen that 0.26 became 0.25.1.
Updated to 0.25.1 but still got the issue. 
I don't know what kind of information I could provide to help here.

doudz, 2019-01-20:
Sorry, I have to read the log you provide, no time yet

biau, 2019-01-20:
It's really not a real problem for me because I do not use colors for this strip. just reporting it to help others.

doudz, 2019-01-21:
could you clarify please, the original issue was "not responding at first time" , to check : do you see command 0x0092 in log when it's not working ?
If yes, so it's a zigbee problem (range, etc) not related to HA or zigate lib
If no, we have to found why event is missing

Next you're talking about a problem of state reporting (on/off state or color) this will be probably fix in the next release 0.26.0, so wait a little and please open a new issue if needed

closed 2019-01-21 by biau

doudz, 2019-01-21:
Pas de souci ;)