doudz/zigate #70: Aqara Light Bulb wrong initial state / missed updates

cpainchaud, 2019-01-14:
when starting HASS or when powering off-on buld manually , this library does not seem to be able to retrieve bulb current status so HASS will keep in last state it knows about it which is sometimes/often wrong.

Hopefully, this bulb sends status every 6-10 seconds with current states, including on-off info:
2019-01-14 22:19:17 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'2b7df3010000ff010042002703283c0521a4000727000000000000000008211601092100010a2100006420016520fe6621d901'
2019-01-14 22:19:17 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:43, addr:7df3, endpoint:1, cluster:0, attribute:65281, status:0, data_type:66, size:39, data:03283c0521a4000727000000000000000008211601092100010a2100006420016520fe6621d901, rssi:48
2019-01-14 22:19:17 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED

on-off state is in bold here:
ON: 0328260521a3000727000000000000000008211601092100010a2100006420**###01**6520fe6621d901
OFF: 03282c0521a3000727000000000000000008211601092100010a2100006420**### 00**6520fe6621d901

Would it be possible to decode attribute 65281 and get useful infos/states from there?

doudz, 2019-01-15:
If you call discover_device service ,does it fix the problem

cpainchaud, 2019-01-17:
here is HASS logs for discovery:

2019-01-17 19:25:45 WARNING (Thread-17) [zigate] No template found for lumi.light.aqcn02
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Start automagic bind and report process for device Device lumi.light.aqcn02 (7df3) 00158d0002c65701
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] bind and report for cluster 0x0006
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] REQUEST : 0x0030 b'\x00\x15\x8d\x00\x02\xc6W\x01\x01\x00\x06\x03\x00\x15\x8d\x00\x01\xedE\x12\x01'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Msg to send b'003000150900158d0002c657010100060300158d0001ed451201'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Encoded Msg to send b'0102103002101502190210158d02100212c657021102110210021602130210158d02100211ed4512021103'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Waiting for status message for command 0x0030
2019-01-17 19:25:45 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xc4\x02\x10q\x02\x100\x02\x10\x03\x01\x800\x02\x10\x02\x13\xb0q\x02\x10r\x03'
2019-01-17 19:25:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00710030'
2019-01-17 19:25:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:113, packet_type:48, error:b'', rssi:0
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8030: b'7100'
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8030 - Bind response : sequence:113, status:0, rssi:114
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] STATUS code to command 0x0030:0
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Waiting for message 0x8030
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Stop waiting, got message 0x8030
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] REQUEST : 0x0120 b'\x02}\xf3\x01\x01\x00\x06\x00\x00\x00\x00\x01\x00\x10\x00\x00\x00\x01\x0e\x10\x00\x00\x00'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Msg to send b'01200017b2027df30101000600000000010010000000010e10000000'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Encoded Msg to send b'01021120021017b202127df30211021102100216021002100210021002110210100210021002100211021e1002100210021003'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Waiting for status message for command 0x0120
2019-01-17 19:25:45 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xe1\x02\x10E\x02\x11 \x02\x10\x03\x01\x81 \x02\x10\x02\x18\x17E}\xf3\x02\x11\x02\x10\x02\x16\x02\x10r\x03'
2019-01-17 19:25:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00450120'
2019-01-17 19:25:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:69, packet_type:288, error:b'', rssi:0
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8120: b'457df301000600'
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8120 - Configure Reporting response : sequence:69, addr:7df3, endpoint:1, cluster:6, status:0, rssi:114
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] STATUS code to command 0x0120:0
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Waiting for message 0x8120
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Stop waiting, got message 0x8120
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] bind and report for cluster 0x0008
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] REQUEST : 0x0030 b'\x00\x15\x8d\x00\x02\xc6W\x01\x01\x00\x08\x03\x00\x15\x8d\x00\x01\xedE\x12\x01'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Msg to send b'003000150700158d0002c657010100080300158d0001ed451201'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Encoded Msg to send b'0102103002101502170210158d02100212c657021102110210021802130210158d02100211ed4512021103'
2019-01-17 19:25:45 DEBUG (Thread-17) [zigate] Waiting for status message for command 0x0030
2019-01-17 19:25:45 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xc7\x02\x10r\x02\x100\x02\x10\x03\x01\x800\x02\x10\x02\x13\xb4r\x02\x10u\x03'
2019-01-17 19:25:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00720030'
2019-01-17 19:25:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:114, packet_type:48, error:b'', rssi:0
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8030: b'7200'
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-17 19:25:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8030 - Bind response : sequence:114, status:0, rssi:117
[... cut ...]
2019-01-17 19:25:49 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-17 19:25:49 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'005a0140'
2019-01-17 19:25:49 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:90, packet_type:320, error:b'', rssi:0
2019-01-17 19:25:49 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-17 19:25:49 DEBUG (Thread-17) [zigate] STATUS code to command 0x0140:0

cpainchaud, 2019-01-18:
more informations: HASS says the bulb was last seen online a few days ago, despite the bulb sending this Hearbeat message every  8 seconds.

doudz, 2019-01-21:
The "hearbeat" is not handled yet,
could you provide the zigate.json file ?

cpainchaud, 2019-01-21:
sure here it is :
```json
{
    "devices": [
        {
            "addr": "e95e",
            "discovery": "templated",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 1,
                                    "data": 5,
                                    "name": "application_version",
                                    "value": 5
                                },
                                {
                                    "attribute": 4,
                                    "data": "LUMI",
                                    "name": "manufacturer",
                                    "value": "LUMI"
                                },
                                {
                                    "attribute": 5,
                                    "data": "lumi.sensor_motion.aq2",
                                    "name": "type",
                                    "type": "str",
                                    "value": "lumi.sensor_motion.aq2"
                                },
                                {
                                    "attribute": 65281,
                                    "data": "01210d0c0328170421a80105210f00062400000000000a2100006410000b210a00",
                                    "name": "battery",
                                    "unit": "V",
                                    "value": 3.085
                                },
                                {
                                    "attribute": 7,
                                    "data": 3,
                                    "name": "power_source",
                                    "value": 3
                                }
                            ],
                            "cluster": 0
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": 14,
                                    "name": "luminosity",
                                    "unit": "lm",
                                    "value": 14
                                }
                            ],
                            "cluster": 1024
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": false,
                                    "expire": 10,
                                    "name": "presence",
                                    "type": "bool",
                                    "value": false
                                }
                            ],
[... 719 lines cut ...]
                        19,
                        6,
                        1,
                        1030,
                        8,
                        768
                    ],
                    "profile": 260
                }
            ],
            "generictype": "",
            "info": {
                "addr": "4c6b",
                "assumed_state": true,
                "ieee": "00158d0002c63bd6",
                "last_seen": "2019-01-18 17:36:05",
                "mac_capability": "10001110",
                "rssi": 48
            }
        }
    ],
    "groups": {},
    "scenes": {}
}
```

doudz, 2019-01-22:
is it better with the last release ?

cpainchaud, 2019-01-23:
It works, thx!

quick question : did you see any unknown attribute which may match to a 'transition' time ? The bulb goes up/down gradually over 1 or 2 seconds. I was wondering if any attribute allows to customize this behavior?

doudz, 2019-01-24:
By default I set the transition to 0, means no transition so I suppose the bulb do what it want
Using the light.turn_on service you could set transition time
[image]

doudz, 2019-01-24:
Closing since it's solved

closed 2019-01-24 by doudz