doudz/zigate #73: Problème de remontée des températures negatives avec capteur Xiaomi

Jsl-1, 2019-01-20:
Bonjour @doudz,
Les températures négatives semblent remonter comme positive dans ha sur les capteurs rond Xiaomi (pas tester les carrés). J'imagine que c'est un problème  sur la librairie. 
@pipiche38 semble avoir corrigé un problème similaire sur domoticz (https://www.domoticz.com/forum/viewtopic.php?t=26254). 
En te remerciant énormément pour tout ton investissement sur la prise en charge de la Zigate.

doudz, 2019-01-20:
It's a bit strange, could you provide some debug log ?

Jsl-1, 2019-01-20:
Yes, as soon as negative temp append ;)
Don't have enough log for now. Maybe tomorrow or tuesday.

ChoOo7, 2019-01-20:
Put it in the freezer !

Jsl-1, 2019-01-20:
I'm suprised to have enough signal in the freezer, but it's working. And in fact there is no problem, ha reports negative correctly. There should be another problem with the sensor. Will investigate.
Sorry for  making you waste your time and thanks you again for your work and your help...

closed 2019-01-20 by Jsl-1