doudz/zigate #76: lumi.ctrl_neutral1 stopped working

pawelka, 2019-01-22:
I'm not sure in details what was changed, but aqara single wall switch stop working.

In the past (1 month ago):
- for cluster 6 two endpoint were discover 1 and 2. In case of this switch endpoint one is fake and second only work.

Latest version from today and previous:
- only first endpoint has action onoff, other are empty.

Because of such discovery only one switch is created (fake one).

I wanted to create template, but I'm not sure if it's make sense if endpoint 2 is not discover.

pawelka, 2019-01-22:
All logs: "Update attribute for ..." after descovering device.

2019-01-22 15:04:39 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update attribute for device Device lumi.ctrl_neutral1 (acc2) 00158d0001f4dcd4 {'endpoint': 1, 'cluster': 0, 'addr': 'acc2', 'attribute': 1, 'data': 21, 'name': 'application_version', 'value': 21}
2019-01-22 15:04:39 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update attribute for device Device lumi.ctrl_neutral1 (acc2) 00158d0001f4dcd4 {'endpoint': 2, 'cluster': 6, 'addr': 'acc2', 'attribute': 0, 'data': False, 'name': 'onoff', 'type': <class 'bool'>, 'value': False}
2019-01-22 15:04:43 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update attribute for device Device lumi.ctrl_neutral1 (acc2) 00158d0001f4dcd4 {'endpoint': 1, 'cluster': 0, 'addr': 'acc2', 'attribute': 5, 'data': 'lumi.ctrl_neutral1', 'name': 'type', 'type': <class 'str'>, 'value': 'lumi.ctrl_neutral1'}

There is endpoint 2 cluster 6, but when switch is created is endpoint 1 :-/ weird.

pawelka, 2019-01-22:
{
            "addr": "acc2",
            "discovery": "",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 5,
                                    "data": "lumi.ctrl_neutral1",
                                    "name": "type",
                                    "type": "str",
                                    "value": "lumi.ctrl_neutral1"
                                },
                                {
                                    "attribute": 1,
                                    "data": 21,
                                    "name": "application_version",
                                    "value": 21
                                }
                            ],
                            "cluster": 0
                        }
                    ],
                    "device": 6,
                    "endpoint": 1,
                    "in_clusters": [
                        0,
                        3,
                        1,
                        2,
                        25,
                        10
                    ],
                    "out_clusters": [
                        0,
                        10,
                        25
                    ],
                    "profile": 260
                },
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": false,
                                    "name": "onoff",
                                    "type": "bool",
                                    "value": false
                                },
                                {
                                    "attribute": 61440,
                                    "data": 61653504
                                }
                            ],
                            "cluster": 6
                        }
                    ],
                    "device": 256,
                    "endpoint": 2,
                    "in_clusters": [
                        16,
                        6,
                        4,
                        5
                    ],
                    "out_clusters": [],
                    "profile": 260
                },
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": false,
                                    "name": "onoff4",
                                    "type": "bool",
                                    "value": false
                                }
                            ],
                            "cluster": 6
                        }
                    ],
                    "device": 0,
                    "endpoint": 4,
                    "in_clusters": [
                        18,
                        6
                    ],
                    "out_clusters": [],
                    "profile": 260
                },
                {
                    "clusters": [],
                    "device": 0,
                    "endpoint": 6,
                    "in_clusters": [
                        18,
                        6
                    ],
                    "out_clusters": [],
                    "profile": 260
                },
                {
                    "clusters": [],
                    "device": 0,
                    "endpoint": 5,
                    "in_clusters": [
                        18,
                        6
                    ],
                    "out_clusters": [],
                    "profile": 260
                }
            ],
            "generictype": "",
            "info": {
                "addr": "acc2",
                "assumed_state": true,
                "bit_field": "0100000000000001",
                "descriptor_capability": "00000000",
                "id": 0,
                "ieee": "00158d0001f4dcd4",
                "last_seen": "2019-01-22 15:09:18",
                "mac_capability": "10000100",
                "manufacturer_code": "1037",
                "max_buffer": 127,
                "max_rx": 100,
                "max_tx": 100,
                "power_type": 1,
                "rssi": 102,
                "server_mask": 1
            }
        }

pawelka, 2019-01-22:
Probably this cause my problem ;-) :                         

```
# Oh please XIAOMI, respect the standard...
                        if ep_id != 1 and self.get_property_value('type') == 'lumi.ctrl_neutral1':
                            ep_id -= 1
                        actions[ep_id].append(ACTIONS_ONOFF)
```

doudz, 2019-01-22:
lol maybe ;)

pawelka, 2019-01-22:
checking by replacing: ep_id -= 1 with continue if resolve problem

pawelka, 2019-01-22:
Works:
```
                    if 0x0006 in endpoint['in_clusters']:
                        # Oh please XIAOMI, respect the standard...
                        if ep_id == 1 and self.get_property_value('type') == 'lumi.ctrl_neutral1':
                            continue
                        actions[ep_id].append(ACTIONS_ONOFF)
```

I'm only skipping ep_id==1 instead of decreasing other endpoint.

Will you add this to main code or you need PR for this ? I tested by hand without branch.

doudz, 2019-01-22:
If you completly remove the 2 lines, could you confirm everything is working ? both on/off command and state return ?

```python
                        if ep_id != 1 and self.get_property_value('type') == 'lumi.ctrl_neutral1':
                            ep_id -= 1
```

pawelka, 2019-01-22:
As described 3min before your comment, everything works as expected if fake endpoint will be skipped.

doudz, 2019-01-22:
Initially this hack has been added because the state was on endpoint 2 but the on/off works on endpoint 1
I will check

doudz, 2019-01-22:
> As described 3min before your comment, everything works as expected if fake endpoint will be skipped.

Ok so I remove the lines.
Thanks for all

closed 2019-01-22 by doudz