doudz/zigate #78: OSRAM Device LIGHTIFY Outdoor Flex RGBW does not respond to commands with 0.26.x

biau, 2019-01-23:
So I've tested 0.26.0 and 0.26.1 and the result is not better than previous version.
Actually it's worst because I can't control the strip anymore.
Please find the log from hass.
08:54:05 I plugged the strip, you can see that zigate see it connects.
08:45:19 I tried to switch the state and nothing happens.
> 2019-01-23 08:54:05 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b"\x01\x02\x10M\x02\x10\x02\x1c'\x8ft|\xb0>\xaa\x02\x10\xa9\xa5\x1d\x8eZ\x03\x01\x87\x02\x11\x02\x10\x02\x13\xdf\x02\x10\x02\x10Z\x03"
> 2019-01-23 08:54:05 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'8f747cb03eaa00a9a51d8e'
> 2019-01-23 08:54:05 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:8f74, ieee:7cb03eaa00a9a51d, mac_capability:10001110, rssi:90
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] Device Announce RESPONSE 0x004D - Device announce : addr:8f74, ieee:7cb03eaa00a9a51d, mac_capability:10001110, rssi:90
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8701: b'0000'
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8701 - Route Discovery Confirmation : status:0, network_status:0, rssi:90
> 2019-01-23 08:54:05 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-23 08:54:19 DEBUG (SyncWorker_12) [zigate] REQUEST : 0x0092 b'\x02\x8ft\x01\x03\x00'
> 2019-01-23 08:54:19 DEBUG (SyncWorker_12) [zigate] Msg to send b'009200066f028f74010300'
> 2019-01-23 08:54:19 DEBUG (SyncWorker_12) [zigate] Encoded Msg to send b'01021092021002166f02128f7402110213021003'
> 2019-01-23 08:54:19 DEBUG (SyncWorker_12) [zigate] Waiting for status message for command 0x0092
> 2019-01-23 08:54:19 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15Z\x02\x10M\x02\x10\x92\x02\x10\x03\x01\x87\x02\x12\x02\x10\x02\x1e[\xd3\x02\x11\x02\x13\x02\x12\x8ft\x02\x11\x02\x14\x02\x10\x02\x1a\x02\x10\x02\x11\xab]\x03'
> 2019-01-23 08:54:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'004d0092'
> 2019-01-23 08:54:19 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:77, packet_type:146, error:b'', rssi:0
> 2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8702: b'd30103028f740104000a0001ab'
> 2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:211, source_endpoint:1, dst_endpoint:3, dst_address_mode:2, dst_address:0x8f74, sequence:171, rssi:93
> 2019-01-23 08:54:19 ERROR (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:211, source_endpoint:1, dst_endpoint:3, dst_address_mode:2, dst_address:0x8f74, sequence:171, rssi:93
> 2019-01-23 08:54:19 DEBUG (SyncWorker_12) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:77, packet_type:146, error:b'', rssi:0
> 2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
> 2019-01-23 08:54:29 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x87\x02\x11\x02\x10\x02\x13\x02\x18\xd0\x02\x10]\x03'
> 2019-01-23 08:54:29 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
> 2019-01-23 08:54:29 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8701: b'd000'
> 2019-01-23 08:54:29 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8701 - Route Discovery Confirmation : status:208, network_status:0, rssi:93
> 2019-01-23 08:54:29 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED

I can also post discovery_device log but it's more verbose and it's not easy to see only what is related to this stpecific device.

doudz, 2019-01-23:
**`2019-01-23 08:54:19 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8702 - APS Data Confirm Fail : status:211, source_endpoint:1, dst_endpoint:3, dst_address_mode:2, dst_address:0x8f74, sequence:171, rssi:93`**

Looks like a range problem

biau, 2019-01-23:
with rssi 93 ?
It's curious because it's at the same place as prior to 0.26.x, with 0.25.x the light strip respond to all command with no delay, my only issue was state reporting not working for color and after the first command.
I'm going to test it in the same room as the zigate.

doudz, 2019-01-23:
Since the strip was offline, maybe a route needs to be discovered, could you post the discover_device log ?

biau, 2019-01-23:
Yes it look's like that. as usual you're right but now I have a better understanding of logs.
I think the route was using my kitchen or my livingroom aqara wall switch as intermediate router, that's why I get a good rssi. 
But I don't know why it looks like the zigate received device announce from the strip but was not able to send any message back.
I unplugged the strip and put it in another corner of the kitchen and then the strip connect to the zigate with rssi at 33. But it works! I guess at this place the strip does not use intermediate router anymore.
Finally I restore the strip at the initial place, get back the rssi around 90 and I can control it again.
Is there anyway to see the actual route for a device ? Like with zigbee2mqtt you can display routes using a graphical node tool.

By the way the strip is still not reporting his color with 0.26.x.

doudz, 2019-01-23:
Some devices just don't support reporting on some attributes, I don't know for this one.
You could try to call "discover_device" (again I know...) and post the result here with the content of the zigate.json file

biau, 2019-01-23:
Don't need to be sorry I'm glad you answer my questions : 
discovery log: https://termbin.com/hgr5
zigate.json: https://termbin.com/x9rw

ruimarinho, 2019-09-27:
@doudz `APS Data Confirm Fail` guaranteed to be range issues? I have some with `RESPONSE 0x8702 - APS Data Confirm Fail : status:212, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:512b, sequence:62, lqi:8` as well. The lqi seems very low but I simply do not understand why the IKEA repeaters are not improving the network..