doudz/zigate #82: [zigate] Command 0x0100 failed Failed (ZigBee event codes) 21 : b''

pheuzoune, 2019-01-25:
Hi, 

I've pushed some template on my fork of your clone generated using your custom component for HA. 

I have multiples logs saying : 
` [zigate] Command 0x0100 failed Failed (ZigBee event codes) 21 : b''`

I suppose it's my lumi.ctrl_neutral2 but not sure. 

My Zigate in 3.0f_pre 
Do you know how I can prevent those messages ? It seems to create loads of latency with command sent to my wallswitch.

doudz, 2019-01-25:
The command 0x0100 is read_attribute_request so if it fails it suppose a reading problem.
Are you using the latest HA component and latest zigate release ?
Could you provide debug log ?

doudz, 2019-01-26:
could you provide your zigate.json too ?

pheuzoune, 2019-01-26:
I've modified my zigate.json to remove all attributes that seems to be useless for my device (power_source, battery_level , etc ... ). 
It's not better, 
It gives only 2 errors of this kind a each refrech / change of state. 
I can't find why there are still 2 left. 

Can this be solved by adding the right template in the lib and removing invalid attributes ? 

```
        {
            "addr": "7bc1",
            "discovery": "templated",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 1,
                                    "data": 1,
                                    "name": "application_version",
                                    "value": 1
                                },
                                {
                                    "attribute": 4,
                                    "data": "LUMI",
                                    "name": "manufacturer",
                                    "value": "LUMI"
                                },
                                {
                                    "attribute": 5,
                                    "data": "lumi.ctrl_neutral2",
                                    "name": "type",
                                    "type": "str",
                                    "value": "lumi.ctrl_neutral2"
                                }
                            ],
                            "cluster": 0
                        }
                    ],
                    "device": 6,
                    "endpoint": 1,
                    "in_clusters": [
                        0,
                        3,
                        1,
                        2,
                        25,
                        10
                    ],
                    "out_clusters": [
                        0,
                        10,
                        25
                    ],
                    "profile": 260
                },
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": true,
                                    "name": "onoff",
                                    "type": "bool",
                                    "value": true
                                }
                            ],
                            "cluster": 6
                        }
                    ],
                    "device": 256,
                    "endpoint": 2,
                    "in_clusters": [
                        16,
                        6,
                        4,
                        5
                    ],
                    "out_clusters": [],
                    "profile": 260
                },
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": true,
                                    "name": "onoff3",
                                    "type": "bool",
                                    "value": true
                                }
                            ],
                            "cluster": 6
                        }
                    ],
                    "device": 256,
                    "endpoint": 3,
                    "in_clusters": [
                        16,
                        6,
                        4,
                        5
                    ],
                    "out_clusters": [],
                    "profile": 260
                }
            ],
```

doudz, 2019-01-27:
the "problem" is `"assumed_state": true`, because of this, HA will try to refresh the value which seems to failed.
assumed_state has been set because configuring reporting has failed, doesn't it work in previous release ?

doudz, 2019-01-27:
Well, I think I understand what's happening:
- Configure reporting fails because device doesn't support it => So I assume I have to poll the state
- BUT the device reports his states by default so I don't have to poll the state
I will make some tests

pheuzoune, 2019-01-27:
Hi, 

you're posting too quick :)

Here is the debug log when launching a discover :+1: 

`2019-01-27 10:02:09 DEBUG (Thread-22) [zigate] REQUEST : 0x0043 b'{\xc1\x04'
2019-01-27 10:02:09 DEBUG (Thread-22) [zigate] Msg to send b'00430003fe7bc104'
2019-01-27 10:02:09 DEBUG (Thread-22) [zigate] Encoded Msg to send b'0102104302100213fe7bc1021403'
2019-01-27 10:02:09 DEBUG (Thread-22) [zigate] Waiting for status message for command 0x0043
2019-01-27 10:02:09 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15R\x02\x10\x94\x02\x10C\x02\x10\x03'
2019-01-27 10:02:09 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-27 10:02:09 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00940043'
2019-01-27 10:02:09 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:148, packet_type:67, error:b'', rssi:0
2019-01-27 10:02:09 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-27 10:02:09 DEBUG (Thread-22) [zigate] STATUS code to command 0x0043:RESPONSE 0x8000 - Status response : status:0, sequence:148, packet_type:67, error:b'', rssi:0
2019-01-27 10:02:10 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80C\x02\x10\x12b\x94\x02\x10{\xc1\x02\x1c\x02\x14\x02\x11\x02\x14\x02\x10\x02\x10\x02\x12\x02\x12\x02\x10\x12\x02\x10\x02\x16\x02\x10\x84\x03'
2019-01-27 10:02:10 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x106\x93k{\xc1\x02\x11\x02\x10\x02\x10\xff\x02\x11\x02\x10B\x02\x10)d\x10\x02\x10e\x10\x02\x10n \x02\x10o \x02\x10\x02\x11!\xe4\x02\x1c\x02\x13(\x1b\x02\x15!\x02\x19\x02\x10\x02\x18!\x16&\x02\x1a!\x02\x10\x02\x10\x99#\x02\x10\x02\x10\x02\x10\x02\x10\x9b!\x02\x10\x02\x10\x84\x03'
2019-01-27 10:02:10 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8043: b'94007bc10c040104000002020012000600'
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8043 - Simple descriptor : sequence:148, status:0, addr:7bc1, length:12, endpoint:4, profile:260, device:0, bit_field:00000010, inout_clusters:b'\x02\x00\x12\x00\x06\x00', rssi:132, in_clusters:(18, 6), out_clusters:()
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] REQUEST : 0x0043 b'{\xc1\x04'
2019-01-27 10:02:10 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Msg to send b'00430003fe7bc104'
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'6b7bc1010000ff01004200296410006510006e20006f20000121e40c03281b05210900082116260a2100009923000000009b210000'
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Encoded Msg to send b'0102104302100213fe7bc1021403'
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:107, addr:7bc1, endpoint:1, cluster:0, attribute:65281, status:0, data_type:66, size:41, data:6410006510006e20006f20000121e40c03281b05210900082116260a2100009923000000009b210000, rssi:132
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Waiting for status message for command 0x0043
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-27 10:02:10 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15S\x02\x10\x95\x02\x10C\x02\x10\x03'
2019-01-27 10:02:10 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00950043'
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:149, packet_type:67, error:b'', rssi:0
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] STATUS code to command 0x0043:RESPONSE 0x8000 - Status response : status:0, sequence:149, packet_type:67, error:b'', rssi:0
2019-01-27 10:02:10 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-27 10:02:11 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80C\x02\x10\x12c\x95\x02\x10{\xc1\x02\x1c\x02\x14\x02\x11\x02\x14\x02\x10\x02\x10\x02\x12\x02\x12\x02\x10\x12\x02\x10\x02\x16\x02\x10\x84\x03'
2019-01-27 10:02:11 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-27 10:02:11 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8043: b'95007bc10c040104000002020012000600'
2019-01-27 10:02:11 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8043 - Simple descriptor : sequence:149, status:0, addr:7bc1, length:12, endpoint:4, profile:260, device:0, bit_field:00000010, inout_clusters:b'\x02\x00\x12\x00\x06\x00', rssi:132, in_clusters:(18, 6), out_clusters:()
2019-01-27 10:02:11 DEBUG (ZiGate-Decode data) [zigate] REQUEST : 0x0043 b'{\xc1\x04'
2019-01-27 10:02:11 DEBUG (ZiGate-Decode data) [zigate] Msg to send b'00430003fe7bc104'
2019-01-27 10:02:11 DEBUG (ZiGate-Decode data) [zigate] Encoded Msg to send b'0102104302100213fe7bc1021403'
2019-01-27 10:02:11 DEBUG (ZiGate-Decode data) [zigate] Waiting for status message for command 0x0043
2019-01-27 10:02:11 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15P\x02\x10\x96\x02\x10C\x02\x10\x03'
2019-01-27 10:02:11 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
[... cut ...]
2019-01-27 10:02:19 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:158, packet_type:67, error:b'', rssi:0
2019-01-27 10:02:19 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-27 10:02:19 DEBUG (ZiGate-Decode data) [zigate] STATUS code to command 0x0043:RESPONSE 0x8000 - Status response : status:0, sequence:158, packet_type:67, error:b'', rssi:0
2019-01-27 10:02:19 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
`

doudz, 2019-01-27:
I think you cut the log, I don't see the bind and report sequence

pheuzoune, 2019-01-27:
Ok. Sorry, I didn't think it was long and so verbose. 

Here is the log in attachment. 
[home-assistant.log](https://github.com/doudz/zigate/files/2800538/home-assistant.log)

pheuzoune, 2019-01-28:
Hi, 

Just tested dev branch. 
No more errors in HA logs
Operations seems really faster and no more hang of the zigate a each state change. 

thanks a lot !

doudz, 2019-01-28:
Thanks to you

closed 2019-01-28 by doudz

ruimarinho, 2019-09-27:
@doudz I am still seeing this on firmware 3.1a every second:

```
2019-09-27 00:48:25 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed Failed (ZigBee event codes) 15 : b''
2019-09-27 00:48:25 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-09-27 00:48:25 DEBUG (SyncWorker_3) [zigate] STATUS code to command 0x0100:RESPONSE 0x8000 - Status response : status:21, sequence:31, packet_type:256, error:b'', lqi:0
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] REQUEST : 0x0100 b'\x02\x92\xf0\x01\x01\x00\x06\x00\x00\x00\x00\x01\x00\x00'
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] Msg to send b'0100000e680292f00101000600000000010000'
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] Encoded Msg to send b'01021102100210021e68021292f00211021102100216021002100210021002110210021003'
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] Waiting for status message for command 0x0100
2019-09-27 00:48:25 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xb1\x15 \x02\x11\x02\x10\x02\x10\x03'
2019-09-27 00:48:25 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-09-27 00:48:25 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'15200100'
2019-09-27 00:48:25 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:21, sequence:32, packet_type:256, error:b'', lqi:0
2019-09-27 00:48:25 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed Failed (ZigBee event codes) 15 : b''
2019-09-27 00:48:25 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] STATUS code to command 0x0100:RESPONSE 0x8000 - Status response : status:21, sequence:32, packet_type:256, error:b'', lqi:0
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] REQUEST : 0x0100 b'\x02\x92\xf0\x01\x01\x00\x08\x00\x00\x00\x00\x01\x00\x00'
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] Msg to send b'0100000e660292f00101000800000000010000'
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] Encoded Msg to send b'01021102100210021e66021292f00211021102100218021002100210021002110210021003'
2019-09-27 00:48:25 DEBUG (SyncWorker_9) [zigate] Waiting for status message for command 0x0100
2019-09-27 00:48:25 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xb0\x15!\x02\x11\x02\x10\x02\x10\x03'
2019-09-27 00:48:25 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-09-27 00:48:25 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'15210100'
2019-09-27 00:48:25 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:21, sequence:33, packet_type:256, error:b'', lqi:0
2019-09-27 00:48:25 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed Failed (ZigBee event codes) 15 : b''
```

Any idea?

medaemon, 2019-12-07:
I have the same issues repeating frequently since I upgraded firware to 3.1a:
Command 0x0100 failed Failed (ZigBee event codes) 15 : b''