doudz/zigate #83: Optionnal address mode argument to action_onoff

Jsl-1, 2019-01-25:
Hi @doudz,
Would it be possible to add address mode as optionnal argument to action_onoff method in core.py ?
I ask because i need to call action_onoff for group address. I have tested with raw command and it's working great.

doudz, 2019-01-25:
It's possible, but I have an other suggestion. Since the group address is supposed to be known maybe I could just choose the right address mode depending of the presence of the address in group list ?
or I can do both, provide optionnal adress mode argument and if not specified, try to choose the right one

Jsl-1, 2019-01-25:
It would be great 👍
I can add command in home assistant when it's done if you want ;)

doudz, 2019-01-25:
I just made a commit in dev, could you try ?

Jsl-1, 2019-01-25:
It's working great

Jsl-1, 2019-01-25:
Sorry to be late, i had to try from home assistant (and create new command) because i'm not alone at home ;) 
Thank you very much !