doudz/zigate #84: lumi.remote.b286acn01 badly recognize

doudz, 2019-01-30:
similar to https://github.com/doudz/zigate/issues/49

pheuzoune, 2019-01-30:
Hello, 

As it seems to be more related to your library, I'll post my intel here. 

Here is an extract of zigate.json : 

[lumi.remote.b286acn01_zigate.json.txt](https://github.com/doudz/zigate/files/2811684/lumi.remote.b286acn01_zigate.json.txt)

And here is extract of debug logs with click, double click and hold on each endpoint (Button left, button right, Both button) : 
[lumi.remote.b286acn01.debug_logs.log](https://github.com/doudz/zigate/files/2811691/lumi.remote.b286acn01.debug_logs.log)

pheuzoune, 2019-01-30:
I closed the pull request as I made a mistake. I though the device ID was an internal ID of your lib based on template, but it's defined by the device itself apparently. 

So I have some questions to implement it the right way. 

1) I'm looking for a way to catch witch  device (lumi.remote.b286acn01) I'm facing in Cluster Object.
Is it available from Cluster Object ? 
Is it possible to add this ? 

2)  Why 0.5 sec of expiry is too fast ? 
We're looking at a wallswitch. Using it with 0.5 allows a usage like "click to switch on , oops wrong button, click to switch off" 
With 2 seconds of expiry, both click are viewed as only one.

doudz, 2019-01-30:
Since it doesn't provide a device ID except on the first endpoint, it could be a good idea to use a template to set a fake device ID.
We can use the same device ID as the first endpoint

0.5 second is a bit short for system like Jeedom or Home assistant, but we could try and see what happens

pheuzoune, 2019-01-31:
I've tried this way and it does not work. 
Left button is  on same endpoint as device attributes, so fake device ID is overwritter. 

I've made another test using a new parameter "decoder" in the template with standard value "generic". 
This parameter is available from Cluster Objects at init and allow special handling.  

Pushing the pull request right now.

doudz, 2019-01-31:
I fixed the problem, could you check ?

doudz, 2019-02-05:
should be ok in 0.27.1

closed 2019-02-05 by doudz

stephane, 2020-11-30:
In my previous HA installation (w/o Docker but venv and zigate), I was able to properly detect the switch (2 switchs on square support) but not with HA OS (5.6 64 bits), HA 0.118.4 and Zigbee HA, this device is now detected has `LUMI lumi.remote.b286acn01`.