doudz/zigate #85: lumi.vibration.aq1: struct.error: unpack requires a bytes object of length 8

pheuzoune, 2019-01-30:
Hi, 

Python exception happen when tilting the vibration sensor. 

Here is the extract of zigate.json : 
[lumi.vibration.aq1_zigate.json.txt](https://github.com/doudz/zigate/files/2811715/lumi.vibration.aq1_zigate.json.txt)

And the debug log (I touched it, waited a bit then tilted it) : 
[lumi.vibration.aq1_debug.log](https://github.com/doudz/zigate/files/2811720/lumi.vibration.aq1_debug.log)

doudz, 2019-01-30:
https://github.com/fairecasoimeme/ZiGate/issues/134

pheuzoune, 2019-01-30:
Confirmed. 

No more stacktrace and values seems correct. 
I'll close this one and we'll have to wait for a fix in the firmware for definitive fix. 

Thanks again !

closed 2019-01-30 by pheuzoune