doudz/zigate #98: Stack after removing zigate.json and recreating it from scratch

pawelka, 2019-02-11:
2019-02-11 23:01:36 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:158, packet_type:67, error:b'', rssi:0
2019-02-11 23:01:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-02-11 23:01:36 DEBUG (ZiGate-Decode data) [zigate] STATUS code to command 0x0043:RESPONSE 0x8000 - Status response : status:0, sequence:158, packet_type:67, error:b'', rssi:0
2019-02-11 23:01:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-02-11 23:01:37 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x106)W\xfe\xda\x02\x11\x02\x10\x02\x10\xff\x02\x11\x02\x10B\x02\x10)d\x10\x02\x10e\x10\x02\x10n \x02\x10o \xff\x02\x11!\xe4\x02\x1c\x02\x13(\x1a\x02\x15!\x1d\x02\x10\x02\x18!\x12&\x02\x1a!\x02\x10\x02\x10\x99#\x02\x10\x02\x10\x02\x10\x02\x10\x9b'
2019-02-11 23:01:38 WARNING (ZiGate-Decode data) [zigate] No response waiting command 0x8120
Exception in thread ZiGate-Decode data:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 474, in decode_data
    self.interpret_response(response)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 556, in interpret_response
    device.load_template()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2678, in load_template
    self._bind_report()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2104, in _bind_report
    for endpoint_id, endpoint in endpoints_list:
RuntimeError: dictionary changed size during iteration

2019-02-11 23:01:38 WARNING (Thread-4) [zigate] No response waiting command 0x8120
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 354, in need_discovery
    if device.need_discovery():
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2621, in need_discovery
    self.load_template()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2678, in load_template
    self._bind_report()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2104, in _bind_report
    for endpoint_id, endpoint in endpoints_list:
RuntimeError: dictionary changed size during iteration

pawelka, 2019-02-11:
And one more with fake endpoint:
2019-02-11 23:36:35 DEBUG (Thread-4) [zigate] Start automagic bind and report process for device Device lumi.ctrl_neutral1 (ac57) 00158d0001f4dcd4
2019-02-11 23:36:35 DEBUG (SyncWorker_27) [zigate] Saving persistent file
2019-02-11 23:36:35 DEBUG (Thread-4) [zigate] Bind and report endpoint 1 for device Device lumi.ctrl_neutral1 (ac57) 00158d0001f4dcd4
2019-02-11 23:36:35 DEBUG (SyncWorker_27) [zigate] File already existing, make a backup before
2019-02-11 23:36:35 DEBUG (Thread-4) [zigate] bind and report for cluster 0x0001
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 354, in need_discovery
    if device.need_discovery():
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2621, in need_discovery
    self.load_template()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2678, in load_template
    self._bind_report()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 2109, in _bind_report
    self._zigate.bind_addr(self.addr, endpoint_id, 0x0001)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 946, in bind_addr
    return self.bind(ieee, endpoint, cluster, dst_addr, dst_endpoint)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 934, in bind
    dst_addr, dst_endpoint)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 918, in _bind_unbind
    dst_addr_mode = self._choose_addr_mode(dst_addr)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 898, in _choose_addr_mode
    if len(addr_ieee) == 4:
TypeError: object of type 'NoneType' has no len()

doudz, 2019-02-12:
I'll take a look.
Just a question, did you recreate the file by hand ?, the last error suppose the IEEE is missing, which is impossible in normal conditions because the value is retrieved at pairing time

doudz, 2019-02-12:
Fix the first error : https://github.com/doudz/zigate/commit/dbe40cec1ccf2d9446782b3dee2d0bce89757815

pawelka, 2019-02-12:
Yes. I removed it by hand.

doudz, 2019-02-12:
Ok I think I found the problem, did you get problem right after flashing the new firmware 3.0f ? and erasing EEProm ?

doudz, 2019-02-12:
It should be ok now

doudz, 2019-02-12:
> And one more with fake endpoint:

Could you explain ?

pawelka, 2019-02-12:
"Ok I think I found the problem, did you get problem right after flashing the new firmware 3.0f ? and erasing EEProm ?" - YES

pawelka, 2019-02-12:
"Could you explain ?" - I did few HA restarts and saw the stack. Maybe I break discovery process, but I had weird situation with attribute reading 0x0100 - this reach in bracket number ~200, so it blocks any zigate operations, so I restart.

doudz, 2019-02-12:
ok  I understand. try the last update, I think it would be better

pawelka, 2019-02-12:
It's better. Thx !!!

closed 2019-02-13 by pawelka