doudz/zigate #100: Fast requests

pawelka, 2019-02-12:
After fast request enable/disable switch. Zigate no longer send response. I had to restart HA to start it working again. Then fast request and problem :-/. 

2019-02-12 22:57:56 DEBUG (SyncWorker_10) [zigate] REQUEST : 0x0092 b'\x02\xeb{\x01\x02\x00'
2019-02-12 22:57:56 DEBUG (SyncWorker_10) [zigate] Msg to send b'009200060502eb7b010200'
2019-02-12 22:57:56 DEBUG (SyncWorker_10) [zigate] Encoded Msg to send b'010210920210021602150212eb7b02110212021003'
2019-02-12 22:57:56 DEBUG (SyncWorker_10) [zigate] Waiting for status message for command 0x0092
2019-02-12 22:57:56 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x02\x19\x02\x10\x1e\x02\x10\x92\x02\x10\x03'
2019-02-12 22:57:56 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-02-12 22:57:56 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'001e0092'
2019-02-12 22:57:56 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:30, packet_type:146, error:b'', rssi:0
2019-02-12 22:57:56 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-02-12 22:57:56 DEBUG (SyncWorker_10) [zigate] STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:0, sequence:30, packet_type:146, error:b'', rssi:0
2019-02-12 22:57:56 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b"\x01\x81\x02\x12\x02\x10\x02\x1e'@\xeb{\x02\x12\x02\x10\x02\x16\x02\x10\x02\x10\x02\x10\x10"
2019-02-12 22:58:01 DEBUG (SyncWorker_6) [zigate] REQUEST : 0x0092 b'\x02\xeb{\x01\x02\x00'
2019-02-12 22:58:01 DEBUG (SyncWorker_6) [zigate] Msg to send b'009200060502eb7b010200'
2019-02-12 22:58:01 DEBUG (SyncWorker_6) [zigate] Encoded Msg to send b'010210920210021602150212eb7b02110212021003'
2019-02-12 22:58:01 DEBUG (SyncWorker_6) [zigate] Waiting for status message for command 0x0092
2019-02-12 22:58:04 WARNING (SyncWorker_6) [zigate] No response after command 0x0092 (1)
2019-02-12 22:58:05 DEBUG (SyncWorker_27) [zigate] REQUEST : 0x0092 b'\x02\xeb{\x01\x02\x00'
2019-02-12 22:58:05 DEBUG (SyncWorker_27) [zigate] Msg to send b'009200060502eb7b010200'
2019-02-12 22:58:05 DEBUG (SyncWorker_27) [zigate] Encoded Msg to send b'010210920210021602150212eb7b02110212021003'
2019-02-12 22:58:05 DEBUG (SyncWorker_27) [zigate] Waiting for status message for command 0x0092
2019-02-12 22:58:08 WARNING (SyncWorker_27) [zigate] No response after command 0x0092 (2)
2019-02-12 22:58:18 DEBUG (SyncWorker_7) [zigate] REQUEST : 0x0092 b'\x02\xeb{\x01\x02\x00'
2019-02-12 22:58:18 DEBUG (SyncWorker_7) [zigate] Msg to send b'009200060502eb7b010200'
2019-02-12 22:58:18 DEBUG (SyncWorker_7) [zigate] Encoded Msg to send b'010210920210021602150212eb7b02110212021003'
2019-02-12 22:58:18 DEBUG (SyncWorker_7) [zigate] Waiting for status message for command 0x0092
2019-02-12 22:58:21 WARNING (SyncWorker_7) [zigate] No response after command 0x0092 (3)
2019-02-12 22:58:43 DEBUG (SyncWorker_17) [zigate] REQUEST : 0x0092 b'\x02\xeb{\x01\x02\x00'
2019-02-12 22:58:43 DEBUG (SyncWorker_17) [zigate] Msg to send b'009200060502eb7b010200'
2019-02-12 22:58:43 DEBUG (SyncWorker_17) [zigate] Encoded Msg to send b'010210920210021602150212eb7b02110212021003'
2019-02-12 22:58:43 DEBUG (SyncWorker_17) [zigate] Waiting for status message for command 0x0092

pawelka, 2019-02-12:
Hmmm, after waiting long time it's unlocked, so maybe it's zigate stick firmware, not this library.

I'm using zigate over wifi with HA.

pawelka, 2019-02-12:
On question to confirm. Does after erase eeprom device list should be empty or old one? I assume empty, but something was returned in my case. I tried are eeprom, pdm from FlashGUI and JNxxxx production flasher and result is the same. Maybe I did something wrong during flashing to 3.0f

pawelka, 2019-02-12:
I moved closer the zigate to switch. Fast on/off doesn't block, but on previous firmware it was  the same place and it works :-/.

How to interpret the rssi value from logs. It should be from -40 to -100, but logs has possitive value. Is it correct that lower value is better?

I'm leaving this ticket open. If someone have some suggestion/comment, I'm waiting. If not, we can close this ticket.

doudz, 2019-02-13:
Erase eeprom the internal zigate memory so any paired device should be pair again.
(But it doesn't delete the content of zigate.json, and it's good)

rssi value has always been a positive value from 1 to 254, you get negative ??? 
And higher value is better, 254 you're close to zigate, 1 you are too far

doudz, 2019-02-13:
Since you're using wifi, take care on the wifi channel and zigbee channel you use to avoid conflict
[image]
https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence

pawelka, 2019-02-13:
I'll check channels at the evening, but I found that from 3.0f there is new command:
0x0806 | Set TX power (v30f uniquement) 

Did you consider to add it ?

pawelka, 2019-02-13:
I don't have negative RSSI value, but some theories about RSSI on the internet mainly mention negative value.

doudz, 2019-02-13:
I just discover that rssi for zigate is in fact lqi. rssi value is not sent

doudz, 2019-02-13:
https://github.com/fairecasoimeme/ZiGate/issues/151

doudz, 2019-02-13:
> I'll check channels at the evening, but I found that from 3.0f there is new command:
> 0x0806 | Set TX power (v30f uniquement)
> 
> Did you consider to add it ?

the command will be added soon

pawelka, 2019-02-13:
Thank you for interesting discussion. No more questions/issues. Closing.

closed 2019-02-13 by pawelka