doudz/zigate #108: Building network map

pdecat, 2019-02-19:
Hi @doudz, I was eager to try the new network map feature from 0.28 but it crashes with a stack overflow:

```
2019-02-19 13:21:10 DEBUG (SyncWorker_4) [zigate] REQUEST : 0x004e b'|\x0e\x00'
2019-02-19 13:21:10 DEBUG (SyncWorker_4) [zigate] Msg to send b'004e00033f7c0e00'
2019-02-19 13:21:10 DEBUG (SyncWorker_4) [zigate] Encoded Msg to send b'0102104e021002133f7c021e021003'
Fatal Python error: Cannot recover from stack overflow.

Thread 0x00007f423612aae8 (most recent call first):
  File "/usr/local/lib/python3.6/threading.py", line 299 in wait
  File "/usr/local/lib/python3.6/threading.py", line 551 in wait
  File "/usr/local/lib/python3.6/threading.py", line 1180 in run
  File "/usr/local/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/local/lib/python3.6/threading.py", line 884 in _bootstrap

[...]

Current thread 0x00007f423d5abae8 (most recent call first):
  File "/usr/local/lib/python3.6/site-packages/colorlog/colorlog.py", line 108 in format
  File "/usr/local/lib/python3.6/logging/__init__.py", line 839 in format
  File "/usr/local/lib/python3.6/logging/__init__.py", line 993 in emit
  File "/usr/local/lib/python3.6/logging/__init__.py", line 864 in handle
  File "/usr/local/lib/python3.6/logging/__init__.py", line 1515 in callHandlers
  File "/usr/local/lib/python3.6/logging/__init__.py", line 1453 in handle
  File "/usr/local/lib/python3.6/logging/__init__.py", line 1443 in _log
  File "/usr/local/lib/python3.6/logging/__init__.py", line 1295 in debug
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 694 in _wait_status
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 436 in send_data
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1055 in lqi_request
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1066 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
[... 28 lines cut ...]
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079 in build_neighbours_table
  ...
[...]
```

Note: this was invoked from HA.

doudz, 2019-02-19:
nice infinite loop...

doudz, 2019-02-19:
Could you provide debug log ?

pdecat, 2019-02-19:
There's this error above:

```
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/logging/__init__.py", line 993, in emit
    msg = self.format(record)
  File "/usr/local/lib/python3.6/logging/__init__.py", line 839, in format
    return fmt.format(record)
  File "/usr/local/lib/python3.6/site-packages/colorlog/colorlog.py", line 108, in format
    record = ColoredRecord(record)
RecursionError: maximum recursion depth exceeded
Call stack:
  File "/usr/local/lib/python3.6/threading.py", line 884, in _bootstrap
    self._bootstrap_inner()
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 69, in _worker
    work_item.run()
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/zigate/__init__.py", line 414, in build_network_map
    myzigate.build_network_map(hass.config.config_dir)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1091, in build_network_map
    table = self.build_neighbours_table()
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079, in build_neighbours_table
    n2 = self.build_neighbours_table(n['addr'])
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079, in build_neighbours_table
    n2 = self.build_neighbours_table(n['addr'])
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1079, in build_neighbours_table
    n2 = self.build_neighbours_table(n['addr'])
  [Previous line repeated 968 more times]
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1066, in build_neighbours_table
    r = self.lqi_request(addr, index, True)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1055, in lqi_request
    r = self.send_data(0x004e, data, wait_response=wait_response)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 436, in send_data
    status = self._wait_status(cmd)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 694, in _wait_status
    LOGGER.debug('Waiting for status message for command 0x{:04x}'.format(cmd))
Message: 'Waiting for status message for command 0x004e'
Arguments: ()
```

Let me redact my logs for sharing.

pdecat, 2019-02-19:
Here are the logs: https://gist.github.com/pdecat/22b751b8a4dcca12dcbf9aacc72c06e4

doudz, 2019-02-19:
thanks.
the network map can be more complex than I supposed, I have to fix that

pdecat, 2019-02-19:
This issue is likely caused by cycles in the network. This could be detected by passing the list of visited nodes in the recursive calls and return if a node was already visited as described in https://www.geeksforgeeks.org/detect-cycle-in-a-graph/

doudz, 2019-02-20:
I think it's ok now in dev branch, could you check with that branch ?

pdecat, 2019-02-20:
Network map generation in dev branch no longer crashes and generates the following:

```
2019-02-20 12:27:59 DEBUG (SyncWorker_0) [zigate] // ZiGate Network
graph zigate_network {
        node [shape=box]
        0000 [label="ZiGate (0000)" shape=doubleoctagon]
        "7c0e" [label="OSRAM Plug 01 (7c0e) 7cb03eaa00acf22e"]
        4093 [label="OSRAM Plug 01 (4093) 84182600000eda6a"]
        "4de2" [label="OSRAM Plug 01 (4de2) 84182600000ec7c9"]
        "899f" [label="OSRAM Plug 01 (899f) 7cb03eaa00aca562"]
        f757 [label="Device TRADFRI wireless dimmer (f757) 000b57fffe277a4b"]
        3356 [label="IKEA of Sweden TRADFRI bulb E27 opal 1000lm (3356) 000b57fffe141018"]
        "6ff0" [label="IKEA of Sweden TRADFRI bulb E27 WS opal 980lm (6ff0) 90fd9ffffe19cf16"]
        a43a [label="IKEA of Sweden TRADFRI bulb E27 WS opal 980lm (a43a) 90fd9ffffe6e7e71"]
        bd94 [label="IKEA of Sweden TRADFRI remote control (bd94) 90fd9ffffe8b4381"]
        "5bd6" [label="IKEA of Sweden TRADFRI bulb E27 W opal 1000lm (5bd6) 000b57fffea1ce3e"]
        "950e" [label="IKEA of Sweden TRADFRI bulb E27 W opal 1000lm (950e) 90fd9ffffe4d1adc"]
        ea13 [label="None TRADFRI motion sensor (ea13) 90fd9ffffe084427"]
        "6c31" [label="Device FNB56-ZSW23HG1.1 (6c31) 00158d00025dd2f4"]
        0000 -- 4093 [label=70]
        0000 -- "7c0e" [label=72]
        bd94 -- "899f" [label=70]
        "950e" -- ea13 [label=255]
        "4de2" -- bd94 [label=74]
        ea13 -- 4093 [label=16]
}
```

Note: the homeassistant docker container managed by hassio is lacking the graphviz executables:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/websocket_api/decorators.py", line 17, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/websocket_api/commands.py", line 148, in handle_call_service
    connection.context(msg))
  File "/usr/local/lib/python3.6/site-packages/homeassistant/core.py", line 1130, in async_call
    self._execute_service(handler, service_call))
  File "/usr/local/lib/python3.6/site-packages/homeassistant/core.py", line 1154, in _execute_service
    await self._hass.async_add_executor_job(handler.func, service_call)
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/zigate/__init__.py", line 414, in build_network_map
    myzigate.build_network_map(hass.config.config_dir)
  File "/config/deps/lib/python3.6/site-packages/zigate/core.py", line 1114, in build_network_map
    dot.render('zigate_network', cleanup=True)
  File "/config/deps/lib/python3.6/site-packages/graphviz/files.py", line 188, in render
    rendered = backend.render(self._engine, format, filepath, renderer, formatter)
  File "/config/deps/lib/python3.6/site-packages/graphviz/backend.py", line 183, in render
    run(cmd, capture_output=True, check=True, quiet=quiet)
  File "/config/deps/lib/python3.6/site-packages/graphviz/backend.py", line 150, in run
    raise ExecutableNotFound(cmd)
graphviz.backend.ExecutableNotFound: failed to execute ['fdp', '-Tpng', '-O', '/config/zigate_network'], make sure the Graphviz executables are on your systems' PATH
```

For test purposes, I've installed it by hand: `docker exec -ti homeassistant -- apk add --no-cache graphviz`.

And it produced the following image:

[image]

I've got an issue with fonts...

doudz, 2019-02-20:
https://dreampuf.github.io/GraphvizOnline/#graph%20zigate_network%20%7B%0A%20%20%20%20layout%3D%22fdp%22%3B%0A%20%20%20%20%20%20%20%20node%20%5Bshape%3Dbox%5D%0A%20%20%20%20%20%20%20%200000%20%5Blabel%3D%22ZiGate%20(0000)%22%20shape%3Ddoubleoctagon%5D%0A%20%20%20%20%20%20%20%20%227c0e%22%20%5Blabel%3D%22OSRAM%20Plug%2001%20(7c0e)%207cb03eaa00acf22e%22%5D%0A%20%20%20%20%20%20%20%204093%20%5Blabel%3D%22OSRAM%20Plug%2001%20(4093)%2084182600000eda6a%22%5D%0A%20%20%20%20%20%20%20%20%224de2%22%20%5Blabel%3D%22OSRAM%20Plug%2001%20(4de2)%2084182600000ec7c9%22%5D%0A%20%20%20%20%20%20%20%20%22899f%22%20%5Blabel%3D%22OSRAM%20Plug%2001%20(899f)%207cb03eaa00aca562%22%5D%0A%20%20%20%20%20%20%20%20f757%20%5Blabel%3D%22Device%20TRADFRI%20wireless%20dimmer%20(f757)%20000b57fffe277a4b%22%5D%0A%20%20%20%20%20%20%20%203356%20%5Blabel%3D%22IKEA%20of%20Sweden%20TRADFRI%20bulb%20E27%20opal%201000lm%20(3356)%20000b57fffe141018%22%5D%0A%20%20%20%20%20%20%20%20%226ff0%22%20%5Blabel%3D%22IKEA%20of%20Sweden%20TRADFRI%20bulb%20E27%20WS%20opal%20980lm%20(6ff0)%2090fd9ffffe19cf16%22%5D%0A%20%20%20%20%20%20%20%20a43a%20%5Blabel%3D%22IKEA%20of%20Sweden%20TRADFRI%20bulb%20E27%20WS%20opal%20980lm%20(a43a)%2090fd9ffffe6e7e71%22%5D%0A%20%20%20%20%20%20%20%20bd94%20%5Blabel%3D%22IKEA%20of%20Sweden%20TRADFRI%20remote%20control%20(bd94)%2090fd9ffffe8b4381%22%5D%0A%20%20%20%20%20%20%20%20%225bd6%22%20%5Blabel%3D%22IKEA%20of%20Sweden%20TRADFRI%20bulb%20E27%20W%20opal%201000lm%20(5bd6)%20000b57fffea1ce3e%22%5D%0A%20%20%20%20%20%20%20%20%22950e%22%20%5Blabel%3D%22IKEA%20of%20Sweden%20TRADFRI%20bulb%20E27%20W%20opal%201000lm%20(950e)%2090fd9ffffe4d1adc%22%5D%0A%20%20%20%20%20%20%20%20ea13%20%5Blabel%3D%22None%20TRADFRI%20motion%20sensor%20(ea13)%2090fd9ffffe084427%22%5D%0A%20%20%20%20%20%20%20%20%226c31%22%20%5Blabel%3D%22Device%20FNB56-ZSW23HG1.1%20(6c31)%2000158d00025dd2f4%22%5D%0A%20%20%20%20%20%20%20%200000%20--%204093%20%5Blabel%3D70%5D%0A%20%20%20%20%20%20%20%200000%20--%20%227c0e%22%20%5Blabel%3D72%5D%0A%20%20%20%20%20%20%20%20bd94%20--%20%22899f%22%20%5Blabel%3D70%5D%0A%20%20%20%20%20%20%20%20%22950e%22%20--%20ea13%20%5Blabel%3D255%5D%0A%20%20%20%20%20%20%20%20%224de2%22%20--%20bd94%20%5Blabel%3D74%5D%0A%20%20%20%20%20%20%20%20ea13%20--%204093%20%5Blabel%3D16%5D%0A%7D
it's not perfect looks like some links are missing

doudz, 2019-02-20:
To avoid using graphviz, I was thinking about generating a html page and use vivagraphjs to render the graph
and use lovelace card to display it
https://developers.home-assistant.io/docs/en/lovelace_custom_card.html

What do you think about it ?

doudz, 2019-02-20:
or better using https://www.home-assistant.io/components/panel_custom/

pdecat, 2019-02-20:
That would be awesome!

pdecat, 2019-02-20:
> it's not perfect looks like some links are missing
> [image]

That's probably because some devices were offline while I generated the map.

doudz, 2019-02-22:
closing, next in https://github.com/doudz/zigate/issues/111

closed 2019-02-22 by doudz