doudz/zigate #111: Network map troubles with lumi.ctrl_ln2.aq1

biau, 2019-02-21:
Hi Doudz,

Very excited with your new functionnality to help me understand my network.
It looks like the aqara wall switch act like routers but link is not done in network map.
So I have a lot of devices displayed isolated and I think it's because they use my wall switches to connect to the zigate.

Please find debug log here:
https://termbin.com/ab8r

Thanks very much

doudz, 2019-02-21:
it's a bit strange; the LUMI switch doesn't answer but the request has been sent succesfully
Example for 77ed
2019-02-21 12:35:50 DEBUG (SyncWorker_15) [zigate] 77ed is a router, search for children
2019-02-21 12:35:50 DEBUG (SyncWorker_15) [zigate] Search for children of 77ed
2019-02-21 12:35:50 DEBUG (SyncWorker_15) [zigate] Msg to send b'004e0003d777ed00'
2019-02-21 12:35:50 DEBUG (SyncWorker_15) [zigate] Waiting for status message for command 0x004e
2019-02-21 12:35:50 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:205, packet_type:78, error:b'', lqi:0
2019-02-21 12:35:50 DEBUG (SyncWorker_15) [zigate] STATUS code to command 0x004e:RESPONSE 0x8000 - Status response : status:0, sequence:205, packet_type:78, error:b'', lqi:0
2019-02-21 12:35:50 DEBUG (SyncWorker_15) [zigate] Waiting for message 0x804e
2019-02-21 12:35:53 WARNING (SyncWorker_15) [zigate] No response waiting command 0x804e
2019-02-21 12:35:53 ERROR (SyncWorker_15) [zigate] Failed to build neighbours table

biau, 2019-02-21:
Here's the log for 5 minutes after the service call:
https://termbin.com/az4v
I can see the four switch sending data after some time but no idea if it's the data your lib was waiting for.

doudz, 2019-02-21:
definitely it doesn't reply but I suspect that I badly decode the bit_field, maybe I reverse the bit order
I have to check

max5962, 2019-02-21:
Hello @doudz ,
First of all, big thanks for this new functionnality !
[image]
The network is really strange is smeems that all "router" are disconnected from the zigate.

doudz, 2019-02-23:
I made a change we will see in 0.28.2

closed 2019-02-23 by doudz