doudz/zigate #119: Error when attempting to remove device with missing ieee

ruimarinho, 2019-03-05:
Hi @doudz,

Just reporting a small error when attempting to remove a device when ieee is not available. Would it make sense to just call `cleanup_devices` internally?

```
2019-03-05 15:23:14 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.140383690686248] Error handling message: {'type': 'call_service', 'domain': 'zigate', 'service': 'remove_device', 'service_data': {'entity_id': 'zigate.6163'}, 'id': 31}
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/components/websocket_api/decorators.py", line 17, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/app/homeassistant/components/websocket_api/commands.py", line 147, in handle_call_service
    connection.context(msg))
  File "/usr/src/app/homeassistant/core.py", line 1133, in async_call
    self._execute_service(handler, service_call))
  File "/usr/src/app/homeassistant/core.py", line 1157, in _execute_service
    await self._hass.async_add_executor_job(handler.func, service_call)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/zigate/__init__.py", line 360, in remove_device
    myzigate.remove_device(addr)
  File "/config/deps/lib/python3.7/site-packages/zigate/core.py", line 914, in remove_device
    ieee = self._devices[addr]['ieee']
  File "/config/deps/lib/python3.7/site-packages/zigate/core.py", line 2515, in __getitem__
    return self.info[key]
KeyError: 'ieee'
```

After calling `zigate.cleanup_devices`:

```
2019-03-05 15:23:36 DEBUG (SyncWorker_13) [zigate] Dispatch ZIGATE_DEVICE_REMOVED
/config/custom_components/zigate/__init__.py:183: RuntimeWarning: coroutine 'EntityComponent.async_remove_entity' was never awaited
  component.async_remove_entity(entity.entity_id)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
2019-03-05 15:23:36 DEBUG (SyncWorker_13) [zigate] Dispatch ZIGATE_DEVICE_REMOVED
2019-03-05 15:23:36 DEBUG (SyncWorker_13) [zigate] Dispatch ZIGATE_DEVICE_REMOVED
2019-03-05 15:23:36 ERROR (SyncWorker_13) [zigate] IEEE is missing for 6163, please pair it again !
2019-03-05 15:23:36 ERROR (SyncWorker_13) [zigate] IEEE is missing for 6163, please pair it again !
2019-03-05 15:23:36 DEBUG (SyncWorker_13) [zigate] Dispatch ZIGATE_DEVICE_REMOVED
```

ruimarinho, 2019-06-02:
@doudz still getting this issue but without the runtime warning. Is there a recommended way of removing a device with a missing IEEE?

doudz, 2019-06-04:
Fixed in 0.29.10

closed 2019-06-04 by doudz

ruimarinho, 2019-08-23:
Using latest version still experiencing this. Cleanup devices does not work fully yet.

```
/config/custom_components/zigate/__init__.py:287: RuntimeWarning: coroutine 'EntityComponent.async_remove_entity' was never awaited
  component.async_remove_entity(entity.entity_id)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```