doudz/zigate #127: Difficultés avec la broker

ChoOo7, 2019-05-11:
Bonjour

Je suis face à un souci, possible de m'aider à y voir plus clair ?

Je viens de mettre à jour la lib avec pip3, mais je ne pense pas que le souci vienne de là, je précise juste que je suis à jour.

Je lance le broker avec
python3 -m zigate.mqtt_broker --device auto --mqtt_host localhost:1883

Et j'envoie un message avec
<?php
require(__DIR__."/include.php");
$payload = '{"function": "permit_join", "args": []}';
$mqtt->publish("zigate/command", $payload, 0);

Lorsque j'envoie le message, le broker explode en vol ! Voici la stack trace
INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB1
INFO:root:MQTT connected with result code 0
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/dist-packages/zigate/mqtt_broker.py", line 112, in <module>
    broker.start()
  File "/usr/local/lib/python3.5/dist-packages/zigate/mqtt_broker.py", line 41, in start
    self.client.loop_forever()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 1481, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 1003, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 1284, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 1849, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 2305, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 2500, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 2647, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/usr/local/lib/python3.5/dist-packages/zigate/mqtt_broker.py", line 86, in on_message
    'result': result
  File "/usr/local/lib/python3.5/dist-packages/zigate/mqtt_broker.py", line 45, in _publish
    payload = json.dumps(payload, cls=DeviceEncoder)
  File "/usr/lib/python3.5/json/__init__.py", line 237, in dumps
    **kw).encode(obj)
  File "/usr/lib/python3.5/json/encoder.py", line 198, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.5/json/encoder.py", line 256, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 2145, in default
    return json.JSONEncoder.default(self, obj)
  File "/usr/lib/python3.5/json/encoder.py", line 179, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: <zigate.responses.R8000 object at 0x7629fd70> is not JSON serializable

J'avais quelques soucis, je ne savais pas trop d'ou cela venait, donc j'ai tenté des resets, factory reset, erase persistant (le erase persistant ne me vidait pas le fichier /home/pi/.zigate)

Je l'ai fait à la main du coup, voici le contenu
pi@raspberrypi:/var/home/raspberry/zigate $ cat /home/pi/.zigate.json
{
    "devices": [],
    "groups": {},
    "scenes": {}
}

$mqtt->close();

Merci d'avance :)

doudz, 2019-05-15:
Il serait intéressant de refaire le test avec le log en DEBUG

ChoOo7, 2019-05-17:
Merci

root@raspberrypi:~# pip3 install zigate --upgrade
Collecting zigate
  Downloading https://files.pythonhosted.org/packages/7e/dc/f3aaa0e9f4a38a3b5d567a8463c5a9b22f06cb3b0c0d3dcba57b997e1335/zigate-0.29.5-py3-none-any.whl (60kB)
    100% |████████████████████████████████| 61kB 904kB/s
Requirement already up-to-date: RPi.GPIO in /usr/lib/python3/dist-packages (from zigate)
Requirement already up-to-date: pyserial in /usr/local/lib/python3.5/dist-packages (from zigate)
Requirement already up-to-date: pydispatcher in /usr/local/lib/python3.5/dist-packages (from zigate)
Requirement already up-to-date: bottle in /usr/local/lib/python3.5/dist-packages (from zigate)
Installing collected packages: zigate
  Found existing installation: zigate 0.29.4
    Uninstalling zigate-0.29.4:
      Successfully uninstalled zigate-0.29.4
Successfully installed zigate-0.29.5

J'ajoute

LOGGER.setLevel(logging.DEBUG)
Après 
LOGGER = logging.getLogger('zigate')
dans /usr/local/lib/python3.5/dist-packages/zigate/core.py

Je lance 
python3 -m zigate.mqtt_broker --device auto --mqtt_host localhost:1883

puis un reset : 
$payload = '{"function": "reset", "args": []}';
$mqtt->publish("zigate/command", $payload, 0);

sleep(3);

$payload = '{"function": "erase_persistent", "args": []}';
$mqtt->publish("zigate/command", $payload, 0);

puis un associate
$payload = '{"function": "permit_join"}';
$mqtt->publish("zigate/command", $payload, 0);

Et voila la sortie
root@raspberrypi:~# python3 -m zigate.mqtt_broker --device auto --mqtt_host localhost:1883
DEBUG:zigate:Try loading persistent file
DEBUG:zigate:Load success
INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB0
DEBUG:zigate:REQUEST : 0x0010
DEBUG:zigate:Msg to send b'0010000010'
DEBUG:zigate:Encoded Msg to send b'01021010021002101003'
DEBUG:zigate:Waiting for status message for command 0x0010
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x95\x02\x10\x02\x10\x02\x10\x10\x02\x10\x03\x01\x80\x10\x02\x10\x02\x15\x9a\x02\x10\x02\x11\x02\x13\x02\x1d\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000010'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'', lqi:0
DEBUG:zigate:Received response 0x8010: b'0001030d'
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8010 - Version list : major:1, installer:30d, lqi:0, version:3.0d
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0010:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'', lqi:0
DEBUG:zigate:Waiting for message 0x8010
DEBUG:zigate:Stop waiting, got message 0x8010
DEBUG:zigate:REQUEST : 0x0021 b'\x03\x18\xc8\x00'
DEBUG:zigate:Msg to send b'00210004f60318c800'
DEBUG:zigate:Encoded Msg to send b'0102102102100214f6021318c8021003'
DEBUG:zigate:Waiting for status message for command 0x0021
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xa4\x02\x10\x02\x10\x02\x10!\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000021'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:33, error:b'', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
[... 343 lines cut ...]
    rc = self._packet_handle()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 2305, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 2500, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.5/dist-packages/paho/mqtt/client.py", line 2647, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/usr/local/lib/python3.5/dist-packages/zigate/mqtt_broker.py", line 86, in on_message
    'result': result
  File "/usr/local/lib/python3.5/dist-packages/zigate/mqtt_broker.py", line 45, in _publish
    payload = json.dumps(payload, cls=DeviceEncoder)
  File "/usr/lib/python3.5/json/__init__.py", line 237, in dumps
    **kw).encode(obj)
  File "/usr/lib/python3.5/json/encoder.py", line 198, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.5/json/encoder.py", line 256, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 2145, in default
    return json.JSONEncoder.default(self, obj)
  File "/usr/lib/python3.5/json/encoder.py", line 179, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: <zigate.responses.R8000 object at 0x76221bd0> is not JSON serializable

En esperant que cela puisse t'aider
Merci

doudz, 2019-05-17:
Je n'ai réussi à reproduire le problème...
je n'utilise pas php mais j'ai testé avec mosquitto_pub et tout fonctionne bien
`mosquitto_pub -h 192.168.1.100 -t "zigate/command" -m '{"function": "permit_join"}'`

ChoOo7, 2019-05-17:
Je vais tacher de rajouter du debug pour voir ce qu'il n'arrive pas à serializer / de serialiser

Sinon une idée de comment reinitialiser le tout un peu plus ?

doudz, 2019-05-17:
Je vois bien ce qu'il se passe mais je comprends pas pourquoi ça ne se produit pas chez moi. J'ai ajouté un test pour mettre en évidence le défaut et je le corrige prochainement

doudz, 2019-05-18:
C'est corrigé dans la branche dev mais j'ai un souci avec Travis, je règle ça avant de publier la mise à jour

ChoOo7, 2019-05-26:
Avec la dernière version, je n'ai plus de soucis, j'ai pu associer avec succès mon capteur de temperature xiaomi. Tu peux donc fermer le ticket je pense.
Merci bcp et bonne soirée

closed 2019-05-27 by doudz