doudz/zigate #133: Use IEEE address mode

jBouyoud, 2019-06-14:
Hello, 

I'd like to use your library with IEEE address to avoid reconfigure my client each time short address change.

When I call : `zigate.action_onoff('20000000000000000', 1, 1)`

I have the following error : 
`Error calling function action_onoff with the following error : 'H' format requires 0 <= number <= 65535`

doudz, 2019-06-15:
I'm not sure zigate support IEEE address in actions
Maybe you could first retrieve the short address using zigate.get_device_from_ieee and then call action_onoff
```
device = z.get_device_from_ieee('0123456789abcdef')
z.action_onoff(device.addr, 1, 1)
```

jBouyoud, 2019-06-16:
Ok for the workaround. I already do that ;-)

IMHO something still be strange : 

* https://github.com/doudz/zigate/blob/master/zigate/core.py#L967
=> If not supported why an exception is not thrown ?

* https://github.com/doudz/zigate/blob/master/zigate/core.py#L981
=> As I understand, you already handle IEEE for bind/unbind request only.

If this is not an issue you can close this issue

doudz, 2019-06-17:
You're right

doudz, 2019-06-17:
I just push some changes to support IEEE on every command, could you give a try ? (dev banch)

jBouyoud, 2019-06-18:
I'll try as soon as I can and give you a feedback. Thanks

jBouyoud, 2019-06-19:
I'm testing though mqtt broker.
When I send : `{"function": "action_onoff", "args":["20918a000000f454", 1, 1]}`
I have the following log : 
```
DEBUG:zigate:REQUEST : 0x0092 b'\x03 \x91\x8a\x00\x00\x00\xf4T\x01\x01\x01'
DEBUG:zigate:Msg to send b'0092000c070320918a000000f454010101'
DEBUG:zigate:Encoded Msg to send b'010210920210021c0217021320918a021002100210f45402110211021103'
DEBUG:zigate:Waiting for status message for command 0x0092
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x11\x02\x16\x02\x10\x02\x10\x92\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'06000092'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:6, sequence:0, packet_type:146, error:b'', lqi:0
ERROR:zigate:Command 0x0092 failed Failed (ZigBee event codes) 6 : b''
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0092:RESPONSE 0x8000 - Status response : status:6, sequence:0, packet_type:146, error:b'', lqi:0
```
 and nothing happen.

And when I send : `{"function": "read_attribute_request", "args":["20918a000000f454", 1, 8, 0]}`
I have the following log : 
```
DEBUG:zigate:REQUEST : 0x0100 b'\x03 \x91\x8a\x00\x00\x00\xf4T\x01\x01\x00\x08\x00\x00\x00\x00\x01\x00\x00'
DEBUG:zigate:Msg to send b'01000014840320918a000000f4540101000800000000010000'
DEBUG:zigate:Encoded Msg to send b'010211021002101484021320918a021002100210f4540211021102100218021002100210021002110210021003'
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x81\x02\x15\x02\x10\x02\x11\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'05000100'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:5, sequence:0, packet_type:256, error:b'', lqi:0
ERROR:zigate:Command 0x0100 failed Stack already started (no new configuration accepted) : b''
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0100:RESPONSE 0x8000 - Status response : status:5, sequence:0, packet_type:256, error:b'', lqi:0
INFO:root:Publish zigate/command/result
```
and no update receive for this attribute.

No regressions with short network addresses.
Tell me if you need more information.

Regards

doudz, 2019-06-20:
as suspected, the zigate firmware doesn't handle IEEE address for command, only short address.
So I have to translate the address first.
I'll push some code soon

doudz, 2019-06-20:
I update the dev branch to support IEEE for action_onoff, could you try ?
if it works, I'll do the same for other commands

doudz, 2019-06-25:
@jBouyoud did you try ? I would like to push a new release including the mod

doudz, 2019-08-05:
Closing since no news, supposed it's working

closed 2019-08-05 by doudz