doudz/zigate #136: Firmware 3.1a - Error decoding response 0x8702

Jsl-1, 2019-07-16:
Hi @doudz,

Just update zigate to firmware 3.1a, and I have this problem : 

2019-07-16 09:06:16 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-07-16 09:06:16 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8702: b'd40101026eadb5'
2019-07-16 09:06:16 ERROR (ZiGate-Decode data) [zigate] Error decoding response 0x8702: b'd40101026eadb5'
2019-07-16 09:06:16 ERROR (ZiGate-Decode data) [zigate] Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 494, in decode_data
    response = RESPONSES.get(msg_type, Response)(value, lqi)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 33, in __init__
    self.decode()
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 1080, in decode
    Response.decode(self)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 95, in decode
    sdata, msg_data = self.__decode(fmt, keys, msg_data)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 106, in __decode
    sdata = OrderedDict(zip(keys, struct.unpack(fmt, data[:size])))
struct.error: unpack requires a buffer of 13 bytes

Maybe because of this : [https://github.com/fairecasoimeme/ZiGate/pull/209/commits/f2658d012da705c62fdf3c2ea646c0c7c988cafb](https://github.com/fairecasoimeme/ZiGate/pull/209/commits/f2658d012da705c62fdf3c2ea646c0c7c988cafb)

doudz, 2019-07-17:
Firmware 3.1a is not yet supported.
will be soon

Jsl-1, 2019-07-17:
Hi,
Ok, thank you ;)

doudz, 2019-07-18:
Fix in 0.31.0

closed 2019-07-18 by doudz