doudz/zigate #139: Unable to connect to a PiZiGate on RPI v4 with zigate package

stephane, 2019-07-28:
RPI v4 on Raspbian.
PiZiGate (bought in June 2019).
Latest zigate installed in a Python venv.

https://github.com/fairecasoimeme/PiZiGate-tools returns:

$ ./PiZiGate_test  /dev/serial0
```
RPI Model searching ...
+ RPI 4 detected
Verifying /boot/cmdline.txt ...
+ /boot/cmdline.txt seems to be OK
Verifying /boot/cmdline.txt ...
+ /boot/config.txt seems to be OK
Searching /dev/serial0...
+ Port : /dev/serial0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/serial0 ...
+ /dev/serial0 opened --> OK
+ Packet 01 02 10 10 02 10 02 10 10 03 sent --> OK
+ Packet received --> OK
size : 7
 80 bc a4 a4 0a 04 c4
```
I don't know if the received packet makes sense....

but the zigate python package (version 0.31.1) returns:

```
>>> import zigate
>>> z = zigate.connect(port='/dev/serial0', gpio=True)
/home/pi/homeassistant/lib/python3.7/site-packages/zigate/core.py:2139: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(27, GPIO.OUT)  # GPIO2
/home/pi/homeassistant/lib/python3.7/site-packages/zigate/core.py:2146: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(17, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)  # GPIO0
No response after command 0x0010 (1)
Failed to retrieve zigate firmware version
No response after command 0x0021 (2)
No response after command 0x0023 (3)
No response after command 0x0009 (4)
Failed to get network state
No response after command 0x0024 (5)
No response after command 0x0015 (6)
```

I have already set the GPIO with gpio commands in `rc.local`.
I checked HomeAssistant wasn't running to avoid concurrency issues.

doudz, 2019-07-29:
Are you sure that the port is /dev/serial0 ?
I suggest you to set port='auto'

stephane, 2019-07-29:
```
>>> z = zigate.connect(port="auto", gpio=True)
/home/pi/homeassistant/lib/python3.7/site-packages/zigate/core.py:2139: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(27, GPIO.OUT)  # GPIO2
/home/pi/homeassistant/lib/python3.7/site-packages/zigate/core.py:2146: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(17, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)  # GPIO0
ZiGate not found
ZiGate has not been found, please check configuration.
```

doudz, 2019-07-29:
is your user in gpio group ?
if it's pi , what's the result of `groups pi` ?

stephane, 2019-07-30:
pi : pi adm dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio

doudz, 2019-07-31:
Looks good
The warning means the port hasn't been freed. 
You should try to disable your command in rc.local, eventually rebooting your rpi

I've no RPI v4, so the code has been checked only on v2 and v3.

I'm not sure about v4 but for v3 we need to enable_uart and disable bluetooth (or set bluetooth using soft uart)
did you do that ?

> If you're using Rpi3, you can have some trouble trying to use PiZiGate. If needed, add the following line into config.txt (If you're using Hass.io you have to access that on the SD card directly. Simply plug it into your PC and edit it there. The config.txt is not accessible from your Hass.io system, you may need to open the SD card on a Windows or Linux system.):
> 
> dtoverlay=pi3-miniuart-bt
> enable_uart=1

stephane, 2019-08-03:
I disabled the command in rc.local, rebooted and it gives the same result (I tried zigate master too).
I enabled UART but you don't need to disable bluetooth with RPIv4:

https://zigate.fr/documentation/compatibilite-raspberry-pi-4-b/

doudz, 2019-08-04:
Sounds like I have to buy a Rpi4...

doudz, 2019-08-07:
My Rpi 4 is on the way...

doudz, 2019-08-08:
After test it works like a charm
What I did :
- add user to groups : gpio, dialout and i2c
- disable console from /boot/cmdline.txt by removing `console=serial0,115200`
- enable uart by adding `enable_uart=1` in /boot/config.txt
- manually config the port /dev/serial0
```
>>> import zigate
>>> z=zigate.connect('/dev/serial0', gpio=True)
>>> z.get_version()
OrderedDict([('major', 3), ('installer', '30f'), ('lqi', 0), ('version', '3.0f')])

```

That's all folks :)

stephane, 2019-08-10:
I checked again my settings and I still had the same errors but I upgraded my Raspian from testing to stable (because I supposed you ran the stable for your test):

```
>>> import zigate
>>> z = zigate.connect(port="/dev/serial0", gpio=True)
No response waiting command 0x8024
>>> z.get_version()
OrderedDict([('major', 3), ('installer', '30f'), ('lqi', 0), ('version', '3.0f')])
```
Great (expected the warning)!

Did you keep the `console=tty` in /boot/cmdline.txt?
Which zigate firmware do you use?

Thank you for your support.
Conclusion `testing` was not stable ;)

closed 2019-08-10 by stephane

doudz, 2019-08-10:
Fine !
I was using up to date raspbian buster stable and Zigate firmware 3.0f 

I have remove console=.... From cmdline.txt

stephane, 2019-08-10:
There were two 'console=' in my cmdline.txt (weird)!
I removed the one remaining and there is no more error about x8024.

stephane, 2019-08-10:
The response returned by PiZiGate-tools is now longer:

```
+ /dev/serial0 opened --> OK
+ Packet 01 02 10 10 02 10 02 10 10 03 sent --> OK
+ Packet received --> OK
size : 82
 01 80 24 00 0d 2b 01 00 00 00 15 8d 00 01 ed ea 16 0b 00 03 
01 80 06 00 02 85 01 00 03 
01 80 00 00 05 95 00 00 00 10 00 03 
01 80 10 00 05 9a 00 03 03 0f 00 03 
```
as I said to the author it will be nice to check the response in the tool to be sure the response is meanginful.