doudz/zigate #140: struct.error: unpack requires a buffer of 13 bytes

ruimarinho, 2019-08-01:
Hi @doudz. Using master code and 3.1a, when clicking the Aqara button switch, an error comes out.

```
2019-08-01 00:27:01 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update attribute for device LUMI lumi.sensor_switch.aq2 (bc08) 00158d000204a1c4 {'endpoint': 1, 'cluster': 0, 'addr': 'bc08', 'attribute': 5, 'data': 'lumi.sensor_switch.aq2', 'name': 'type', 'type': <class 'str'>, 'value': 'lumi.sensor_switch.aq2'}
2019-08-01 00:27:01 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-08-01 00:27:01 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-08-01 00:27:02 WARNING (ZiGate-Decode data) [zigate] No response waiting command 0x8120
2019-08-01 00:27:02 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED
2019-08-01 00:27:02 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x11\x02\x10\x02\x17v\x02\x10\x02\x11\x02\x10\x02\x16\x02\x16\x82r\x03'
2019-08-01 00:27:03 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update device LUMI lumi.sensor_switch.aq2 (bc08) 00158d000204a1c4
2019-08-01 00:27:03 ERROR (ZiGate-Decode data) [zigate] Traceback (most recent call last):
  File "/config/deps/lib/python3.7/site-packages/zigate/core.py", line 491, in decode_data
    response = RESPONSES.get(msg_type, Response)(value, lqi)
  File "/config/deps/lib/python3.7/site-packages/zigate/responses.py", line 33, in __init__
    self.decode()
  File "/config/deps/lib/python3.7/site-packages/zigate/responses.py", line 1080, in decode
    Response.decode(self)
  File "/config/deps/lib/python3.7/site-packages/zigate/responses.py", line 95, in decode
    sdata, msg_data = self.__decode(fmt, keys, msg_data)
  File "/config/deps/lib/python3.7/site-packages/zigate/responses.py", line 106, in __decode
    sdata = OrderedDict(zip(keys, struct.unpack(fmt, data[:size])))
struct.error: unpack requires a buffer of 13 bytes
```

doudz, 2019-08-01:
I don't yet fully support the firmware 3.a1 , soon

doudz, 2019-08-05:
I can't reproduce the problem, what lib version are you using ?
Maybe I've already fixed the problem

doudz, 2019-08-05:
I think you cut too much the log, the right "Raw packet received" line is missing

ruimarinho, 2019-08-05:
I was using `0.31.0`. Isn't line 6 what you're asking for?

`2019-08-01 00:27:02 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x11\x02\x10\x02\x17v\x02\x10\x02\x11\x02\x10\x02\x16\x02\x16\x82r\x03'
`

ruimarinho, 2019-08-05:
Ah, got you. I will scrub through logs but at this time I don't think I will be able to recover it. My bad.

I'll re-open this ticket if I find the issue again and then report the right packet line.

closed 2019-08-05 by ruimarinho