doudz/zigate #143: Unable to remove IKEA Signal Repeater

ruimarinho, 2019-08-25:
Hi @doudz,

It looks like I am unable to remove this device from the network. When I require `zigate.remove_device`, a message is sent but even after physically clicking once on the repeater, the device remains on the network:

```
2019-08-25 13:04:39 DEBUG (SyncWorker_7) [zigate] REQUEST : 0x0026 b'\x00\x15\x8d\x00\x01\xedF\xb0\x00\ro\xff\xfe\x9f*\xab'
2019-08-25 13:04:39 DEBUG (SyncWorker_7) [zigate] Msg to send b'00260010c900158d0001ed46b0000d6ffffe9f2aab'
2019-08-25 13:04:39 DEBUG (SyncWorker_7) [zigate] Encoded Msg to send b'01021026021010c90210158d02100211ed46b00210021d6ffffe9f2aab03'
2019-08-25 13:04:39 DEBUG (SyncWorker_7) [zigate] Waiting for status message for command 0x0026
2019-08-25 13:04:39 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xa3\x02\x10\x02\x10\x02\x10&\x02\x10\x03'
2019-08-25 13:04:39 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-08-25 13:04:39 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00000026'
2019-08-25 13:04:39 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:38, error:b'', lqi:0
2019-08-25 13:04:39 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-08-25 13:04:39 DEBUG (SyncWorker_7) [zigate] STATUS code to command 0x0026:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:38, error:b'', lqi:0
```

Also tried to hold for 5 seconds but zigate never interprets the device removal. What am I doing wrong?

ruimarinho, 2019-11-01:
@doudz one thing I've noticed during the pairing process is the presence of several `0x0140` errors. Could this be the reason why pairing/unpairing is so problematic with this device?

```
2019-11-01 12:46:31 DEBUG (SyncWorker_0) [zigate] STATUS code to command 0x0049:RESPONSE 0x8000 - Status response : status:0, sequence:153, packet_type:73, error:b'', lqi:0

2019-11-01 12:46:32 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x02\x10M\x02\x10\x02\x1cP\xb5W\x02\x10\x02\x1do\xff\xfe\x9f*\xab\x8e\x02\x10\x03'

2019-11-01 12:46:32 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-11-01 12:46:32 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'b557000d6ffffe9f2aab8e'
2019-11-01 12:46:32 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:b557, ieee:000d6ffffe9f2aab, mac_capability:10001110, lqi:0
2019-11-01 12:46:32 DEBUG (ZiGate-Decode data) [zigate] Device Announce RESPONSE 0x004D - Device announce : addr:b557, ieee:000d6ffffe9f2aab, mac_capability:10001110, lqi:0
2019-11-01 12:46:32 WARNING (ZiGate-Decode data) [zigate] Device already exists with another addr f03d, rename it.

2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Acquire Lock on device Device  (f03d) 000d6ffffe9f2aab
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Release Lock on device Device  (b557) 000d6ffffe9f2aab
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_ADDRESS_CHANGED
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update device Device  (b557) 000d6ffffe9f2aab
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] discover_device b557
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] no endpoints
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] REQUEST : 0x0045 b557
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Msg to send b'00450002a5b557'
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Encoded Msg to send b'0102104502100212a5b55703'
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Waiting for status message for command 0x0045

2019-11-01 12:46:33 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15Z\x02\x10\x9a\x02\x10E\x02\x10\x03'
2019-11-01 12:46:33 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'009a0045'
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:154, packet_type:69, error:b'', lqi:0
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] STATUS code to command 0x0045:RESPONSE 0x8000 - Status response : status:0, sequence:154, packet_type:69, error:b'', lqi:0
2019-11-01 12:46:33 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED

2019-11-01 12:46:36 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x02\x10M\x02\x10\x02\x1cP\xb5W\x02\x10\x02\x1do\xff\xfe\x9f*\xab\x8e\x02\x10\x03'
2019-11-01 12:46:36 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x02\x10M\x02\x10\x02\x1c\xcf\xb5W\x02\x10\x02\x1do\xff\xfe\x9f*\xab\x8e\x9f\x03\x01\x87\x02\x11\x02\x10\x02\x13\x85\x02\x10\x02\x10\x02\x10\x03'
2019-11-01 12:46:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'b557000d6ffffe9f2aab8e'
2019-11-01 12:46:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:b557, ieee:000d6ffffe9f2aab, mac_capability:10001110, lqi:0
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'b557000d6ffffe9f2aab8e'
2019-11-01 12:46:36 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Device Announce RESPONSE 0x004D - Device announce : addr:b557, ieee:000d6ffffe9f2aab, mac_capability:10001110, lqi:0
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:b557, ieee:000d6ffffe9f2aab, mac_capability:10001110, lqi:159
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Device Announce RESPONSE 0x004D - Device announce : addr:b557, ieee:000d6ffffe9f2aab, mac_capability:10001110, lqi:159
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Acquire Lock on device Device  (b557) 000d6ffffe9f2aab
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Acquire Lock on device Device  (b557) 000d6ffffe9f2aab
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8701: b'0000'
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Release Lock on device Device  (b557) 000d6ffffe9f2aab
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8701 - Route Discovery Confirmation : status:0, network_status:0, lqi:0
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Release Lock on device Device  (b557) 000d6ffffe9f2aab
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-11-01 12:46:36 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update device Device  (b557) 000d6ffffe9f2aab
[... cut ...]
2019-11-01 12:46:43 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x02\x1f\xb9\x13\xf1(\x02\x11\x02\x14\x02\x12\x02\x10\x02\x10\x02\x10)\x02\x10\x02\x12\x02\x18\x8fT\x03\x01\x81\x02\x12\x02\x10\x02\x1fM\x14\xf1(\x02\x11\x02\x14\x02\x15\x02\x10\x02\x10\x02\x10!\x02\x10\x02\x12\x1beQ\x03\x01\x81\x02\x12\x02\x10\x02\x1f\xe5\x15\xf1(\x02\x11\x02\x14\x02\x13\x02\x10\x02\x10\x02\x10)\x02\x10\x02\x12\x02\x13\xdfT\x03\x01\x81\x02\x12\x02\x10\x02\x1e\xd1\x15\xf1(\x02\x11\x02\x14\x02\x13\x02\x10\x14\x02\x10(\x02\x10\x02\x11\xffT\x03\x01\x81\x02\x12\x02\x10\x02\x1f\xb5\x15\xf1(\x02\x11\x02\x14\x02\x13\x02\x10\x10\x02\x10)\x02\x10\x02\x12&\xbaT\x03'
```

doudz, 2019-12-21:
It's now easier using new admin panel

closed 2019-12-21 by doudz