doudz/zigate #152: [zigate] Command 0x0100 failed Failed (ZigBee event codes) 21 : b''

pawelka, 2019-12-11:
Opening new one, because the same problem is observer with firmware 3.1a and latest library.

Someone else noticed the same issue with 3.1a firmware in https://github.com/doudz/zigate/issues/82, but ticket is closed and I'm not sure if closed tickets are monitored.

doudz, 2019-12-21:
Is it better with last release ?

pawelka, 2019-12-25:
only message changed:

> 2019-12-25 21:28:48 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:48 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:48 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:48 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:49 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:50 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:50 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''
2019-12-25 21:28:50 ERROR (ZiGate-Decode data) [zigate] Command 0x0100 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b''

pawelka, 2020-01-03:
Is it possible to add to the error information about related device?