doudz/zigate #164: Network map is buggy

doudz, 2020-01-20:
Since 0.38.0 Network map is buggy (for me) many connections missing

doudz, 2020-01-20:
@pdecat I revert what you've done because it doesn't work at all for me

pdecat, 2020-01-21:
Hi, what kind of issue did you have? Did you get timeouts?

doudz, 2020-01-21:
Using your code, only 3 or 4 links are display but using mine, there's a lot more
see below
[image]

MichaelBitard, 2020-04-21:
Using version 0.38.9 I think it's missing a lot of links too.
Can I help on this?

[image]

hvanoch, 2020-04-21:
Having the same issue of missing links.
[image]

BriceDet, 2020-09-07:
Same error 
with jeedom log :
```file = http://xx.xx.xx.xx/index.php?v=d&m=zigate&p=zigate line 2 > injectedScript
line = 99
Message = TypeError: eqLs[element[1]] is undefined
```

komega13, 2021-01-24:
Hi !
I got same bug, only one link is showing up on my network map .
[image]