doudz/zigate #170: Error decoding some 0x8030 responses "struct.error: unpack requires a buffer of 7 bytes"

pdecat, 2020-01-26:
Since I upgraded my Zigate's firmware to v3.1b and afterward v3.1c, I often see the following error in the logs:

```
2020-01-26 11:18:39 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8030: b'd400020ae8'
2020-01-26 11:18:39 ERROR (ZiGate-Decode data) [zigate] Error decoding response 0x8030: b'd400020ae8'
2020-01-26 11:18:33 ERROR (ZiGate-Decode data) [zigate] Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 586, in decode_data
    response = RESPONSES.get(msg_type, Response)(value, lqi)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 36, in __init__
    self.decode()
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 449, in decode
    Response.decode(self)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 98, in decode
    sdata, msg_data = self._decode(fmt, keys, msg_data)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 109, in _decode
    sdata = OrderedDict(zip(keys, struct.unpack(fmt, data[:size])))
struct.error: unpack requires a buffer of 7 bytes
```

Not sure what is affected by these errors yet.

This is with v0.38.5:
[image]

pdecat, 2020-01-26:
Just got another one:

```
2020-01-26 11:26:44 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8030: b'300002a43a'
2020-01-26 11:26:44 ERROR (ZiGate-Decode data) [zigate] Error decoding response 0x8030: b'300002a43a'
2020-01-26 11:26:44 ERROR (ZiGate-Decode data) [zigate] Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 586, in decode_data
    response = RESPONSES.get(msg_type, Response)(value, lqi)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 36, in __init__
    self.decode()
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 449, in decode
    Response.decode(self)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 98, in decode
    sdata, msg_data = self._decode(fmt, keys, msg_data)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 109, in _decode
    sdata = OrderedDict(zip(keys, struct.unpack(fmt, data[:size])))
struct.error: unpack requires a buffer of 7 bytes
```

pdecat, 2020-01-26:
One more:

```
2020-01-26 11:28:50 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8030: b'3200026ff0'
2020-01-26 11:28:50 ERROR (ZiGate-Decode data) [zigate] Error decoding response 0x8030: b'3200026ff0'
2020-01-26 11:28:50 ERROR (ZiGate-Decode data) [zigate] Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigate/core.py", line 586, in decode_data
    response = RESPONSES.get(msg_type, Response)(value, lqi)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 36, in __init__
    self.decode()
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 449, in decode
    Response.decode(self)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 98, in decode
    sdata, msg_data = self._decode(fmt, keys, msg_data)
  File "/usr/local/lib/python3.7/site-packages/zigate/responses.py", line 109, in _decode
    sdata = OrderedDict(zip(keys, struct.unpack(fmt, data[:size])))
struct.error: unpack requires a buffer of 7 bytes
```

doudz, 2020-01-27:
According to doc https://zigate.fr/documentation/commandes-zigate/ the response 0x8030 should include the cluster ID but it looks missing in your samples
```
32 00 02 6ff0
32 : sequence
00 : status
02 : address mode
6ff0 : address
... : cluster ?
```

fairecasoimeme, 2020-01-27:
Hi,
sorry, the pb is in the doc. I just fixed it.
The cluster is not required because it's is always the same (bind or unbin cluster)

Fred

closed 2020-01-27 by doudz

pdecat, 2020-01-27:
Thanks @fairecasoimeme @doudz !

pdecat, 2020-01-27:
Can confirm the issue is resolved with latest lib version:

```
2020-01-27 12:38:42 DEBUG (SyncWorker_9) [zigate] Waiting for message 0x8030
2020-01-27 12:38:42 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x800\x02\x10\x02\x16Q\x98\x02\x10\x02\x12\x02\x1a\xe8\x9f\x03'
2020-01-27 12:38:42 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-01-27 12:38:42 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8030: b'9800020ae8'
2020-01-27 12:38:42 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8030 - Bind response : sequence:152, status:0, address_mode:2, addr:0ae8, lqi:159
2020-01-27 12:38:42 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-01-27 12:38:42 DEBUG (SyncWorker_9) [zigate] Stop waiting, got message 0x8030
```

commit c77e73e5 (2020-01-27): Fix #170