doudz/zigate #172: Error demon Zigate

Theandone90, 2020-01-30:
J'ai une Zigate dans le dernier fw, mais impossible de lancer de Demon sur mon Jeedom.
Je suis en V3.3.39

La version du plugin Zigate est la 1.4.1 et la clé est bien vue mais impossible de lancer le Démon
Voici mes logs.

[2020-01-30 08:30:04][ERROR] : Attention je pense qu'il y a un soucis avec le démon que j'ai relancé plus de 3 fois consecutivement
[2020-01-30 08:30:05][INFO] : Lancement démon zigate : /usr/bin/python3 /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py  --device /dev/ttyUSB0 --gpio 0 --loglevel debug --apikey hkJ9ptHP1VgTMqUuKwmdI4RFNghkK2cL --pid /tmp/jeedom/zigate/daemon.pid --socket /tmp/jeedom/zigate/daemon.sock --callback http://127.0.0.1:80/plugins/zigate/core/php/jeeZiGate.php --sharedata 0 --channel 11
[2020-01-30 07:30:06][INFO][root](MainThread) : Start zigated
[2020-01-30 07:30:06][INFO][root](MainThread) : Log level : debug
[2020-01-30 07:30:06][INFO][root](MainThread) : Socket : /tmp/jeedom/zigate/daemon.sock
[2020-01-30 07:30:06][INFO][root](MainThread) : PID file : /tmp/jeedom/zigate/daemon.pid
[2020-01-30 07:30:06][INFO][root](MainThread) : Apikey : hkJ9ptHP1VgTMqUuKwmdI4RFNghkK2cL
[2020-01-30 07:30:06][INFO][root](MainThread) : Device : /dev/ttyUSB0
[2020-01-30 07:30:06][INFO][root](MainThread) : Callback : http://127.0.0.1:80/plugins/zigate/core/php/jeeZiGate.php
[2020-01-30 07:30:06][INFO][root](MainThread) : Python version : 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516]
[2020-01-30 07:30:06][INFO][root](MainThread) : zigate version : 0.33.1
[2020-01-30 07:30:06][INFO][root](MainThread) : Plugin Abeille détecté, vérification de l'état
[2020-01-30 07:30:06][INFO][root](MainThread) : Ok, le plugin Abeille ne semble pas en fonctionnement.
[2020-01-30 07:30:06][DEBUG][root](MainThread) : Writing PID 4848 to /tmp/jeedom/zigate/daemon.pid
[2020-01-30 07:30:06][DEBUG][root](MainThread) : Send to test to jeedom
[2020-01-30 07:30:06][DEBUG][root](MainThread) : Send to jeedom :  {'action': 'test'}
[2020-01-30 07:30:07][DEBUG][requests.packages.urllib3.connectionpool](MainThread) : Starting new HTTP connection (1): 127.0.0.1
[2020-01-30 08:30:07][DEBUG] : Array (     [action] => test )
[2020-01-30 07:30:07][DEBUG][requests.packages.urllib3.connectionpool](MainThread) : http://127.0.0.1:80 "POST /plugins/zigate/core/php/jeeZiGate.php?apikey=hkJ9ptHP1VgTMqUuKwmdI4RFNghkK2cL HTTP/1.1" 200 16
[2020-01-30 07:30:07][DEBUG][root](MainThread) : Jeedom reply :  {'success': True}
[2020-01-30 07:30:07][INFO][root](MainThread) : Démarrage ZiGate /dev/ttyUSB0
[2020-01-30 07:30:07][DEBUG][zigate](MainThread) : Try loading persistent file
[2020-01-30 07:30:07][DEBUG][zigate](MainThread) : Trying to load /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json
[2020-01-30 07:30:07][WARNING][zigate](MainThread) : Persistent file /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json doesn't exist
[2020-01-30 07:30:07][DEBUG][zigate](MainThread) : Trying to load /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json.0
[2020-01-30 07:30:07][WARNING][zigate](MainThread) : Persistent file /var/www/html/plugins/zigate/core/class/../../resources/zigated/zigate.json.0 doesn't exist
[2020-01-30 07:30:07][DEBUG][zigate](MainThread) : No file to load
[2020-01-30 07:30:07][ERROR][zigate](MainThread) : Cannot connect to ZiGate using port /dev/ttyUSB0
Traceback (most recent call last):
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 208, in reconnect
self.serial = self.initSerial()
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 200, in initSerial
return serial.Serial(self._port, 115200)
AttributeError: module 'serial' has no attribute 'Serial'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 329, in <module>
z.autoStart(args.channel)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 339, in autoStart
self.startup(channel)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 356, in startup
self.setup_connection()
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 221, in setup_connection
self.connection = ThreadSerialConnection(self, self._port)
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 192, in __init__
self.reconnect(False)
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 217, in reconnect
raise ZIGATE_CANNOT_CONNECT('Cannot connect to ZiGate using port {}'.format(self._port))
zigate.transport.ZIGATE_CANNOT_CONNECT: Cannot connect to ZiGate using port /dev/ttyUSB0
[2020-01-30 08:30:13][ERROR] : Impossible de lancer le démon zigate, relancer le démon en debug et vérifiez la log

doudz, 2020-01-30:
C'est "normal", les dernières versions du plugin sont uniquement compatible avec Jeedom V4, il faudrait donc migrer

Theandone90, 2020-01-30:
Même avec la version du plugin en 1.4.1 et le fw de la clé en 3.1a cela ne fonctionne pas ?

doudz, 2020-01-30:
En réalité, le problème indiqué est tout autre, la commande suivante devrait résoudre le problème
`sudo pip3 install pyserial`
Toutefois la version 1.4.1 ne gère pas correctement le firmware 3.1a donc il y a aura des problèmes
La dernière version du plugin est 1.6.4, compatible uniquement Jeedom V4

Theandone90, 2020-01-30:
D'accord.
je vais voir pour migrer en V4.
Mais en V4 je trouve la partie Widget moins bien que la V3.

Theandone90, 2020-01-30:
je viens de passer en v4 mais toujours la même erreur.

doudz, 2020-01-31:
Merci d'essayer la commande sudo pip3 install pyserial

Et si ça ne fonctionne pas sudo pip3 uninstall pyserial

doudz, 2020-06-30:
Mauvais endroit, merci d'utiliser https://github.com/jeedom-zigate/jeedom-plugin-zigate/issues

closed 2020-06-30 by doudz