doudz/zigate #176: Fast change detected

biau, 2020-02-25:
Since latest update my debug log is flooded with messages like this:

> 2020-02-25 07:37:01 DEBUG (Thread-3233908) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': True}
> 2020-02-25 07:37:01 DEBUG (Thread-3233909) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': True}
> 2020-02-25 07:37:01 DEBUG (Thread-3233910) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233911) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233912) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233913) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233914) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233915) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233916) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': True}
> 2020-02-25 07:37:01 DEBUG (Thread-3233917) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233918) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233919) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': True}
> 2020-02-25 07:37:01 DEBUG (Thread-3233920) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233922) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233921) [zigate] Fast change detected, delay it for 1 6 {'attribute': 0, 'data': False}
> 2020-02-25 07:37:01 DEBUG (Thread-3233923) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': True}
> 2020-02-25 07:37:01 DEBUG (Thread-3233924) [zigate] Fast change detected, delay it for 2 6 {'attribute': 0, 'data': False}
It could be comming because of my xiaomi micro module https://zigate.fr/produits-xiaomi-compatibles-zigate/compatible/micro-modulexiaomi/ because he doesn't transmit switch state change anymore.

doudz, 2020-02-25:
looks like my new feature doesn't work at all... :thinking:

doudz, 2020-02-25:
Please update to disable the feature

biau, 2020-02-25:
Thanks for the very fast update (I was searching how to rollback with hacs and without release in github but you're faster !)
Problem solved.

closed 2020-02-25 by biau