doudz/zigate #177: Intégration des capteurs Xiaomi Mijia : lumi.sen_ill.mgl01

Smaky07, 2020-02-28:
J'ai des capteurs de luminosité Xiaomi MIJIA lumi.sen_ill.mgl01 qui ne semblent pas pris en charge par la librairie et donc par jeedom au final.

J'arrive à les inclures, certaines choses sont détectées, des commandes sont créées, mais ensuite plus rien n'est mis à jour et il n'y a plus de communication et de remontée d'information.

Si je peux aider à intégrer ces capteurs j'en serais ravis.

Le capteur est le suivant GZCGQ01LM et il est en Zigbee 3

A force de recherche, la clef zigate semble capable de les gérer car déja intégré dans Domoticz visiblement (j'espère ne pas trop m'avancé la dessus).

doudz, 2020-02-28:
est-ce que je pourrais avoir l'extrait du zigate.json pour voir comment est reconnu le capteur ?

doudz, 2020-02-28:
a priori c'est compatible https://zigate.fr/produits-xiaomi-compatibles-zigate/compatible/capteurdeluminosit/

doudz, 2020-02-28:
idéalement il faut lancer le démon Jeedom en mode debug pour tenter de voir ce qui se passe dans le log

Smaky07, 2020-02-28:
Voici le json 
et les logs en mode debug.

Merci
[zigate.txt](https://github.com/doudz/zigate/files/4268114/zigate.txt)
[zigate.json.txt](https://github.com/doudz/zigate/files/4268119/zigate.json.txt)

doudz, 2020-03-05:
Tout semble bien fonctionner sauf qu'il n'y a pas de trace d'une variation de lumière dans le log.
Il faudrait faire l'essai, toujours en debug, de faire varier la lumière pour forcer le capteur à envoyer à la zigate une information

Smaky07, 2020-03-05:
Dans les logs, il y a le moment ou j'ai éclairé le module avec une torche et ou je l'ai mis dans le noir.
Je vois aussi aucune remontée d'information. 
Je retente dés que j'en ai l'occasion.

doudz, 2020-03-06:
Il est nécessaire de relancer le démon lorsqu'on passe le niveau de log sur debug car là dans le log je ne vois pas d'information du démon donc impossible de savoir si la Zigate a reçu qqc ou pas

Smaky07, 2020-03-06:
Je m'y remets ce soir. Mais le plugin est en debug depuis le début. Il y a
même eu un redémarrage complet de la machine. Peut etre une erreur
néanmoins de ma part.
Je continue de creuser.
J'ai même tenter du côté du plugin abeille mais le fonctionnement n'est pas
le même et je n'ai pas le temps de tout recommencer a 0 pour refaire les
inclusions et revoir les scénarios, ne serait-ce que pour vérifier la prise
en compte de l'equipement.

Merci.

rroblik, 2021-05-26:
Hello @doudz 

Depuis le temps je pense (et je constate) que le module est plutot bien supporté mais les valeurs sont mises à jour très fréquemment sans action de ma part : 
```log
[2021-05-26 11:14:23]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33331
[2021-05-26 11:14:38]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33262
[2021-05-26 11:14:43]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33229
[2021-05-26 11:14:48]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33325
[2021-05-26 11:14:53]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33381
[2021-05-26 11:14:58]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33405
[2021-05-26 11:15:03]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33311
[2021-05-26 11:15:08]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33137
[2021-05-26 11:15:13]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33097
[2021-05-26 11:15:28]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33482
[2021-05-26 11:15:33]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33166
[2021-05-26 11:15:38]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33233
[2021-05-26 11:15:43]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33204
[2021-05-26 11:15:53]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33133
[2021-05-26 11:15:58]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33105
[2021-05-26 11:16:08]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33210
[2021-05-26 11:16:13]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33256
[2021-05-26 11:16:23]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33214
[2021-05-26 11:16:28]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33235
[2021-05-26 11:16:33]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33355
[2021-05-26 11:16:37]INFO : Evènement sur la commande [Extérieur][Lumi 01][luminosity] valeur : 33409
```

En soit pas de soucis c'est même peut-être le fonctionnement nominal ? Je m'interroge sur la durée de vie de la pile en faite.

Sinon j'ai aussi pas mal de commande "fantômes" qui ont été créées lors de l'inclusion, toutes préfixées par `1_` :
[image]

Merci