doudz/zigate #182: Erreur de dépendance

durdur33, 2020-04-28:
Bonjour
Depuis 2 jours je n'arrive plus a installer les dépendances du plugin Zigate
Jeedom 4.052 sur RPI 3B+
Zigate USB
pluggin zigate 1.64

merci par avance de votre aide
Didier 

voici les log:

Installation des dépendances
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_12.x stretch InRelease
Fetched 25.4 kB in 1s (14.2 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-serial is already the newest version (3.2.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-usb is already the newest version (1.0.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-requests is already the newest version (2.12.4-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-setuptools is already the newest version (33.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-rpi.gpio is already the newest version (0.6.5~stretch-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-dev is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: zigate==0.38.6 in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: pyserial>=3.2 in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pyusb in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: bottle in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pydispatcher>=2.0.5 in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Installation des dépendances terminée !
Vérification de la version installée
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/usr/local/lib/python3.5/dist-packages/zigate/__init__.py", line 9, in <module>
from .core import (ZiGate, ZiGateWiFi, ZiGateGPIO)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 16, in <module>
from .transport import (ThreadSerialConnection,
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 12, in <module>
import serial.tools.list_ports
ImportError: No module named 'serial.tools'

[image]

doudz, 2020-04-28:
Il faudrait essayer de lancer la commande suivante 
sudo apt-get remove python3-serial

Puis de relancer l'installation des dépendances

durdur33, 2020-04-28:
manipulation  effectuée
suppression de python3-serial ==> OK
reboot du raspberry
lancement  des dépendances  toujours même message

A+
Didier

Installation des dépendances
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:3 https://deb.nodesource.com/node_12.x stretch InRelease
Fetched 25.4 kB in 2s (12.3 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-serial is already the newest version (3.2.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-usb is already the newest version (1.0.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-requests is already the newest version (2.12.4-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-setuptools is already the newest version (33.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-rpi.gpio is already the newest version (0.6.5~stretch-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-dev is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: zigate==0.38.6 in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: pyserial>=3.2 in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pyusb in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pydispatcher>=2.0.5 in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: bottle in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Installation des dépendances terminée !
Vérification de la version installée
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/usr/local/lib/python3.5/dist-packages/zigate/__init__.py", line 9, in <module>
from .core import (ZiGate, ZiGateWiFi, ZiGateGPIO)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 16, in <module>
from .transport import (ThreadSerialConnection,
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 12, in <module>
import serial.tools.list_ports
ImportError: No module named 'serial.tools'

doudz, 2020-04-28:
ok, merci d'essayer la commande suivante également
sudo python3 -m pip uninstall pyserial

doudz, 2020-04-28:
par contre j'ai un doute sur la suppression de python3-serial car le dernier log semble indiqué que le paquet est toujours présent
`python3-serial is already the newest version (3.2.1-1).`

durdur33, 2020-04-28:
je vais reessayé

durdur33, 2020-04-28:
[image]

toujours même message

Installation des dépendances
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_12.x stretch InRelease
Fetched 25.4 kB in 2s (11.0 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-serial is already the newest version (3.2.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-usb is already the newest version (1.0.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-requests is already the newest version (2.12.4-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-setuptools is already the newest version (33.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-rpi.gpio is already the newest version (0.6.5~stretch-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-dev is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: zigate==0.38.6 in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: pyusb in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pyserial>=3.2 in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pydispatcher>=2.0.5 in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: bottle in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Installation des dépendances terminée !
Vérification de la version installée
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/usr/local/lib/python3.5/dist-packages/zigate/__init__.py", line 9, in <module>
from .core import (ZiGate, ZiGateWiFi, ZiGateGPIO)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 16, in <module>
from .transport import (ThreadSerialConnection,
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 12, in <module>
import serial.tools.list_ports
ImportError: No module named 'serial.tools'

durdur33, 2020-04-28:
j'ai rappliquer les deux commandes
sudo apt-get remove python3-serial
puis
sudo python3 -m pip uninstall pyserial

[image]

A priori le package est bien réinstallé
malgré cela le message reste le même :(

=======================================

Installation des dépendances
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:3 https://deb.nodesource.com/node_12.x stretch InRelease
Fetched 25.4 kB in 1s (15.2 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
Suggested packages:
python3-wxgtk3.0 | python3-wxgtk
The following NEW packages will be installed:
python3-serial
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (This frontend requires a controlling tty.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin:
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/65.3 kB of archives.
After this operation, 327 kB of additional disk space will be used.
Selecting previously unselected package python3-serial.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 101534 files and directories currently installed.)
Preparing to unpack .../python3-serial_3.2.1-1_all.deb ...
Unpacking python3-serial (3.2.1-1) ...
Setting up python3-serial (3.2.1-1) ...
Reading package lists...
Building dependency tree...
Reading state information...
python3-usb is already the newest version (1.0.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-requests is already the newest version (2.12.4-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-setuptools is already the newest version (33.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-rpi.gpio is already the newest version (0.6.5~stretch-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-dev is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: zigate==0.38.6 in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: pyserial>=3.2 in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pyusb in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: bottle in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from zigate==0.38.6)
Requirement already satisfied: pydispatcher>=2.0.5 in /usr/local/lib/python3.5/dist-packages (from zigate==0.38.6)
Installation des dépendances terminée !
Vérification de la version installée
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/usr/local/lib/python3.5/dist-packages/zigate/__init__.py", line 9, in <module>
from .core import (ZiGate, ZiGateWiFi, ZiGateGPIO)
File "/usr/local/lib/python3.5/dist-packages/zigate/core.py", line 16, in <module>
from .transport import (ThreadSerialConnection,
File "/usr/local/lib/python3.5/dist-packages/zigate/transport.py", line 12, in <module>
import serial.tools.list_ports
ImportError: No module named 'serial.tools'

doudz, 2020-04-28:
et que donne `sudo python3 -m pip uninstall pyserial`  ?

durdur33, 2020-04-28:
Dans la console du dessus lors de la suppression par atp
tu as le résultat

je l'ai refait après le lancement des dépendances
 voila le résultat

[image]

doudz, 2020-04-28:
On va tenter d'installer la version pip

sudo apt remove python3-serial
sudo python3 -m pip install pyserial

durdur33, 2020-04-28:
[image]

durdur33, 2020-04-28:
Yes

[image]

durdur33, 2020-04-28:
Merci
je vérifie maintenant que tous fonctionne

durdur33, 2020-04-28:
Bonsoir
Les périphériques Xiaomi et hue ne fonctionnaient plus
je les ai effacé  et re inclut et cela fonctionne a nouveau

Merci

closed 2020-04-29 by doudz

maximebsu, 2020-05-06:
Les dépendances sont OK mais c'est le démon qui ne veut jamais démarrer.

> [2020-05-06 14:04:22][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:04:45][ERROR][zigate](MainThread) : Failed to start network
[2020-05-06 14:04:50][ERROR][zigate](ZiGate-Listen) : OOPS connection lost, reconnect...
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 14:05:29][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:05:52][ERROR][zigate](MainThread) : Failed to start network
[2020-05-06 14:05:57][ERROR][zigate](ZiGate-Decode data) : Bad length 89 != 22 : b'\xab\x00\x00\x08\x18\x13
\x00\x00!\xb3\x1b\x95\xb1\x1d\x8c\xfe%D\x81\x00'
[2020-05-06 14:05:57][ERROR][zigate](ZiGate-Decode data) : Bad length 100 != 11 : b'\xab\x00\x00\x11\x18\x14
\x00\x00)'
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 14:07:28][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:07:38][ERROR][zigate](ZiGate-Decode data) : Bad length 138 != 22 : b'\xb0\x00\x00\x08\x18\x06
\x00\x00!\x86\x19\xban\x1d<\xfe%D\x81\x00'
[2020-05-06 14:07:38][ERROR][zigate](ZiGate-Decode data) : Bad length 149 != 11 : b'\xb0\x00\x00\x11\x18\x07
\x00\x00)'
[2020-05-06 14:07:51][ERROR][zigate](MainThread) : Failed to start network
[2020-05-06 14:07:56][ERROR][zigate](ZiGate-Listen) : OOPS connection lost, reconnect...
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 14:08:56][ERROR][zigate](ZiGate-Decode data) : Bad length 124 != 22 : b'\xb3\x00\x00\x08\x18\x12
\x00\x00!\xf2\x16\xban\x1dH\xfe%D\x81\x00'
[2020-05-06 14:08:56][ERROR][zigate](ZiGate-Decode data) : Bad length 135 != 11 : b'\xb3\x00\x00\x11\x18\x13
\x00\x00)'
[2020-05-06 14:09:01][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:09:23][ERROR][zigate](ZiGate-Decode data) : Bad length 114 != 22 : b'\xb5\x00\x00\x08\x18\x15
\x00\x00!a\x19\xban\x1d1\xfe%D\x81\x00'
[2020-05-06 14:09:23][ERROR][zigate](ZiGate-Decode data) : Bad length 125 != 11 : b'\xb5\x00\x00\x11\x18\x16
\x00\x00)'
[2020-05-06 14:09:24][ERROR][zigate](MainThread) : Failed to start network
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 14:10:34][ERROR][zigate](ZiGate-Decode data) : Bad length 134 != 22 : b'\xb8\x00\x00\x08\x18\x1e
\x00\x00!z\x1c\xban\x1de\xfe%D\x81\x00'
[2020-05-06 14:10:34][ERROR][zigate](ZiGate-Decode data) : Bad length 145 != 11 : b'\xb8\x00\x00\x11\x18\x1f
\x00\x00)'
[2020-05-06 14:10:50][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:10:58][ERROR][zigate](ZiGate-Decode data) : Bad length 103 != 22 : b'\xb9\x00\x00\x08\x18$
\x00\x00!\x98\x16\xban\x1d\x17\xfe%D\x81\x00'
[2020-05-06 14:10:58][ERROR][zigate](ZiGate-Decode data) : Bad length 114 != 11 : b'\xb9\x00\x00\x11\x18%
\x00\x00)'
[2020-05-06 14:11:13][ERROR][zigate](MainThread) : Failed to start network
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 14:12:10][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:12:33][ERROR][zigate](MainThread) : Failed to start network
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 15:19:32][ERROR] : Impossible d'arrêter le démon zigate, tuons-le
[2020-05-06 15:19:33][ERROR] : Erreur sur la fonction deamon_start du plugin : Vous devez attendre au moins 45 secondes entre deux lancements du démon. Dernier lancement : 2020-05-06 15:19:21
[2020-05-06 15:19:38][ERROR] : Erreur sur la fonction deamon_start du plugin : Vous devez attendre au moins 45 secondes entre deux lancements du démon. Dernier lancement : 2020-05-06 15:19:21
[2020-05-06 15:19:52][ERROR] : Erreur sur la fonction deamon_start du plugin : Vous devez attendre au moins 45 secondes entre deux lancements du démon. Dernier lancement : 2020-05-06 15:19:21
[2020-05-06 15:19:55][ERROR] : Erreur sur la fonction deamon_start du plugin : Vous devez attendre au moins 45 secondes entre deux lancements du démon. Dernier lancement : 2020-05-06 15:19:21
[2020-05-06 14:20:09][ERROR][zigate](ZiGate-Listen) : OOPS connection lost, reconnect...
[2020-05-06 14:20:34][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:20:57][ERROR][zigate](MainThread) : Failed to start network
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'
[2020-05-06 14:21:47][ERROR][zigate](ZiGate-Listen) : Malformed packet received, ignore it
[2020-05-06 14:22:06][ERROR][zigate](MainThread) : Failed to get network state
[2020-05-06 14:22:29][ERROR][zigate](MainThread) : Failed to start network
Traceback (most recent call last):
File "/var/www/html/plugins/zigate/core/class/../../resources/zigated/zigated.py", line 336, in <module>
if version < '3.0d':
TypeError: '<' not supported between instances of 'NoneType' and 'str'

doudz, 2020-05-06:
Le démon ne communique pas avec la ZiGate, c'est pour ça qu'il plante @maximebsu 
Il faut vérifier le port USB sélectionné

maximebsu, 2020-05-06:
j'ai beau choisir les ports ACM0, j'en ai d'ailleurs plusieurs, rien y fait :/

maximebsu, 2020-05-06:
[image]

doudz, 2020-05-06:
Quel type de Zigate ?

maximebsu, 2020-05-06:
C'est un sniffer cc2531
Ça a déjà fonctionné puisque j'ai chopé le signal d'une lampe ikea la semaine dernière avec domoticz. Mais la je voulais tout réinstaller proprement et migrer sur jeedom.

La clé est bien alimentee et la led c'est activée quand j'ai lancé le npm start zigbee2mqtt

doudz, 2020-05-07:
> C'est un sniffer cc2531
> Ça a déjà fonctionné puisque j'ai chopé le signal d'une lampe ikea la semaine dernière avec domoticz. Mais la je voulais tout réinstaller proprement et migrer sur jeedom.
> 
> La clé est bien alimentee et la led c'est activée quand j'ai lancé le npm start zigbee2mqtt

Le plugin Zigate ne gère que les clés Zigate équipé du firmware Zigate , sauf erreur de ma part un dongle cc2531 n'est pas une Zigate