doudz/zigate #184: Erreur lors de l'installation des dépendances

maximebsu, 2020-04-30:
Hello doudz.
J'ai jeedom sur rapsberry pi 4, et je n'arrive pas à installer zigate sur jeedom.
Il s'arrête toujours à 80% avant de s'arrêter.
Voici les logs :

File "<string>", line 1, in <module>
File "/usr/local/lib/python3.7/dist-packages/zigate/__init__.py", line 9, in <module>
from .core import (ZiGate, ZiGateWiFi, ZiGateGPIO)
File "/usr/local/lib/python3.7/dist-packages/zigate/core.py", line 16, in <module>
from .transport import (ThreadSerialConnection,
File "/usr/local/lib/python3.7/dist-packages/zigate/transport.py", line 12, in <module>
import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'

doudz, 2020-04-30:
Voir #182

doudz, 2020-04-30:
`sudo python3 -m pip install pyserial`