doudz/zigate #188: Unknown response 0x8035 on Aqara Window Door Sensor

ruimarinho, 2020-05-14:
One of my Aqara Door Sensors has stopped responding in a very strange way. It's not the battery as I replaced it by a brand new one.

There seems to be some kind of PDM memory corruption on the zigate that's generating a `0x8035` response when trying to add or remove this device.

Sometimes, under circumstance I have yet to understand, I get the occasional warning that the device already exists and should be renamed. 

```
2020-05-14 01:05:40 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16C\x02\x13\x02\x10\x02\x10\xf1\x02\x12\x02\x10\x03\x01\x805\x02\x10\x02\x16E\x02\x13\x02\x10\x02\x10\xf1\x02\x14\x02\x10\x03\x01\x805\x02\x10\x02\x16@\x02\x13\x02\x10\x02\x10\xf1\x02\x11\x02\x10\x03'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f102'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f104'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x02', lqi:0
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f101'
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x04', lqi:0
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x01', lqi:0
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03\x01\x805\x02\x10\x02\x16C\x02\x13\x02\x10\x02\x10\xf1\x02\x12\x02\x10\x03\x01'
2020-05-14 01:05:40 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x805\x02\x10\x02\x16E\x02\x13\x02\x10\x02\x10\xf1\x02\x14\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f103'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f103'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x03', lqi:0
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f102'
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x03', lqi:0
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f104'
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x02', lqi:0
2020-05-14 01:05:40 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f103'
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x04', lqi:0
2020-05-14 01:05:40 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x03', lqi:0
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:40 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:05:43 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16E\x02\x13\x02\x10\x02\x10\xf1\x02\x14\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13'
2020-05-14 01:05:43 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x02\x10\x03'
2020-05-14 01:05:43 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:43 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f104'
2020-05-14 01:05:43 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:05:43 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
```

When trying to reset and adding it again, the adding seems successful but then I don't get any events from it.

```
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'971b00158d0002c9ef2a8000'
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:971b, ieee:00158d0002c9ef2a, mac_capability:10000000, rejoin_status:False, lqi:0
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] Device Announce RESPONSE 0x004D - Device announce : addr:971b, ieee:00158d0002c9ef2a, mac_capability:10000000, rejoin_status:False, lqi:0
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] Acquire Lock on device Device  (971b) 00158d0002c9ef2a
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] Release Lock on device Device  (971b) 00158d0002c9ef2a
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update device Device  (971b) 00158d0002c9ef2a
2020-05-14 01:09:14 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:09:19 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16C\x02\x13\x02\x10\x02\x10\xf1\x02\x12\x02\x10\x03\x01\x805\x02\x10\x02\x16E\x02\x13\x02\x10\x02\x10\xf1\x02\x14\x02\x10\x03\x01\x805\x02\x10\x02\x16@\x02\x13\x02\x10\x02\x10\xf1\x02\x11\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03'
[... 72 lines cut ...]
2020-05-14 01:09:44 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:10:03 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16C\x02\x13\x02\x10\x02\x10\xf1\x02\x12\x02\x10\x03'
2020-05-14 01:10:03 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x805\x02\x10\x02\x16E\x02\x13\x02\x10\x02\x10\xf1\x02\x14\x02\x10\x03\x01\x805\x02\x10\x02\x16@\x02\x13\x02\x10\x02\x10\xf1\x02\x11\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03\x01\x805\x02\x10\x02\x16B\x02\x13\x02\x10\x02\x10\xf1\x02\x13\x02\x10\x03'
2020-05-14 01:10:03 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:10:03 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8035: b'030000f101'
2020-05-14 01:10:03 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-05-14 01:10:03 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x04', lqi:0
2020-05-14 01:10:03 WARNING (ZiGate-Decode data) [zigate] Unknown response 0x8035
[... cut ...]
2020-05-14 01:10:09 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x0000 - Base response : additional:b'\x03\x00\x00\xf1\x03', lqi:0
2020-05-14 01:10:09 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-05-14 01:10:10 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x02\x1e\xaf\xd6H\xe1\x02\x18\x02\x10\x02\x16\x02\x10\x02\x10\x02\x10 \x02\x10\x02\x11\x02\x10r\x03'
2020-05-14 01:10:10 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
```

Could this simply be a damaged device?

doudz, 2020-08-28:
@fairecasoimeme could you provide any help on this message 8035 with status 3 ?

doudz, 2020-08-28:
it's just a warning, but nothing to worry about

ruimarinho, 2020-08-28:
@doudz but the sensor isn’t functional and I can’t re-pair it..

doudz, 2020-08-29:
> @doudz but the sensor isn’t functional and I can’t re-pair it..

yes... maybe @fairecasoimeme can go deeper with that ?

ruimarinho, 2020-08-30:
I think my zigate device is having memory issues. @fairecasoimeme:

[image]